raspberrypi/firmware #1203: RPi 3b+ won't boot after update

servusDei2018, 2019-07-23:
I have a SSD attached to my RPi 3b+. **I was able to get it to boot directly from the SSD (without any SD card)**, but last Friday (July 19), after I `sudo apt-get update` and `sudo apt-get upgrade`, my Pi refused to boot from the SSD :(

When I copy a backup of `/boot` to the SSD, it boots, but there's lots of errors resulting from the old kernel without its modules -- `[FAILURE] Failed to start Load Kernel Modules`.

**I would appreciate any help to get my Pi booting either wholly or partially from my SSD again.**

popcornmix, 2019-07-23:
Do you have a backup of /lib/modules?
Do you know what when you last updated kernel?

You could use rpi-update to revert back to a stable kernel. See:
https://github.com/Hexxeh/rpi-firmware/commits/stable

Click on a commit from the right date and use the hash (the long hex number at end of url) as parameter to rpi-update and you should get a consistent firmware/kernel/modules.

servusDei2018, 2019-07-23:
Yes, I have a backup of /lib/modules (which I didn't know existed). I don't know when I last updated the kernel.

Inside `/lib/modules`, there are two folders: `4.19.57+` and `4.19.57-v7+`.

popcornmix, 2019-07-23:
Run ``uname -a`` to see what the kernel version you currently have is. If it matches the backed up ``/lib/modules`` then you should be able to copy them back.

If not and we can assume you were happy with kernel version (4.19.57) when you backed up modules then ad7aa88db0d8f367ef366fbe4dc34ddf20ec9cba is the right hash for 4.19.57 kernel.

```sudo rpi-update ad7aa88db0d8f367ef366fbe4dc34ddf20ec9cba```

should get you back to stable kernel 4.19.57.

servusDei2018, 2019-07-24:
Thanks all, so that worked. Just in case anyone else runs into the same issue, I did this:

* I deleted everything inside `/boot` and replaced it with my backup `/boot`,
* I copied the kernel modules from my backup `/lib/modules` to the SSD,
* I rebooted the Pi, and it ran from the SSD. After entering emergency mode, I
```
apt-get update
apt-get remove raspberrypi-kernel
apt-get raspberrypi-kernel
reboot
```

After that, everything worked fine, and running the kernel `4.19.57-v7+` :)

**Thanks again! Without your help I'd still be stuck!**

closed 2019-07-24 by servusDei2018

pelwell, 2019-07-24:
Thanks for the update. [I think you mean:
```
apt-get install raspberrypi-kernel
```
]

servusDei2018, 2019-07-24:
> Thanks for the update. [I think you mean:
> 
> ```
> apt-get install raspberrypi-kernel
> ```
> 
> ]

Lol, thanks for pointing that out :)