raspberrypi/firmware #1204: RPI 3B+ fails to boot with vc4-fkms-v3d and Hyperpixel 4 TFT screen on kernel 4.19.59

ande122, 2019-07-24:
Enabling vc4-fkms-v3d on Raspbian at kernel 4.19.59 causes kernel panic on boot. I can confirm that on kernel 4.19.40 this does not happen and screen is working perfectly with vc4-fkms-v3d.

# Screen working with rpi-update 6cb1da772a8c56f44591bec904046b15f1126949
pi@raspberrypi:~ $ vcgencmd version
Mar 27 2019 15:45:53 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.40-v7+ #1217 SMP Tue May 7 17:22:08 BST 2019 armv7l GNU/Linux

# Fails to boot with rpi-update 2096a8d567eed47cb742ddb1ce9bf9ac25726980
pi@raspberrypi:~ $ vcgencmd version
Jul 23 2019 19:29:32 
Copyright (c) 2012 Broadcom
version 4b11cff8f84f480b87a9e7a2a59d26014f4e7c91 (clean) (release) (start)
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.59-v7+ #1246 SMP Tue Jul 23 19:18:42 BST 2019 armv7l GNU/Linux

# /boot/config.txt
# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Enable GPU
dtoverlay=vc4-fkms-v3d

dtoverlay=hyperpixel4
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
framebuffer_width=800
framebuffer_height=480
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
display_rotate=3
hdmi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6

Patryk27, 2019-07-27:
Same happens on my RPi Zero W:

[image]

JamesH65, 2019-07-27:
I believe that a fix for an error very similar to this was merged end of last week, not sure if it will have made it in to rpi-update yet, but might be worth a try. @6by9

Patryk27, 2019-07-27:
I've just run `rpi-update` - the crash's still happening, although the cause seems to be different:

[image]

JamesH65, 2019-07-27:
I think one bug has been fixed and that has exposed another. I'll have to take a look Monday.

6by9, 2019-07-28:
Pi3 - this will be the funny one we were looking at last thing on Friday. Adding ```max_framebuffers=2``` to config.txt seemed to fix it, but for reasons that weren't immediately clear.

JamesH65, 2019-07-28:
Ah yes, getting my bugs mixed up. To the people seeing the issue, please try the max_framebuffers=2 to see if it helps.

Patryk27, 2019-07-28:
`max_framebuffers=2` avoids the crash, but instead the screen becomes unresponsive and just stops refreshing some time during the boot (usually just after the `Mounted /boot` message).

The entire OS is working though and I can connect to it using `ssh` + there's no error message in `dmesg`.

JamesH65, 2019-07-28:
Thanks for the report, does look like it's the issue I was looking at last thing Friday. Hopefully some movement on this early next week.

JamesH65, 2019-07-29:
Just pushed a prospective fix for at least some of this to our linux repo. See  https://github.com/raspberrypi/linux/pull/3110

It'll be a few days before this hits rpi-update, so you may wish to build your own kernel to try it out.

ChrKuhn, 2019-07-29:
It works for me, too. I have changed config.txt from

```
[all]
dtoverlay=vc4-fkms-v3d

```
to

```
[all]
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
framebuffer_priority=2

```
and now it works. Thank you very much.

JamesH65, 2019-07-30:
@ChrKuhn Note, just setting max_framebuffers avoids the problem, which is actually still there.  OK to use that until the actual fix get officially released though.