raspberrypi/firmware #1208: no sound no video with raspbery pi 4 since last firmware upgrade

mruwzum, 2019-07-26:
Upgraded from release 1.20190709  to  1.20190718 and no able of getting my sound working, now i just hear some white noise when the raspberry boots up or shouts down. And even not been able yet of play any videos through HDMI on any release. Any stable release soon?

pelwell, 2019-07-26:
1) Are you referring to sound through the headphone socket or over HDMI?
2) Is there an image on the HDMI monitor?
3) Are you saying you can't play a video? If so, what kind of video, and what tool are you using to play it?
4) Are you using the vc4-fkms-v3d driver (it's the default on for a new install on a Pi 4), or have you changed it?

mruwzum, 2019-07-26:
Thanks for answering so quickly @pelwell ! 

1. I've never been able of use any HDMI on any firmware with the pi4, just able to use soundcard with 1.20190709 through the headphone jack.
2. Nope
3. Not able to see the commandline or even execute kodi or startx. Can't start any video output. I can see a gui and controll my pi visually just through VNC.
4. Yes and uncommented it from /boot/config.txt with no results.

I've using all the raspberry pi versions and sizes and not able to do a normal HDMI use of my pi 4.

pelwell, 2019-07-26:
6. I don't understand why you have not reported the non-working HDMI before. Is your application normally "headless" (with no display)?

7. In your answer to (1) you mention a "soundcard". Is this an external device (a HAT, or a USB dongle), or do you mean the onboard audio?

mruwzum, 2019-07-26:
No external soundcard, using the pi's snd_bcm.

pelwell, 2019-07-26:
Does the display work with one of your other Pis? Please grab the EDID, decode it and post the results:
```
$ tvservice -d edid.bin
$ edidparser edid.bin
```
Run on this on the Pi4 if you can, and one of your other Pis.

I assume you have tried both HDMI ports?

Are you sure the micro HDMI cable is a good one? Or are you using a full-sized cable with an adaptor?

mruwzum, 2019-07-26:
Yes, it works perfectly and i've bought also a full-sized micro HDMI cable for the pi. 

Those are the results of my pi4:

> Enabling fuzzy format match...
Parsing edid.bin...
HDMI:EDID version 1.3, 1 extensions, unknown aspect ratio
HDMI:EDID features - videodef 0x80 !standby !suspend !active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 50-76 Hz, horizontal is 30-80 kHz, max pixel clock is 160 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is AAA
HDMI:EDID found preferred DMT detail timing format: 1366x768p @ 60 Hz (81)
HDMI:EDID found DMT detail timing format: 1280x768p @ 60 Hz (22)
HDMI:EDID established timing I/II bytes are A3 08 00
HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 8, 800x600p @ 56 Hz in established timing I/II
HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
HDMI:EDID standard timings block x 8: 0x81C0 0101 0101 0101 0101 0101 0101 0101 
HDMI:EDID found DMT format: code 85, 1280x720p @ 60 Hz (16:9) in standard timing 0
HDMI:EDID parsing v3 CEA extension 0
HDMI:EDID monitor support - underscan IT formats:no, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:4
HDMI:EDID found CEA detail timing format: 1280x720p @ 50 Hz (19)
HDMI:EDID found CEA detail timing format: 1920x1080i @ 50 Hz (20)
HDMI:EDID found CEA detail timing format: 720x480p @ 60 Hz (2)
HDMI:EDID found CEA detail timing format: 720x576p @ 50 Hz (17)
HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz (native)
HDMI:EDID found CEA format: code 5, 1920x1080i @ 60Hz 
HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz 
HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz 
HDMI:EDID found CEA format: code 7, 1440x480i @ 60Hz 
HDMI:EDID found CEA format: code 2, 720x480p @ 60Hz 
HDMI:EDID found CEA format: code 6, 1440x480i @ 60Hz 
HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz 
HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz 
HDMI:EDID found CEA format: code 20, 1920x1080i @ 50Hz 
HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz 
HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz 
HDMI:EDID found CEA format: code 22, 1440x576i @ 50Hz 
HDMI:EDID found CEA format: code 17, 720x576p @ 50Hz 
HDMI:EDID found CEA format: code 21, 1440x576i @ 50Hz 
HDMI:EDID found CEA format: code 32, 1920x1080p @ 24Hz 
HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48 kHz, sample size: 16|20 bits
HDMI:EDID found HDMI VSDB length 6
HDMI:EDID HDMI VSDB has physical address 1.0.0.0
HDMI:EDID HDMI VSDB supports AI:no, dual link DVI:no
HDMI:EDID HDMI VSDB deep colour support - 48-bit:no 36-bit:no 30-bit:no DC_yuv444:no
HDMI:EDID HDMI VSDB has no latency information
HDMI:EDID filtering formats with pixel clock unlimited MHz or h. blanking unlimited
HDMI:EDID best score mode initialised to DMT (4) 640x480p @ 60 Hz with pixel clock 25 MHz (score 0)
HDMI:EDID best score mode is now CEA (1) 640x480p @ 60 Hz with pixel clock 25 MHz (score 43432)
HDMI:EDID best score mode is now CEA (2) 720x480p @ 60 Hz with pixel clock 27 MHz (score 3045736)
HDMI:EDID CEA mode (3) 720x480p @ 60 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID CEA mode (4) 1280x720p @ 60 Hz with pixel clock 74 MHz has a score of 80296
HDMI:EDID DMT mode (4) 640x480p @ 60 Hz with pixel clock 25 MHz has a score of 36864
HDMI:EDID CEA mode (5) 1920x1080i @ 60 Hz with pixel clock 74 MHz has a score of 87208
HDMI:EDID CEA mode (6) 1440x480i @ 60 Hz with pixel clock 27 MHz has a score of 35368
HDMI:EDID CEA mode (7) 1440x480i @ 60 Hz with pixel clock 27 MHz has a score of 35368
HDMI:EDID DMT mode (8) 800x600p @ 56 Hz with pixel clock 36 MHz has a score of 53760
HDMI:EDID DMT mode (9) 800x600p @ 60 Hz with pixel clock 40 MHz has a score of 57600
HDMI:EDID CEA mode (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz has a score of 211624
HDMI:EDID DMT mode (16) 1024x768p @ 60 Hz with pixel clock 65 MHz has a score of 94370
HDMI:EDID CEA mode (17) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 2545736
HDMI:EDID CEA mode (18) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID best score mode is now CEA (19) 1280x720p @ 50 Hz with pixel clock 74 MHz (score 4071080)
HDMI:EDID CEA mode (20) 1920x1080i @ 50 Hz with pixel clock 74 MHz has a score of 3628680
HDMI:EDID CEA mode (21) 1440x576i @ 50 Hz with pixel clock 27 MHz has a score of 35368
[... 62 lines cut ...]
HDMI:EDID best score mode is now CEA (2) 720x480p @ 60 Hz with pixel clock 27 MHz (score 3045736)
HDMI:EDID CEA mode (3) 720x480p @ 60 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID CEA mode (4) 1280x720p @ 60 Hz with pixel clock 74 MHz has a score of 80296
HDMI:EDID DMT mode (4) 640x480p @ 60 Hz with pixel clock 25 MHz has a score of 36864
HDMI:EDID CEA mode (5) 1920x1080i @ 60 Hz with pixel clock 74 MHz has a score of 87208
HDMI:EDID CEA mode (6) 1440x480i @ 60 Hz with pixel clock 27 MHz has a score of 35368
HDMI:EDID CEA mode (7) 1440x480i @ 60 Hz with pixel clock 27 MHz has a score of 35368
HDMI:EDID DMT mode (8) 800x600p @ 56 Hz with pixel clock 36 MHz has a score of 53760
[... cut ...]
HDMI:EDID CEA mode (31) 1920x1080p @ 50 Hz with pixel clock 148 MHz has a score of 128680
HDMI:EDID CEA mode (32) 1920x1080p @ 24 Hz with pixel clock 74 MHz has a score of 74766
HDMI:EDID best score mode is now DMT (81) 1366x768p @ 60 Hz with pixel clock 85 MHz (score 5188835)
HDMI:EDID DMT mode (85) 1280x720p @ 60 Hz with pixel clock 74 MHz has a score of 135592
HDMI0:EDID preferred mode remained as DMT (81) 1366x768p @ 60 Hz with pixel clock 85 MHz
HDMI:EDID has HDMI support and audio support
edidparser exited with code 0

pelwell, 2019-07-26:
Did you try both HDMI ports on the Pi4?

It appears from the EDIDs that you used a different port on your monitor for the Pi4 (port 1) and Pi3 (port 2) - please try the Pi4 in port 1.

mruwzum, 2019-07-26:
Sorry, thats the pi4 port1:
>Enabling fuzzy format match...
Parsing edid.bin...
HDMI:EDID version 1.3, 1 extensions, unknown aspect ratio
HDMI:EDID features - videodef 0x80 !standby !suspend !active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 50-76 Hz, horizontal is 30-80 kHz, max pixel clock is 160 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is AAA
HDMI:EDID found preferred DMT detail timing format: 1366x768p @ 60 Hz (81)
HDMI:EDID found DMT detail timing format: 1280x768p @ 60 Hz (22)
HDMI:EDID established timing I/II bytes are A3 08 00
HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 8, 800x600p @ 56 Hz in established timing I/II
HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
HDMI:EDID standard timings block x 8: 0x81C0 0101 0101 0101 0101 0101 0101 0101 
HDMI:EDID found DMT format: code 85, 1280x720p @ 60 Hz (16:9) in standard timing 0
HDMI:EDID parsing v3 CEA extension 0
HDMI:EDID monitor support - underscan IT formats:no, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:4
HDMI:EDID found CEA detail timing format: 1280x720p @ 50 Hz (19)
HDMI:EDID found CEA detail timing format: 1920x1080i @ 50 Hz (20)
HDMI:EDID found CEA detail timing format: 720x480p @ 60 Hz (2)
HDMI:EDID found CEA detail timing format: 720x576p @ 50 Hz (17)
HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz (native)
HDMI:EDID found CEA format: code 5, 1920x1080i @ 60Hz 
HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz 
HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz 
HDMI:EDID found CEA format: code 7, 1440x480i @ 60Hz 
HDMI:EDID found CEA format: code 2, 720x480p @ 60Hz 
HDMI:EDID found CEA format: code 6, 1440x480i @ 60Hz 
HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz 
HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz 
HDMI:EDID found CEA format: code 20, 1920x1080i @ 50Hz 
HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz 
HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz 
HDMI:EDID found CEA format: code 22, 1440x576i @ 50Hz 
HDMI:EDID found CEA format: code 17, 720x576p @ 50Hz 
HDMI:EDID found CEA format: code 21, 1440x576i @ 50Hz 
HDMI:EDID found CEA format: code 32, 1920x1080p @ 24Hz 
HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48 kHz, sample size: 16|20 bits
HDMI:EDID found HDMI VSDB length 6
HDMI:EDID HDMI VSDB has physical address 1.0.0.0
HDMI:EDID HDMI VSDB supports AI:no, dual link DVI:no
HDMI:EDID HDMI VSDB deep colour support - 48-bit:no 36-bit:no 30-bit:no DC_yuv444:no
HDMI:EDID HDMI VSDB has no latency information
HDMI:EDID filtering formats with pixel clock unlimited MHz or h. blanking unlimited
HDMI:EDID best score mode initialised to DMT (4) 640x480p @ 60 Hz with pixel clock 25 MHz (score 0)
HDMI:EDID best score mode is now CEA (1) 640x480p @ 60 Hz with pixel clock 25 MHz (score 43432)
HDMI:EDID best score mode is now CEA (2) 720x480p @ 60 Hz with pixel clock 27 MHz (score 3045736)
HDMI:EDID CEA mode (3) 720x480p @ 60 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID CEA mode (4) 1280x720p @ 60 Hz with pixel clock 74 MHz has a score of 80296
HDMI:EDID DMT mode (4) 640x480p @ 60 Hz with pixel clock 25 MHz has a score of 36864
HDMI:EDID CEA mode (5) 1920x1080i @ 60 Hz with pixel clock 74 MHz has a score of 87208
HDMI:EDID CEA mode (6) 1440x480i @ 60 Hz with pixel clock 27 MHz has a score of 35368
HDMI:EDID CEA mode (7) 1440x480i @ 60 Hz with pixel clock 27 MHz has a score of 35368
HDMI:EDID DMT mode (8) 800x600p @ 56 Hz with pixel clock 36 MHz has a score of 53760
HDMI:EDID DMT mode (9) 800x600p @ 60 Hz with pixel clock 40 MHz has a score of 57600
HDMI:EDID CEA mode (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz has a score of 211624
HDMI:EDID DMT mode (16) 1024x768p @ 60 Hz with pixel clock 65 MHz has a score of 94370
HDMI:EDID CEA mode (17) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 2545736
HDMI:EDID CEA mode (18) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 45736
HDMI:EDID best score mode is now CEA (19) 1280x720p @ 50 Hz with pixel clock 74 MHz (score 4071080)
HDMI:EDID CEA mode (20) 1920x1080i @ 50 Hz with pixel clock 74 MHz has a score of 3628680
HDMI:EDID CEA mode (21) 1440x576i @ 50 Hz with pixel clock 27 MHz has a score of 35368
HDMI:EDID CEA mode (22) 1440x576i @ 50 Hz with pixel clock 27 MHz has a score of 35368
HDMI:EDID best score mode is now DMT (22) 1280x768p @ 60 Hz with pixel clock 68 MHz (score 4617964)
HDMI:EDID CEA mode (31) 1920x1080p @ 50 Hz with pixel clock 148 MHz has a score of 128680
HDMI:EDID CEA mode (32) 1920x1080p @ 24 Hz with pixel clock 74 MHz has a score of 74766
HDMI:EDID best score mode is now DMT (81) 1366x768p @ 60 Hz with pixel clock 85 MHz (score 5188835)
HDMI:EDID DMT mode (85) 1280x720p @ 60 Hz with pixel clock 74 MHz has a score of 135592
HDMI0:EDID preferred mode remained as DMT (81) 1366x768p @ 60 Hz with pixel clock 85 MHz
HDMI:EDID has HDMI support and audio support
edidparser exited with code 0

pelwell, 2019-07-26:
If you get no image out of either port on the Pi 4 and with either input on the monitor then it sounds like there is a fault on the Pi 4. You could try a `sudo rpi-update` to get the very latest firmware and kernel, but I don't expect it to make a difference.

mruwzum, 2019-07-26:
Don’t expect to make a difference? So don’t spect to get any video output never?

chriscamacho, 2019-07-26:
careful assuming the cable is good, I had an adaptor I could get edid data from a monitor with but no picture (but it did work on a different display) trying an alternative adaptor fixed my issue....

pelwell, 2019-07-27:
@chriscamacho makes a good point. Is your capable one of ours? It isn't that you have to use an "official" cable, more that we know they are well made - you get some _terrible_ cables - and we can rule out some possibilities.

JamesH65, 2019-07-30:
@mruwzum Any luck with a different cable?

mruwzum, 2019-07-31:
I've bought a new cable and also a adaptor too. It's not a cable or adaptor problem. I've realized that I can't use the DRM VC4 V3D driver. If I disable it, I can get video output, but no way to make it work with the new driver.

pelwell, 2019-07-31:
There isn't a full VC4 driver for Pi4 yet, so don't use vc4-kms-v3d on a Pi4. Why isn't vc4-fkms-v3d suitable for your use cases?

6by9, 2019-07-31:
Please post the raw edid.bin file (NOT the parsed output) if you believe there is an EDID parsing issue with your display and vc4-fkms-v3d.

mruwzum, 2019-07-31:
I've activated the driver again, connected it to HDMI0 and that's my raw edid.bin

00000000: 00000000 11111111 11111111 11111111 11111111 11111111  ......
00000006: 11111111 00000000 01100011 00011000 00000000 00000000  ..c...
0000000c: 00000000 00000000 00000000 00000000 00010010 00011000  ......
00000012: 00000001 00000011 10000000 00000000 00000000 00000000  ......
00000018: 00001010 11010111 10100101 10100010 01011001 01001010  ....YJ
0000001e: 10010110 00100100 00010100 01010000 01010100 10100011  .$.PT.
00000024: 00001000 00000000 10000001 11000000 00000001 00000001  ......
0000002a: 00000001 00000001 00000001 00000001 00000001 00000001  ......
00000030: 00000001 00000001 00000001 00000001 00000001 00000001  ......
00000036: 01111111 00100001 01010110 10101010 01010001 00000000  .!V.Q.
0000003c: 00011110 00110000 01000110 10001111 00110011 00000000  .0F.3.
00000042: 00111111 01000011 00100001 00000000 00000000 00011110  ?C!...
00000048: 10101001 00011010 00000000 10100000 01010000 00000000  ....P.
0000004e: 00010110 00110000 00110000 00100000 00110111 00000000  .00 7.
00000054: 00111111 01000011 00100001 00000000 00000000 00011010  ?C!...
0000005a: 00000000 00000000 00000000 11111101 00000000 00110010  .....2
00000060: 01001100 00011110 01010000 00010000 00000000 00001010  L.P...
00000066: 00100000 00100000 00100000 00100000 00100000 00100000        
0000006c: 00000000 00000000 00000000 11111100 00000000 01000001  .....A
00000072: 01000001 01000001 00001010 00100000 00100000 00100000  AA.   
00000078: 00100000 00100000 00100000 00100000 00100000 00100000        
0000007e: 00000001 10110100 00000010 00000011 00100100 01110100  ....$t
00000084: 01010000 10010000 00000101 00000100 00000011 00000111  P.....
0000008a: 00000010 00000110 00000001 00011111 00010100 00010011  ......
00000090: 00010010 00010110 00010001 00010101 00100000 00100011  .... #
00000096: 00001001 00000111 00000011 10000011 00000001 00000000  ......
0000009c: 00000000 01100110 00000011 00001100 00000000 00100000  .f... 
000000a2: 00000000 00000000 00000001 00011101 00000000 10111100  ......
000000a8: 01010010 11010000 00011110 00100000 10111000 00101000  R.. .(
000000ae: 01010101 01000000 11000100 10001110 00100001 00000000  U@..!.
000000b4: 00000000 00011110 00000001 00011101 10000000 11010000  ......
000000ba: 01110010 00011100 00010110 00100000 00010000 00101100  r.. .,
000000c0: 00100101 10000000 11000100 10001110 00100001 00000000  %...!.
000000c6: 00000000 10011110 10001100 00001010 11010000 10001010  ......
000000cc: 00100000 11100000 00101101 00010000 00010000 00111110   .-..>
000000d2: 10010110 00000000 00010011 10001110 00100001 00000000  ....!.
000000d8: 00000000 00011000 10001100 00001010 11010000 10010000  ......
000000de: 00100000 01000000 00110001 00100000 00001100 01000000   @1 .@
000000e4: 01010101 00000000 00010011 10001110 00100001 00000000  U...!.
000000ea: 00000000 00011000 00000000 00000000 00000000 00000000  ......
000000f0: 00000000 00000000 00000000 00000000 00000000 00000000  ......
000000f6: 00000000 00000000 00000000 00000000 00000000 00000000  ......
000000fc: 00000000 00000000 00000000 11110011                    ....

mruwzum, 2019-07-31:
Also got a couple of videos of the boot of my pi with the new driver or without it:
With the driver and no video:
https://a.uguu.se/KinGt1HpnA4z_IMG_0014.MOV

Without the driver:
https://a.uguu.se/07fIx11M5Nfw_IMG_0015.MOV

6by9, 2019-07-31:
Attach it as a file, not dumped via hexdump or similar. Otherwise I'll just be wasting time converting your dump back into a binary file.

6by9, 2019-07-31:
From your video it looks like DRM is choosing a different mode from the firmware, and your monitor doesn't like it. I'll have to run the EDID to work out what is going on.

pelwell, 2019-07-31:
`base64` is installed by default, works well for small files, and hexdumps annoy @popcornmix, so what's not to like?

mruwzum, 2019-07-31:
https://file.io/7PniMq

6by9, 2019-07-31:
https://file.io/7PniMq - 404 Page not found

mruwzum, 2019-07-31:
https://file.io/rkAHQq

mruwzum, 2019-07-31:
https://we.tl/t-znIGjCbTni

6by9, 2019-07-31:
Thank you. You can also drag and drop or paste files directly into the Github issue.

mruwzum, 2019-07-31:
> Thank you. You can also drag and drop or paste files directly into the Github issue.

Thanks, monday morning quarterback.

closed 2019-08-17 by mruwzum