raspberrypi/firmware #1209: Pi4 incompatible usb sata bridge

chriscamacho, 2019-07-27:
idVendor           0x152d JMicron Technology Corp. / JMicron USA Technology Corp.
  idProduct          0x0578 JMS567 SATA 6Gb/s bridge

initially this bridge appears to work fine and indeed reports faster hdparm speeds than with a different compatible bridge.  However if used to mount a root file system the PI will randomly lockup usually before the system is fully up.

P33M, 2019-07-27:
Please try the workaround detailed here:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245931

chriscamacho, 2019-07-28:
sorry, I completely missed that, it does indeed seem to be working, thanks!

closed 2019-07-28 by chriscamacho

Lipown, 2020-01-12:
Hi, this work around works, but not as much as UASP. This bridge should work with UASP but not working. Has somebody tried to update firmware of the device?