raspberrypi/firmware #1210: RPi 4, dpms force off, backlight remains on

mm191, 2019-07-27:
On the RPi 4, xset dpms force off (or xset dpms 0 0 60 etc) will only show a black screen, but not turn of the backlight of the screen.

Seems to be a bug: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=244425&p=1496192&hilit=backlight#p1496192

Any workarounds known?

6by9, 2019-07-28:
Duplicate of https://github.com/raspberrypi/linux/issues/3050
Not yet implemented.

JamesH65, 2019-07-28:
Please used the other issue.

crocy, 2019-08-02:
I'd like to reopen this issue since I have the exact same problem on an official 7" screen connected via the DSI connector on RPi 4B. The exact same display worked fine on RPi 3B.

The related issue #3050 talks about how this hasn't been implemented yet for a HDMI interface, but this uses a different interface yet results in the same problem. I can provide more info if needed.

6by9, 2019-08-02:
The raised issue didn't specify the display device, and the linked thread wasn't very clear, hence the initial reading that it was a duplicate.
@JamesH65 please reopen and rename to specify "DSI backlight remains on".

mm191, 2019-08-08:
> The raised issue didn't specify the display device, and the linked thread wasn't very clear, hence the initial reading that it was a duplicate.

I am new to Raspis and have also no knowledge about details of dpms, but I somebody can specify what kind of information can help to solve the problem, I can provide more information.

A strange behaviour that I observed in the meantime is the following (on dpms force off or dpms reaching the off timeout). It uses omxplayer and a small self-build python GUI.

- Backlight remains on
- A python gui (tkinter in case it matters) disappears
- omxplayer keeps showing and updating a live stream on the otherwise black screen with backlight turned on

When you move the mouse or press a key, thus forcing dpms to go back to "on", the screen blanks completely *including the backlight*. After approx. 1 second screen will turn on (backlight on and all windows shown) as expected after a mouse movement.

AGTDesigns, 2019-08-21:
Hi,
I've encountered a similar issue. With  RPI connected by DSI to a 7-inch touchscreen (official Raspberry), I do the following instructions on a start-up script, which worked flawlessly on a RPi3 with the same display:

#!/bin/bash
export DISPLAY=:0
xset s 10

On the Pi3, after 10 seconds of inactivity the backlight turned off, and the screen went black. If the mouse was moved/screen touched/key pressed the screen would turn back on.

On the Pi4, the screen goes black, the backlight keeps on, and it does not awake back when I move the mouse / press any key. 

Is there any specific additional information I can provide to help solve this issue? I'm using the latest Raspbian, and did made a apt-upgrade.

Thanks!

crocy, 2019-09-09:
Any progress on this?

nor500, 2019-10-01:
Isn`t there any progress yet?? Nearly two month has gone...Is there any hardware limitation of the pi4, which cannot be overcome by firmware side to make hdmi displays to go standby mode to save power?

JamesH65, 2019-10-01:
Sorry only just seen the request to reopen. I doubt anyone has looked at this, we are fairly busy on more important stuff. Now its reopened there might be some progress.

JamesH65, 2019-10-01:
@nor500 This thread is for DSI display backlight according to the original and subsequent posts. This one is for HDMI #3050

frankgould, 2019-10-03:
I would like to know when the screen will turn off using dpms. This is not related to #3050, this is firmware controlling hardware, not command line.

JamesH65, 2019-10-03:
I'll try and take a look at this next week. Just to confirm, the official LCD is not turning its backlight off when `xset dpms force off`. To confirm, this is when using the Mesa driver (FKMS enabled), on a Pi4, using the latest Buster and firmware?

AGTDesigns, 2019-10-03:
Can you expand a bit on the "Mesa driver (FKMS enabled)"? I've downloaded the latest Operating system and tried the commands that work with no problem in the RPi3, and the RPi4 does not turn the backlight off. I'm trying to switch off the backlight after 10seconds.

AGTDesigns, 2019-10-03:
I've opened #1234 a while back when this thread was closed, but all the info is there. Thanks for the help!

6by9, 2019-10-03:
Pi 0-3 had several modes of operation:
- "legacy" (the default). The firmware does all 3D and composition
- "Full KMS". The Linux kernel does all 3D and composition. Things that use DispmanX for putting overlays on the screen (eg omxplayer and raspivid) will fail.
- "Fake KMS" (aka FKMS). The Linux kernel does all 3D, but composition is done in the firmware but supporting the DRM/KMS kernel APIs. 

Pi4 doesn't have a firmware 3D driver, therefore legacy mode will drop everything into software rendering. 
FKMS is the default on Pi4, so xset etc now has to go through the DRM/KMS APIs, and not the framebuffer APIs.
(Full KMS is still being worked on for the Pi4)

frankgould, 2019-10-03:
I just added the following issue similar to this original post but regarding HDMI monitors, not DSI monitors. They appear to both behave the same by **not** turning off the screen.

https://github.com/raspberrypi/firmware/issues/1260

JamesH65, 2019-11-05:
Fix for this in rpi-update - reports on whether the fix is working would be useful. Please close if fixed.

nor500, 2019-11-05:
I have already updated to latest firmware. My firmware version is:
$ /opt/vc/bin/vcgencmd version
Nov  1 2019 14:39:41 
Copyright (c) 2012 Broadcom
version 2faac616a157856a1d205f3a9713a7130defa643 (clean) (release) (start)

What the firmware did was completely turn off hdmi port instead of monitor standby.

My dpms settings:
$ xset q:

DPMS (Energy Star):
  Standby: 60    Suspend: 65535    Off: 65535
  DPMS is Enabled

So with that is settings the hdmi port will be turned off in 60s isntead of sending the monitor into standby state.
With the same settings raspberry 2, or raspberry 3 is sending the monitor to  standby state properly.

So hdmi blanking is still not working properly with this newest firmware.

GXTracker, 2020-01-03:
I just ran rpi-update on a Pi4B 4GB and after a reboot the backlight is now going off when triggered through xscreensaver when it wasnt before.

Unfortunately though when you touch the screen to wake it up the backlight will activate at full brightness. If you set it back to a specific brightness with: 

`echo 40 | sudo tee /sys/class/backlight/rpi_backlight/brightness`

then the brightness stays at that until the screen blanks again, then it resets to 255 again upon touch.

JamesH65, 2020-01-03:
That is a known issue, and is still an issue because I haven't figured out a way of fixing it. There currently no way of determining what the current brightness is from the driver (no idea why this API is missing - it's fairly old code), so when the command to turn off (by setting back light to 0) is executed I cannot store the previous value. Although I may have just had an idea...will have a think and check out the backlight driver code.