raspberrypi/firmware #1213: Firmware mailbox hang after disabling EMMC2 clock

vianpl, 2019-07-29:
Seen while working on bringing up the board on 64bits with mainline Linux.

With that setup, the `sdhci-iproc` driver (emmc2 in dtb) probe happens in parallel with all firmware dependent devices'. It turns out `sdhci-iproc` can't successfully probe as it depends on some GPIO controlled voltage regulator, returning -EPROBE_DEFER. One of the clean-up routines is `clk_disable_unprepare(pltfm_host->clk)`, which disables BCM2835_CLOCK_EMMC2.

For some reason the firmware mailbox interface stops responding after this call (I get timeouts). If re-enabled it'll come back alive. I carefully double-checked all the clk code and it matches the downstream implementation, so I'm somewhat confident it might be a firmware issue.

Any ideas?

lategoodbye, 2019-07-31:
As workaround you could mark this clock as critical in the clk-bcm2835 driver. But i agree, that you want to know the reason behind this.

popcornmix, 2019-07-31:
Can you identify the mailbox messsage (with length/params) being sent?
e.g. can you run /opt/vc/bin/vcmailbox with the same parameters and see the same failure?

lategoodbye, 2019-07-31:
The clk-bcm2835 driver doesn't use the mailbox interface, it uses the registers directly.

popcornmix, 2019-07-31:
You might need to identify the exact register write that kills firmware.
If you are just disabling emmc2 clock then I wouldn't expect any problems.
But there are many other registers that could kill firmware (e.g. disabling its clock).

vianpl, 2019-07-31:
> Can you identify the mailbox messsage (with length/params) being sent?

Anything going trough the _property tag_ mailboxes would fail (as in anything going trough the mailbox driver).

> e.g. can you run /opt/vc/bin/vcmailbox with the same parameters and see the same failure?

I'll give it a try later.

> You might need to identify the exact register write that kills firmware.

The offending line of code was this one, using the EMMC2 reg/offset (extracted from the clk driver's `bcm2835_clock_off()`):
```
cprman_write(cprman, data->ctl_reg,
		         cprman_read(cprman, data->ctl_reg) & ~CM_ENABLE);
```
That said I'll provide the exact address and changes.

lategoodbye, 2019-07-31:
@vianpl Could you please identify your firmware version?

vianpl, 2019-07-31:
> @vianpl Could you please identify your firmware version?

Yes sorry,  I reproduced it both with: 20190718 (latest) and 20190620 (first to support the RPi4)

lategoodbye, 2019-07-31:
@vianpl Could you please provide your config or do you use arm64/defconfig?

ptesarik, 2019-07-31:
@vianpl Are you sure your `data->ctl_reg` is indeed `CM_EMMC2CTL`?

I have tried with a crude user-space program that writes the registers through /dev/mem, and I was not able to get the hanging behaviour of the firmware. See:

```
#define _FILE_OFFSET_BITS 64

#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <sys/mman.h>
#include <unistd.h>
#include <stdio.h>

static const char devmem[] = "/dev/mem";

/* cprman */
#define CM_BASE 0xfe101000

#define CM_EMMC2CTL (CM_BASE + 0x1d0)
#define CM_EMMC2DIV (CM_BASE + 0x1d4)

#define CM_PASSWORD 0x5a000000

static unsigned long pagesize;

static unsigned readreg(int fd, unsigned long addr)
{
	unsigned *regs;
	unsigned val;

	regs = mmap(NULL, pagesize, PROT_READ, MAP_SHARED,
		    fd, addr & ~(off_t)(pagesize - 1));
	val = regs[(addr & (pagesize - 1)) / sizeof(*regs)];
	munmap(regs, pagesize);
	return val;
}

static void writereg(int fd, unsigned long addr, unsigned val)
{
	unsigned *regs;

	regs = mmap(NULL, pagesize, PROT_WRITE, MAP_SHARED,
		    fd, addr & ~(off_t)(pagesize - 1));
	regs[(addr & (pagesize - 1)) / sizeof(*regs)] = val;
	munmap(regs, pagesize);
}

int main()
{
	int fd;
	unsigned ctl, oldctl;
	unsigned div, olddiv;

	pagesize = sysconf(_SC_PAGESIZE);

	fd = open(devmem, O_RDWR | O_SYNC);
	if (fd < 0) {
		perror("Cannot open memory device");
		return 1;
	}

	oldctl = readreg(fd, CM_EMMC2CTL);
	olddiv = readreg(fd, CM_EMMC2DIV);
	writereg(fd, CM_EMMC2CTL, CM_PASSWORD | (readreg(fd, CM_EMMC2CTL) & ~(1U<<4)));
	ctl = readreg(fd, CM_EMMC2CTL);
	div = readreg(fd, CM_EMMC2DIV);
	printf("CM_EMMC2CTL: 0x%08x -> 0x%08x\n", oldctl, ctl);
	printf("CM_EMMC2DIV: 0x%08x -> 0x%08x\n", olddiv, div);

	puts("Press Enter...");
	getchar();

	oldctl = readreg(fd, CM_EMMC2CTL);
	olddiv = readreg(fd, CM_EMMC2DIV);
	writereg(fd, CM_EMMC2CTL, CM_PASSWORD | readreg(fd, CM_EMMC2CTL) | (1U<<4));
	ctl = readreg(fd, CM_EMMC2CTL);
	div = readreg(fd, CM_EMMC2DIV);
	printf("CM_EMMC2CTL: 0x%08x -> 0x%08x\n", oldctl, ctl);
	printf("CM_EMMC2DIV: 0x%08x -> 0x%08x\n", olddiv, div);

	close(fd);
	return 0;
}
```

While the program was waiting for input, I could still run mailbox commands in another terminal.

vianpl, 2019-07-31:
> @vianpl Could you please provide your config or do you use arm64/defconfig?

It's my bcm2711-arm64 branch with arm64's default defconfig.

> @vianpl Are you sure your `data->ctl_reg` is indeed `CM_EMMC2CTL`?
> 
> I have tried with a crude user-space program that writes the registers through /dev/mem, and I was not able to get the hanging behaviour of the firmware.

You're right, it was not the right clock, it's it's parent who's causing trouble: `plld_per`. I missed that `clk_core_unprepare()` also calls unprepare on the clock's parent (recursively).  I observed it's specifically the `plld_per` code that 's causing trouble, not `plld` nor `emmc2`.

As a proof I edited your small program and it now makes the mailbox fail as promised (BTW thanks for that :smile: )

```
#define _FILE_OFFSET_BITS 64

#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <sys/mman.h>
#include <unistd.h>
#include <stdio.h>

static const char devmem[] = "/dev/mem";

/* cprman */
#define CM_BASE 0xfe101000

#define CM_PLLD			(CM_BASE+ 0x10c)
#define A2W_PLLD_PER		(CM_BASE + 0x1540)

#define CM_PLLD_LOADPER		(1 << 6)
#define CM_PLLD_HOLDPER		(1 << 7)

#define A2W_PLL_CHANNEL_DISABLE	(1 << 8)

#define CM_PASSWORD 0x5a000000

static unsigned long pagesize;

static unsigned readreg(int fd, unsigned long addr)
{
	unsigned *regs;
	unsigned val;

	regs = mmap(NULL, pagesize, PROT_READ, MAP_SHARED,
		    fd, addr & ~(off_t)(pagesize - 1));
	val = regs[(addr & (pagesize - 1)) / sizeof(*regs)];
	munmap(regs, pagesize);
	return val;
}

static void writereg(int fd, unsigned long addr, unsigned val)
{
	unsigned *regs;

	regs = mmap(NULL, pagesize, PROT_WRITE, MAP_SHARED,
		    fd, addr & ~(off_t)(pagesize - 1));
	regs[(addr & (pagesize - 1)) / sizeof(*regs)] = val;
	munmap(regs, pagesize);
}

int main()
{
	int fd;
	unsigned cmreg, oldcmreg;
	unsigned a2wreg, olda2wreg;

	pagesize = sysconf(_SC_PAGESIZE);

	fd = open(devmem, O_RDWR | O_SYNC);
	if (fd < 0) {
		perror("Cannot open memory device");
		return 1;
	}

	oldcmreg = readreg(fd, CM_PLLD);
	olda2wreg = readreg(fd, A2W_PLLD_PER);
	writereg(fd, CM_PLLD, CM_PASSWORD | ((oldcmreg & ~CM_PLLD_LOADPER) | CM_PLLD_HOLDPER));
	writereg(fd, A2W_PLLD_PER, CM_PASSWORD | (olda2wreg | A2W_PLL_CHANNEL_DISABLE));
	cmreg = readreg(fd, CM_PLLD);
	a2wreg = readreg(fd, A2W_PLLD_PER);
	printf("CM_PLLD: 0x%08x -> 0x%08x\n", oldcmreg, cmreg);
	printf("A2W_PLLD_PER: 0x%08x -> 0x%08x\n", olda2wreg, a2wreg);

	puts("Press Enter...");
	getchar();

	oldcmreg = readreg(fd, CM_PLLD);
	olda2wreg = readreg(fd, A2W_PLLD_PER);
	writereg(fd, A2W_PLLD_PER, CM_PASSWORD | (olda2wreg & ~A2W_PLL_CHANNEL_DISABLE));
	writereg(fd, CM_PLLD, CM_PASSWORD | (oldcmreg & ~CM_PLLD_HOLDPER));
	cmreg = readreg(fd, CM_PLLD);
	a2wreg = readreg(fd, A2W_PLLD_PER);
	printf("CM_PLLD: 0x%08x -> 0x%08x\n", oldcmreg, cmreg);
	printf("A2W_PLLD_PER: 0x%08x -> 0x%08x\n", olda2wreg, a2wreg);

	close(fd);
	return 0;
}
```

lategoodbye, 2019-07-31:
The more important question i think is, why the clock core disables `plld_per`. The clock should be flagged as CLK_IGNORE_UNUSED, so it shouldn't been disabled.

lategoodbye, 2019-07-31:
Some time ago, i asked @anholt about some clock patches which hasn't applied to upstream. One of them was the [following](https://github.com/raspberrypi/linux/commit/783b1d3fbfe95aea79cf9bdb8d82d65a3be770dd#diff-f332406d43d20f215b1bcb1168dcaadc). At that time it was basically necessary for firmware-kms.

@vianpl Does adding CLK_IS_CRITICAL to the plld_per flags fix the issue?

timg236, 2019-08-01:
PLLD is designed to be the fixed clock source and is never modified. Various bits of bridge infrastructure rely upon this so stopping or changing PLLD will likely cause the lockups.

timg236, 2019-08-01:
I think this can be closed. Changing the CM post-dividers is reasonably safe but the firmware will assume that peripheral channels are not modified by the host-OS.

vianpl, 2019-08-01:
> The more important question i think is, why the clock core disables `plld_per`?

Well `emmc2` seems to be the sole user of the divisor at that specific moment, so the clk core automatically unprepares it (and the PLL too for the same reasons). I guess it makes sense from the point of view of the clk framework. 

> Some time ago, i asked @anholt about some clock patches which hasn't applied to upstream. One of them was the [...]

I tested that patch, it fixes the `plld_per` issue, yet generates other issues on it's own:

```
[    0.349206] bcm2835-clk fe101000.cprman: pllh: couldn't lock PLL
[    0.349237] Enabling unprepared pllh_rcal_prediv
[    0.349293] WARNING: CPU: 1 PID: 1 at drivers/clk/clk.c:984 clk_core_enable+0x98/0xb0
[    0.349309] Modules linked in:
[    0.349328] CPU: 1 PID: 1 Comm: swapper/0 Not tainted 5.3.0-rc2-00077-ga6c23787c97e-dirty #597
[    0.349344] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[...]
[    0.349630] Call trace:
[    0.349645]  clk_core_enable+0x98/0xb0
[    0.349658]  __clk_register+0x578/0x748
[    0.349672]  devm_clk_hw_register+0x4c/0xb0
[    0.349686]  bcm2835_register_pll_divider+0x144/0x1a0
[    0.349700]  bcm2835_clk_probe+0x118/0x1c8
[    0.349716]  platform_drv_probe+0x50/0xa0
[    0.349730]  really_probe+0xdc/0x290
[    0.349743]  driver_probe_device+0x54/0xe8
[...]
```
I'm looking into it, I think I'll try to revamp the patch as it's seems the right solution to @timg236 comment : "Changing the CM post-dividers is reasonably safe but the firmware will assume that peripheral channels are not modified by the host-OS."

> Edit: Another question, is this behavior specific to the BCM2711?

Don't know, I'll test it later.

@timg236 I know it's not at fw issue anymore, I'll close it as soon as we agree on how to fix it.

timg236, 2019-08-01:
PLLH doesn't exist on 2711. The HDMI 2.0 core has its own PLL for the pixel output and the state-machine clock doesn't need to be actively managed. This will probably be more obvious once we get around to doing a a vanilla KMS + HDMI driver, but that won't be any time soon since FKMS is the priority as it provides backwards compatibility for dispmanx etc.

vianpl, 2019-08-01:
@timg236 Thanks, I was about to ask you about `pllh` :slightly_smiling_face: 

@lategoodbye I guess we'll have to remove `pllh` from 2711 using your latest SoC clock filtering patches. And then apply on top of that phill's patch. Do you want me to do a pull request on top of bcm2711-initial-v2 or you'll take care of it?

lategoodbye, 2019-08-11:
@timg236 Since we need to adapt the clk driver now for BCM2711, we need to modelize all the clocks properly. Are there any other BCM2835 clocks (beside PLLH) which are not available for BCM2711?

timg236, 2019-08-12:
I don't think any other clocks have been removed. This change might also be of interest
https://github.com/raspberrypi/linux/pull/3140/commits/c9c27b7a88869664714eb8b9713507ff0500ad2c

lategoodbye, 2019-08-12:
Thanks for clarification.

The mentioned change is currently not critical, since we start without official V3D support in mainline. An acceptable solution for mainline, would be to write a clk driver for the V3D clock. Otherwise the driver will only be useable with the Raspberry Pi.

vianpl, 2019-08-20:
As I've seen @lategoodbye updated his series with fixes both for PLLH and PLLD_PER I think we're good. Closing.

closed 2019-08-20 by vianpl