raspberrypi/firmware #1214: Chromium Video blocked by selecting Analog audio in latest rpi-update

ThreeMileJump, 2019-07-31:
Last week there was an rpi-update which introduced a bug in chromium video playback. 

I have 2 Pi 4 systems setup (one at work and one at home). I did the rpi-update on both systems, and one stopped playing youtubes in chromium and the other one didn't. I reverted to the previous rpi-update and the problem disappeared. 

It turned out that the only difference between the two setups was that the working one used HDMI audio and the one that broke used analog audio. I tested this and while playing a youtube in chromium, setting the audio to analog freezes the playback, changing back to HDMI resumes playback. There seems to be no effect on any other graphics program apart from chromium videos.

The previous image identifies as  4.19.58-v7l+  and the new problematic version as  4.19.60-v7l+

Details of the updated image -

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi4B 4G

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage5

* Which firmware version (`vcgencmd version`)?
Jul 25 2019 17:52:38 
Copyright (c) 2012 Broadcom
version f09a65b56d867fa06b7c572d4bacea3e25fe5756 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.60-v7l+ #1247 SMP Thu Jul 25 14:54:07 BST 2019 armv7l GNU/Linux

pelwell, 2019-07-31:
I've literally just submitted a fix for this to the internal repo. Pi4 has an extra PWM block which is dedicated to the audio output. The initial Pi4 firmware was incapable of redirecting the audio to the header pins (12&13 or 18&19) because that meant reverting to the other PWM block. I added a patch on the 23rd that re-enabled access to PWM0, but between testing and pushing I dropped a necessary change, breaking the audio output - sorry.

We should be able to push a firmware update in the next few days, initially available via rpi-update. Until the fix is released to the stable branch, the easiest way to revert to the previous working version is to run `sudo rpi-update 35e22a2`.

popcornmix, 2019-08-01:
This is hopefully fixed in latest rpi-update firmware.

ThreeMileJump, 2019-08-03:
Confirmed as fixed.

closed 2019-08-03 by pelwell