raspberrypi/firmware #1215: New AWB algorithm breaks AWB gains after resolution change

jtc42, 2019-08-01:
**Describe the bug**
When using the new AWB algorithm, setting AWB gains is broken after changing the camera resolution. Upon changing the resolution, the AWB gains seem to reset to their Auto values, and changing them back is broken.

**To reproduce**
Easiest to demonstrate in Python
```
import picamera as cam
import time

c = cam.PiCamera()
c.start_preview()

# Set the AWB mode to change the gains
c.awb_mode = "tungsten"
time.sleep(2)

# Show, and store, the gains it settled on
print(c.awb_gains)
g = c.awb_gains

# Turn auto AWB off, fixing the gains
c.awb_mode = "off"
time.sleep(2)

# Store the current resolution
r=c.resolution

# Apply a "new" resolution
c.resolution=r

print("At this point, the AWB gains will have reset if using the new AWB algorithm")
time.sleep(1)

# Try to set the AWB gains back
c.awb_gains = g

# Note: it didn't work
print(c.awb_gains)
# Wait for it to work...
time.sleep(2)

# It still doesn't work...
print(c.awb_gains)
print(c.awb_mode)
c.stop_preview()
```

**Expected behaviour**
1) Changing resolution shouldn't reset the AWB gains
2) AWB gains should be changeable after this anyway, if AWB is off.

**Actual behaviour**
1) Changing resolution resets AWB gains, seemingly to their values when AWB is set to "auto"
2) Turning the AWB off and trying to set new AWB gains after this point silently fails. AWB gains remain fixed at their reset values.

**System**
* Pi3B+
* Firmware 38e81f25e639d19fc0ce6e67fd39998c340a15d5, and 1.20190718 tested

**Additional context**
These issues seem related:
https://github.com/waveform80/picamera/issues/581
https://github.com/waveform80/picamera/issues/580

H4R0, 2019-08-24:
@jtc42 i got the same problem after upgrading, could you find any solution for it ?

Do you know if the bug occurs in the kernel or in the picamera library ? Because i would like to downgrade to fix it.

I was able to set awb_gains before, now it does not change.

My test Code:

> from picamera import PiCamera
camera = PiCamera(resolution=(2560, 1440), framerate=10)
time.sleep(3)
print("exposure: " + str(camera.exposure_speed))
print("awb_gains: " + str(camera.awb_gains))
camera.exposure_mode = 'off'
camera.awb_mode = 'off'
camera.awb_gains = (Fraction(1, 4), Fraction(225, 256))
time.sleep(2)
print("awb_gains: " + str(camera.awb_gains))
camera.awb_gains = (Fraction(1, 100), Fraction(50, 100))
time.sleep(2)
print("awb_gains: " + str(camera.awb_gains))
camera.awb_gains = 1
time.sleep(2)
print("awb_gains: " + str(camera.awb_gains))
camera.awb_gains = 2.0
time.sleep(2)
print("awb_gains: " + str(camera.awb_gains))

Output:
> exposure: 3240
awb_gains: (Fraction(201, 128), Fraction(307, 256))
awb_gains: (Fraction(201, 128), Fraction(307, 256))
awb_gains: (Fraction(201, 128), Fraction(307, 256))
awb_gains: (Fraction(201, 128), Fraction(307, 256))
awb_gains: (Fraction(201, 128), Fraction(307, 256))

H4R0, 2019-08-24:
@jtc42 i got the same problem on a raspberry pi zero running  4.19.57+

Could you find any fix for now besides downgrading ?

JeffJassky, 2019-08-31:
For anyone else experiencing this, this work-around shared by @6by9 works well for me. Run this code before running raspistill, raspivid, or using picamera.

`sudo vcdbg set awb_mode 0`

6by9, 2019-08-31:
The fix for the issue is in the latest rpi-update firmware.

```sudo vcdbg set awb_mode 0``` just switches back to the old algorithm, which doesn't help on a Pi4 as the old algorithm can't work due to the change in 3D hardware. Whilst it will work in the short term, it can not be recommended longer term.

JeffJassky, 2019-08-31:
@6by9 thank you!

And for reference - how can I switch it back to the new awb algorithm?

H4R0, 2019-09-25:
For 2 days i have the problem now that even "sudo vcdbg set awb_mode 0" doesnt effectivly works.

I take pictures every 10 minutes with picamera and around 15-30% of the pictures have visually no awb_gain.

Im not changing the awb_gains value between shots.

Its totally random and i have no idea why, nothing in the system logs either.

The only thing that changed is that i installed upgrades with apt upgrade, but there was no kernel bump.

I rebooted the pi but it didnt fix it. 

Anyone has the same bug ?

Does someone know how long it will take for the kernel update to be available via apt upgrade ? Has been a month already.

dgalland, 2020-03-30:
With the new firmware and the new awb algorithm I also note the following behavior:
- When I try the different modes (sunlight, cloudy, ...) it's OK
- When I go to awb_mode off and I fix the gains it's OK
- But after I can't change awb_mode which stays off
The behavior is normal if I return to the old algorithm with sudo vcdbg set awb_mode 0
Note: from Python with Picamera 1.13
Regards

JamesH65, 2020-03-30:
vcdbg set_awb_mode is no longer recommended (and will be removed at some point). Use the greyworld AWB option.

6by9, 2020-03-30:
Resuming AWB after having switched off was fixed in https://github.com/Hexxeh/rpi-firmware/commit/15261920fa8cb85efed3920c715d10e5b1e848dd - 30 Jan  2020.
Is your firmware after that date? If not, then update.

dgalland, 2020-03-30:
@ 6by9
Thank you for answering.
I did rpi-update is that enough?
Here is what uname -a gives
Linux raspberrypi 4.19.113-v7+ #1300 SMP Thu Mar 26 16:53:09 GMT 2020 armv7l GNU/Linux
Not really blocking for me but I wanted to report
Anyway in my case the awb auto algorithm did not really give satisfactory results. Do you think the new algorithm is better?
@JamesH65 
is there a trick to use "Greyworld with the Picamera library?

6by9, 2020-03-30:
Blast. Sorry, we have two firmware source branches at present, one for Pi0-3, and one for Pi4. The fix is only on the Pi4 branch. I'll pull it across to the other one.

Re using greyworld, see https://github.com/waveform80/picamera/pull/576.
I know waveform80 is busy on other stuff at present, but we'll look at what can be done to get it merged.

6by9, 2020-03-30:
Actually if you want an interim fix on a non-critical Pi, then ```sudo BRANCH=next rpi-update``` will get a test firmware where the Pi0-3 fw is built from the same tree as Pi4.  See https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=267576 for more info.

Cherry-pick done, so all being well a fix would be in the next "normal" rpi-update anyway.

dgalland, 2020-04-02:
@6by9
It also seems to me that there is a change in behavior with the AGC algorithm. I posted in the raspberry forum, I don't know if I should open an issue?
Thanks for your help and best regards.

6by9, 2020-04-02:
@dgalland Your comment is totally unrelated to AWB changing, therefore does not belong on this issue at all.

If you have a reproducable test case then raise an issue, but seeing as you've posted on the forums it'd be nice if you left more than 5 minutes for someone to have a chance to reply before cross posting here. And always add a link if you have asked the question on the forums so that we don't cover the same ground twice.

jtc42, 2020-04-28:
@6by9 thanks for the updates on this. Has the updated pi3 firmware been officially released now? We're currently still using the old AWB debug mode for our microscopes, but I'm keen to move away from that ASAP.

Thanks.

6by9, 2020-04-28:
https://github.com/Hexxeh/rpi-firmware/commit/837c2456a7ea5a2e17d54378e7e8d5f221c54e1a
```firmware: isp/tuner: Resetting to a lamp mode cancels manual_gains_used_ (master)```

But also note https://github.com/Hexxeh/rpi-firmware/commit/b6e327eae03bb047d6b49f3debce004c689730e1
```firmware: Switch to building from common firmware branch```
so all Pi variants from that point on are now built off the same firmware tree.

yujiayabe, 2020-06-15:
I checked many comments, but my Pi and camera don't work well. I can't stop AWB. 
I check AWB with seeing picture(movie) using red paper covering lens on/off.
camera.exposure_mode = 'off' moves well.

- RasperryPi 4
- PiCamera V2.1
- Distributor ID:	Raspbian
  Description:	Raspbian GNU/Linux 10 (buster)
  Release:	10 
  Codename:	buster
- Linux raspberrypi 5.4.45-v7l+ #1321 SMP Wed Jun 10 17:39:20 BST 2020 armv7l GNU/Linux
- Python 3.7.3
- openCV 3.2.0

```
# -*- coding: utf-8 -*-
import cv2
import numpy as np
import picamera
import time

from picamera.array import PiRGBArray
from picamera import PiCamera

def main():
    #camera mode
    #camera mode
    camera = PiCamera()
    camera.resolution = (832, 624)
    camera.iso = 100    
    camera.shutter_speed = 10000
    camera.exposure_mode = 'off'
    capture = PiRGBArray(camera, size=(832, 624))
    camera.awb_mode = 'off'
    camera.awb_gains = (1.8,1.8)
    
    # allow the camera to warmup
    time.sleep(0.1)

    for frame in camera.capture_continuous(capture, format="bgr", use_video_port=True):
        image = frame.array
        hsv = cv2.cvtColor(image, cv2.COLOR_BGR2HSV)

        cv2.imshow("Image", image)
        
        key = cv2.waitKey(1) & 0xFF
        capture.truncate(0)

if __name__ == '__main__':
    main() 

```

Regards.