raspberrypi/firmware #1216: Error in pairing with Secure Simple Pairing (SSP) disabled

Douglas6, 2019-08-02:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

Douglas6, 2019-08-02:
There seems to be a problem with Bluetooth pairing when Secure Simple Pairing is disabled. Several posts on this forum alerted me to the problem (which I can verify): https://www.raspberrypi.org/forums/view ... 9&t=195090 and https://www.raspberrypi.org/forums/view ... 8&t=244241. 

To reproduce the issue:
1. Flash the latest Buster full image and upgrade as of today. 
2. Disable SSP with 'sudo hciconfig hci0 sspmode 0'
3. Enter bluetoothctl and make the Pi discoverable. 
4. Attempt to pair from another device (I used an Android phone running Pie, ironically.)
5. No joy. 

An hcidump transcription goes like this:

HCI sniffer - Bluetooth packet analyzer ver 5.50
device: hci0 snap_len: 1500 filter: 0xffffffff
> HCI Event: Command Complete (0x0e) plen 4
    Write Simple Pairing Mode (0x03|0x0056) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    Write Current IAC LAP (0x03|0x003a) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    Write Scan Enable (0x03|0x001a) ncmd 1
    status 0x00
> HCI Event: Connect Request (0x04) plen 10
    bdaddr 0C:CB:85:0D:6B:AB class 0x5a020c type ACL
> HCI Event: Command Status (0x0f) plen 4
    Accept Connection Request (0x01|0x0009) status 0x00 ncmd 1
> HCI Event: Command Complete (0x0e) plen 10
    PIN Code Request Negative Reply (0x01|0x000e) ncmd 1
    status 0x00 bdaddr 0C:CB:85:0D:6B:AB
> HCI Event: Connect Complete (0x03) plen 11
    status 0x18 handle 12 bdaddr 0C:CB:85:0D:6B:AB type ACL encrypt 0x00
    Error: Pairing Not Allowed

Attempts with a dongle that does not support SSP worked fine, which is why I suspect Raspbian rather than BlueZ (as well as the lack of outcry from other distros).

pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage4
pi@raspberrypi:~ $ vcgencmd version
Jul  9 2019 14:40:53
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start)
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux

pelwell, 2019-08-02:
Did you read "Is this the right place for my bug report?"? Your report has nothing to do with the firmware - the [linux](https://github.com/raspberrypi/linux) repo is a better place to start.

closed 2019-08-02 by pelwell