raspberrypi/firmware #1217: Excessive overscan on Pi4

prusa3duser, 2019-08-03:
I have installed the latest Raspian Buster Lite (that is, there is no graphics, just console) on the SDCard for the PI 4 (1 GB model).

The Pi starts to boots up fine and I can see the raspberry logos and the boot up sequence executing, but a few seconds into the boot up the Pi seems to change resolution and continues to execute more steps in the sequence, but after that resolution change the overscan hides the first 2 characters on the left of the screen, and eventually when it finishes booting, the LOGIN prompt is not visible at all (it is off the bottom of the screen).

I tried playing with the overscan and other display settings in raspi-config but wasn't able to do much.

The Pi is connected via HDMI 0 (closest to the USB-C) to an LG television set (US Model).  

I tried the same sdcard on Pi Zero, Pi 3 B+, Pi 2  and connected to the same TV and it works just fine (it does not do the resolution change).

Could you please help me troubleshoot?

mullcom, 2019-08-03:
Maybe power drop? Have you test to boot it up with raspberry power adapter
instead?

send from mobile

6by9, 2019-08-04:
Your monitor/TV is applying the overscan, not the Pi.

The Pi firmware driver has code in to effectively crop the size of the framebuffer to allow for the display cropping.
With the Pi4, the driver swaps to the Linux mainline DRM/KMS drivers part way through boot (the mode switch you observe). This brings the Pi in line with all other Linux distributions, however it has only just received support for setting up overscan margins. I'm waiting on https://github.com/raspberrypi/linux/pull/3103 to be merged, and then you can set up margins from the kernel command line. At that point the firmware can be updated to pass through the config.txt overscan settings to the kernel.

At present you can remove/comment out the line ```dtoverlay=vc4-fkms-v3d``` from config.txt. This will disable the DRM/KMS driver so the overscan options will be active, but also means you have no 3D acceleration on your system.

There may be some useful information about your display in the EDID. If you run ```tvservice -d edid.dat```, zip that file, and then attach it to this issue (drag and drop into your web browser), then we can analyse it. I'm not holding out any great hopes though.

prusa3duser, 2019-08-06:
@mullcom : I am using the original raspberry pi 4 power adapter.
@6by9 : Thanks for the description of what is happening...  Please see attached edid.dat.

BTW, the TV is Samsung not LG.  Also, I noticed that tvservice --status reports differences between the raspberry pi 3 and pi 4, I don't know if this information is relevant or not but here it is:

PI 3 B+
---------
pi@raspberrypi:~ $ tvservice --status
state 0xa [HDMI CEA (34) RGB lim 16:9], 1920x1080 @ 30.00Hz, progressive

PI 4
[edid.dat.gz](https://github.com/raspberrypi/firmware/files/3473781/edid.dat.gz)

----
pi@raspberrypi:~ $ tvservice --status
state 0xa [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive

persson1589, 2019-09-27:
Im using the rasberry pie 4, and I've changed every option to try to customize overscan in /boot/config.txt, but the screen remains the same. Overscan isn't working whatsoever for me. What am I doing wrong?

popcornmix, 2019-09-30:
@persson1589 overscan support was only added recently for Pi4 with fkms. (September 3rd).
If you have an older firmware (``vcgencmd version``) then you can update with rpi-update or wait for support to reach "apt upgrade" firmware (an update is expected in next week or so).

persson1589, 2019-10-01:
@popcornmix Thanks for the tip, I just downloaded the 26 september release of raspbian and now overscan is working!

knuckleheadsmiff, 2019-10-09:
Finally! Thanks for this info. It was baffling that it was never working before.

ashwinterminator, 2019-11-16:
> 
> 
> @persson1589 overscan support was only added recently for Pi4 with fkms. (September 3rd).
> If you have an older firmware (`vcgencmd version`) then you can update with rpi-update or wait for support to reach "apt upgrade" firmware (an update is expected in next week or so).

I installed the latest Raspbian and installed all the updates through sudo apt-get update and upgrade as well.

But the overscan settings don't seem to work with composite output. Is that a known issue? The overscan settings do work when I comment out dtoverlay=vc4-fkms-v3d though

pelwell, 2019-11-16:
Overscan on the composite output is covered by https://github.com/raspberrypi/firmware/issues/1285.

alanbork, 2021-05-14:
seems like this has been fixed and can be closed?

persson1589, 2021-05-14:
yes

closed 2023-12-09 by hobbieman