raspberrypi/firmware #1223: Pi 4 - Terrible performance with enable_tvout=1

ashwinterminator, 2019-08-09:
**Describe the bug**
On the latest Raspbian, when enable_tvout is set to 1 on config.txt, the Pi 4 takes upto 10 minutes to completely boot. Kodi never opens and Chromium is unusable. But the same is not noticed when enable_tvout is absent from the config.txt.

**To reproduce**
1) In config.txt, add enable_tvout=1
2) Reboot

**Expected behaviour**
The https://www.raspberrypi.org/documentation/configuration/config-txt/video.md says performance will be hit by enable_tvout=1 to a small degree. But the device is completely unusable.

**Actual behaviour**
On the latest Raspbian, I set enable_tvout is set to 1 on config.txt, the Pi 4 took 10 minutes to completely boot. Kodi never opens and Chromium is unusable. But the same is not noticed when enable_tvout is absent from the config.txt.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Pi4
* Which OS and version (`cat /etc/rpi-issue`)? Raspbian 2019-07-10

6by9, 2019-08-10:
Should already have been fixed via https://github.com/Hexxeh/rpi-firmware/commit/2096a8d567eed47cb742ddb1ce9bf9ac25726980
Please use rpi-update and retest.

ashwinterminator, 2019-08-11:
Updated to the Aug 9th firmware using rpi-update

But the issue persists. The 4 raspberries come on, a splash screen for a split second and a heartbreaking grey screen again. :(

ashwinterminator, 2019-08-12:
> 
> 
> Should already have been fixed via [Hexxeh/rpi-firmware@2096a8d](https://github.com/Hexxeh/rpi-firmware/commit/2096a8d567eed47cb742ddb1ce9bf9ac25726980)
> Please use rpi-update and retest.

Updated to the Aug 9th firmware using rpi-update

But the issue persists. The 4 raspberries come on, a splash screen for a split second and a heartbreaking grey screen again. :(

UPDATE: Left it on for at least 10 minutes. It was stuck on the grey screen and never went into the desktop at all.

ashwinterminator, 2019-08-19:
Any idea on when this might be resolved? My Pi has been lying around because of this :\

szachrisson, 2019-09-02:
I have the same experience on my Raspberry 4. Enable_tvout = 1 in the config and when I start up I see the rainbow-square, followed by four pi-logos and then the welcome screen which you barely see before it all turns black. I boughed my Pi 4 to plug in to my CRT TV. Is there a way to get around this or is a software update on the way?

ashwinterminator, 2019-09-02:
> I have the same experience on my Raspberry 4. Enable_tvout = 1 in the config and when I start up I see the rainbow-square, followed by four pi-logos and then the welcome screen which you barely see [...]

Still no luck for me.

I tried updating the latest raspbian with the sudo apt-get dist-upgrade command and tried. No difference.

I then did an rpi-update and tried again. Still no difference.

I even posted on their forums: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=250154, but no one is replying.

And they categorically say they don't provide e-mail support. Not sure what else we an do

pelwell, 2019-09-02:
You can be patient. It's still early days for Pi 4, and as well as being a small team with many projects we're just coming to the end of the holiday season. Normal service should be restored fairly soon.

JamesH65, 2019-09-02:
We are listening, and when I get a chance will look at this. When I was testing composite prior to launch I didn't see the problem, so perhaps something has happened since then. Anything unusual on your setups that might be relevant?

ashwinterminator, 2019-09-02:
> 
> 
> We are listening, and when I get a chance will look at this. When I was testing composite prior to launch I didn't see the problem, so perhaps something has happened since then. Anything unusual on your setups that might be relevant?

Thank you for the reply, @JamesH65 

You had mentioned you pushed a bug-fix for this here: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245655&p=1522243#p1501372 (this is around 1.5 months ago)

But though I ran rpi-update today and tested, the situation has not changed. Any idea why? uname-a mentioned the version date as 30th August as well.

As for anything unusual about the setup,
Location > India
Language > English
Time Zone > Kolkata
Connected over a 2.4Ghz WiFi connection
No Password set (behaviour was the same with a password too)
Running Raspbian Buster Full loaded onto a 32GB Class 10 SanDisk card using Etcher

No apps installed

Ran dist-upgrade and checked - no improvement
Ran rpi-update and checked - no improvement

Anything other information that might help you understand the problem?

szachrisson, 2019-09-02:
I found a reddit-thread which had a solution which I havn't tried yet. https://www.reddit.com/r/raspberry_pi/comments/cqjdgn/raspberry_pi_4b_slow_boot_on_composite_out

> Hey all, I've been struggling with getting my Pi 4 to boot on composite output. Basically, I launch my pi and I am brought to a black screen with 4 Raspberries. Then, it cuts to the "welcome to pi" white screen for a split-second, before black-screening again. After about 3 minutes, a default full-screen terminal prompt pops up. I leave this for about a minute and then it blackscreens. Then, about 3 mins later, normal raspberry pi desktop loads and I can use the board as intended. Anyone have any clue why this might be happening/advice for how to fix? I refuse to use microhdmi, as I purchased this for use with a crt tv!

> FIXED!!!
Solution: Disable splash screen via ssh using raspi-config. Also, boot to terminal without login. Login with credentials, run startx, and it works!!!

szachrisson, 2019-09-06:
I disabled splash screen by removing the "quiet splash" in cmdline to see the boot sequence and tried to boot to terminal without login as explained in [this article](https://www.kiwi-electronics.nl/blog?journal_blog_post_id=5&lang=en). Sadly, the screen turned black before the terminal would appear.

I'm not sure if this is helping but this is the last thing that I see in the console before it turns black. It seems everything loaded ok and that something happened when leaving the booting sequence? https://imgur.com/Hi9DJBv

szachrisson, 2019-09-09:
@pelwell @JamesH65 is there anything you want us to test to assist you further? Do you think that composite video output is a general problem with all devices or do you know of a way to just get to the terminal with analog video?

JamesH65, 2019-09-09:
It wasn't something we saw during bringup/release, so I'll need to investigate - will try to do something this week.

ashwinterminator, 2019-09-09:
> 
> 
> I found a reddit-thread which had a solution which I haven't tried yet. https://www.reddit.com/r/raspberry_pi/comments/cqjdgn/raspberry_pi_4b_slow_boot_on_composite_out
> 
> > Hey all, I've been struggling with getting my Pi 4 to boot on composite output. Basically, I launch my pi and I am brought to a black screen with 4 Raspberries. Then, it cuts to the "welcome to pi" white screen for a split-second, before black-screening again. After about 3 minutes, a default full-screen terminal prompt pops up. I leave this for about a minute and then it blackscreens. Then, about 3 mins later, normal raspberry pi desktop loads and I can use the board as intended. Anyone have any clue why this might be happening/advice for how to fix? I refuse to use microhdmi, as I purchased this for use with a crt tv!
> 
> > FIXED!!!
> > Solution: Disable splash screen via ssh using raspi-config. Also, boot to terminal without login. Login with credentials, run startx, and it works!!!

I tried this. But it made no difference at all :\

Let's hope they can find a fix for this sometime this week. Thanks for your responses, @JamesH65! :)

szachrisson, 2019-09-09:
Thank you @JamesH65 for your reply! Good luck with investigating :)

JamesH65, 2019-09-10:
This definitely appears to be an issue, so just to document what I have seen so far so I don't forget..

During debugging a number of possible issues have come to light, in firmware and in Linux space.  

1. During startup, there is an assert in the firmware as a clock is out of range.  This is probably due to something not realising that composite, which requires a slower clock, is enabled.
2. There are a number of HDMI initialisations that are done even if composite is enabled, which should probably be disabled, although I suspect they are harmless.
3. With FKMS enabled, the DRM driver is rejecting the mode being passed to it for some reason - I believe it thinks the frame rate is out of range when composite is selected.
4. With FKMS disabled, it starts out fine, but the firmware then receives a mailbox, presumably from the ARM framebuffer driver, to set the framebuffer up to HD mode (3560x2160) - which is much too high for composite. Not yet sure what has change to cause that. However, the composite monitor I have DOES display something, unlike when using FKMS.

I need to talk to the DRM expert to sort this out and he's back in the office tomorrow.

JamesH65, 2019-09-11:
OK, think we have got to the bottom of this. I'm pushed a couple of patches out for code review so hopefully will hit the real world in a couple of days.

ashwinterminator, 2019-09-13:
> 
> 
> OK, think we have got to the bottom of this. I'm pushed a couple of patches out for code review so hopefully will hit the real world in a couple of days.

Thanks, @JamesH65! I presume it will currently be available through rpi-update and not dist-upgrade?

JamesH65, 2019-09-13:
I doubt it's made it to rpi-update yet. I know one change to the firmware has been merged, but there is still a change in FKMS required, that hasn't been, and it takes a few days to get from merged to rpi-update.

ashwinterminator, 2019-09-13:
> 
> 
> I doubt it's made it to rpi-update yet. I know one change to the firmware has been merged, but there is still a change in FKMS required, that hasn't been, and it takes a few days to get from merged to rpi-update.

Thank you for your reply, @JamesH65 .

Will await your update. :)

JamesH65, 2019-09-13:
Sorry, just checked, and both PR's have been merged, so will be in rpi-update fairly soon I suspect. Last one was 4 days ago, so probably this weekend sometime.

popcornmix, 2019-09-13:
rpi-update firmware should now contain the fixes.

ashwinterminator, 2019-09-13:
Thank you, @popcornmix & @JamesH65!

I am sorry if I was rude earlier. :)

Will update the Pi this weekend and report back here.

JamesH65, 2019-09-13:
Rude? Not that I was aware!

szachrisson, 2019-09-13:
I can confirm everything is working properly as far as I can see! Both video and audio working as expected. Thank you very much for fixing this and being transparent and open in communicating with us!

JamesH65, 2019-09-13:
No worries! Thanks for letting us know its working. I'll leave this issue open for a few more days so more people can test it more thoroughly.

54dfg254, 2019-09-15:
Works great for me.

Lakka also has this problem. Can you explain exactly what you changed? The developers over in the Lakka issue for this problem aren't sure what was changed. https://github.com/libretro/Lakka-LibreELEC/issues/743

Thanks.

ashwinterminator, 2019-09-16:
> 
> 
> No worries! Thanks for letting us know its working. I'll leave this issue open for a few more days so more people can test it more thoroughly.

Works just fine, @JamesH65! Worked flawlessly for a good 5 hours yesterday. Thank you very much! :)

I will close the issue now.

closed 2019-09-16 by ashwinterminator

JamesH65, 2019-09-16:
> Works great for me.
> 
> Lakka also has this problem. Can you explain exactly what you changed? The developers over in the Lakka issue for this problem aren't sure what was changed. [libretro/Lakka-LibreELEC#743](https://github.com/libretro/Lakka-LibreELEC/issues/743)
> 
> Thanks.

There was a fix to the fkms driver  (https://github.com/raspberrypi/linux/pull/3223), plus a fix to the firmware, which is currently in rpi-update.