raspberrypi/firmware #1224: RPi4B does not support vcgencmd display_power 0

frankgould, 2019-08-12:
**Is this the right place for my bug report?**
I was referred here by RPF forum moderator.

**Describe the bug**
I am migrating my app from the previous version of Raspbian that supported the `vcgencmd display_power 0` command to turn off RPi4B attached HDMI monitor. It does not appear to work returns `display_power=1` but does not turn off the monitor. This same SDcard installed in _RPi3B+ does turn the monitor on/off using vcgencmd_. 

I tried `tvservice -o` but that appears to suspend all processes and my app never returns after a 60 second scheduled return. Only after I issue the `tvservice -p` command does my app return to restore the HDMI monitor using `tvservice -p.` I cannot find the TV address using cec-client. What is the recommended way for Buster to turn off and on the HDMI monitor?

**To reproduce**
With HDMI monitor attached to RPi4B, CLI 'vcgencmd display_power 0'

**Expected behaviour**
HDMI monitor turned off.

**Actual behaviour**
Nothing but echos 'display_power=1' in terminal

**System**
https://pastebin.com/haejHaZ3

* Which model of Raspberry Pi? 
When inserting RPi4B SDcard in RPi3B+, HDMI off/on works with vcgencmd

**Logs**
see pastebin above

**Additional context**
Add any other relevant context for the problem.

popcornmix, 2019-08-12:
This is the right place. It is something we are aware of but it's not top of the list of issues.

Basically the HDMI hardware is different on Pi4 and 4kp60 scrambling adds another layer of difficulty to achieving this.

It will come but we can't say when.

6by9, 2019-08-12:
https://github.com/raspberrypi/linux/issues/3050

Janghou, 2019-08-22:
IMHO this issue is not only a nuisance for users but also harming the planet. 

KODI doesn't turn off idle monitors, burning unnecessary WATTS for hampering monitors turning off / going into powersave mode.

E.g. this great plugin stopped working on RPI4 LibreElec: 
https://github.com/asavah/script.pidisplaypower 

A workaround could be using tvservice -o, and tvservice -p, but tvservice seems to disable more than just HDMI output.

Also: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=244425&p=1493575&hilit=hdmi+power+off+monitor+raspberry+4

LorenKeagle, 2019-09-06:
This also affects those of us using the Pi 4 with various MagicMirror setups.  The scripts that turn on/off the display based on sensor input (e.g. PIR sensor) depend on 'vcgencmd display_power <x>'.

popcornmix, 2019-09-09:
Latest firmware should support vcgencmd display_power on Pi4.

Janghou, 2019-09-12:
Not working in LibreElec 9.2 Beta1 (11-09-2019)  LibreELEC-RPi4.arm-9.1.501 (Kernel 4.19.66)

popcornmix, 2019-09-13:
@Janghou @MilhouseVH latest rpi-update firmware should fix querying the display_power.

giddyhup, 2019-10-21:
@popcornmix, it doesn't appear fixed for me.
```
$ vcgencmd version

Sep 24 2019 17:34:30
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start)
```

giddyhup, 2019-10-21:
I should mention that I use HDMI output 1 only, not HDMI 0.

JamesH65, 2019-10-21:
@giddyhup That version is too old. Do a `sudo apt update && sudo apt full-upgrade` and see if that helps.

giddyhup, 2019-10-21:
@JamesH65 Thanks for the comment but I think I'm as up-to-date as I can be:
```
$ sudo apt update && sudo apt full-upgrade
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:3 http://repo.mosquitto.org/debian buster InRelease
Hit:4 https://deb.nodesource.com/node_10.x buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

```
I actually did an rpi-update earlier to get the latest fixes.

JamesH65, 2019-10-21:
Hmm, just tried this and its not working for me either, which is weird because I was looking at this the other day and ISTR it working fine. I might have a dodgy test firmware installed, so will need to check. however, off on a few days off now, so will need to be next week.

giddyhup, 2019-10-28:
@JamesH65 could this issue please be reopened?

frankgould, 2019-10-28:
I just wanted to comment that my original post was satisfied, this LibreElec problem is not what I reported or experienced, and this LibreElec report should be a new issue, not reopen my original issue. We also need the fix for dpms to power off the display but that's already in process.

JamesH65, 2019-10-28:
I've just been trying vcgencmd display_power 0 on a Pi4, and it doesn't seem to work - there's some odd stuff going on when I run it through the debugger.  Will double check.

JamesH65, 2019-10-28:
Ah, I'm using HDMI1 only which doesn't work correctly.  I'll need to figure out how this reacts with the other change going through.

JamesH65, 2019-10-28:
OK, check the other changes awaiting merge and this should be fixed when those go through. In the meantime, it should work if you use HDMI 0 rather than HDMI 1.

giddyhup, 2019-10-28:
Regrettably, I cant use HDMI 0 since the adapter I use is too wide and doesn't fit next to the power connector. I'll wait for the fix.

Somehow related (not needed by me right now, though), are there plans to give control over the individual HDMI connectors (e.g. turning HDMI 1 on and HDMI 0 off)?

JamesH65, 2019-10-28:
That requires a change to the gencmd, which I will do after the main changes are merged - the underlying code will be able to support it at that point. In fact the code change adds a new mailbox that allows you to specify which display you want to turn on/off, including HDMI, Composite, and DSI/DPI., so much more comprehensive than before.

frankgould, 2019-10-28:
Will this new "mailbox" cause code change from the original vcgencmd? While we're pontificating, will dpms also support powering off the monitor?

JamesH65, 2019-10-28:
The mailbox change doesn't change the vcgencmd. A subsequent merge will change it to add an extra optional parameter which is the display number you want to turn off. It will retain backwards compatibility. 

In my tests running FKMS, the monitor does power off (to power save mode) when the screen saver kicks in (providing the screen saver setup is set correctly). ISTR also testing it from xset and that also did the same. Note, this is all requires FKMS to be running.  It may be possible to change legacy mode to do the same, I just don't know how. Might be done in the framebuffer driver, but not sure.

frankgould, 2019-10-28:
Thanks @JamesH65 for the mailbox clarification.

In my tests running FKMS, the monitor does blank the screen but the backlight is still on and the power indicator stays blue, in that it's getting an HDMI signal. It needs to kill power to the monitor so the energy saver kicks in and the light goes red. I let the blank screen sit for a while to see if it would power the monitor off but nothing changed in 15 seconds or so.

JamesH65, 2019-10-28:
@frankgould This is exactly the fix currently being code reviewed. I'll try and hurry it along.

frankgould, 2019-10-28:
@JamesH65 No rush necessary but thank you for the offer. I'm patient these days and appreciate your attention to the matter.

It's so cool to have the RPi4B running a desktop so nicely with a few side issues that are just inconvenient. Getting them fixed is especially rewarding and time it takes for quality.

Thanks for all you reps do!

JamesH65, 2019-10-28:
Starting to trickle through. Have also sorted out the legacy (pre-FKMS)  framebuffer driver so that will also power down the display correctly when screensaver kicks in but that requires everything else to be merged first. Should be good to go in a few days.

mohyddintash, 2019-11-15:
Hello @JamesH65! I am also running Buster on RPI 4, and I haven't succeeded yet in figuring out how to turn off specific HDMI port and not both. I tried `tvservice -o -v 7` and it worked but when I did `tvservice -p -v 7` I only got a blank screen, however, `vcgencmd display_power 0`  worked but it turned both ports off at the same time. Do you have an idea when will this fix be released? and do you have a suggestion about using tvservice -p -v7 instead?
This is critical for my case as I use one of the HDMI ports for 3d printer projector.
Thanks Alot!!

JamesH65, 2019-11-15:
I'd need to check whether turning off an individual port is possible. Cannot quite remember how the code works.

mohyddintash, 2019-11-16:
@JamesH65 Thank you so much for your help! I will also be grateful if you can confirm if it is not possible to turn off an individual port, thus not to waste my time working on it. I think that it could be done using tvservice but I can't yet get my head around how to bring the monitor back to display the same window, something to do with fbset and x server refresh but hadn't worked for me yet.

JamesH65, 2019-11-16:
I'll take a look at the code when I get the chance. I vaguely remember thinking about this at some point, ISTR maybe you can turn off individual ports, but maybe tvservice doesn't take advantage of that? Will need to double check.

mohyddintash, 2019-11-17:
@JamesH65 Thanks a lot mate! I hope it will be possible. If you can, please let me know when you find out.

mohyddintash, 2019-11-20:
@JamesH65 Someone pointed out that they added support for individually turning on/off the HDMI ports in the development firmware [here](https://github.com/raspberrypi/firmware/commit/b79618b5db4e4e7c02f9ad9d3ada51713825313e), in case someone also needs this, but I'm not quite sure how long this usually takes to be pushed to stable :/

JamesH65, 2019-11-20:
You can use vcgencmd display_power to turn off individual displays.  See docs here https://www.raspberrypi.org/documentation/raspbian/applications/vcgencmd.md

Note, there's an incoming PR to update these docs to give a bit more information, but should be enough to get by. for the moment.

mohyddintash, 2019-11-20:
@JamesH65 Oh that is great!! Thank you so much just what I need! but I'm a bit confused, how long has this been in the docs? I feel silly for missing it because it was supposed to be still not pushed to stable!

JamesH65, 2019-11-20:
@mohyddintash All this stuff is recent - in the last month. So not surprised you had missed it. Of course, I'm assuming it has been pushed! There is a bug fix going through at the moment to do with return codes for this, but that has no effect of functionality.

mohyddintash, 2019-11-20:
@JamesH65  Glad to hear this! Thank you very much to you and all the contributors out there!! 👍

mohyddintash, 2019-11-21:
@JamesH65 Sorry, for the late feedback, but I've just tried to run `vcgencmd display_power 0 7
` after doing apt update then apt upgrade, but I'm still getting `error=2 error_msg="Invalid arguments"`, although it's already in the [docs](https://www.raspberrypi.org/documentation/raspbian/applications/vcgencmd.md) you referrenced earlier.

JamesH65, 2019-11-22:
Probably not made it to apt yet. rpi-update would get it I believe. Usual caveats apply when using rpi-update.

mohyddintash, 2019-11-22:
@JamesH65 hahahaha, so ture! Documentation is getting better with time, more than we could ask for :)
I think I'll have to risk an rpi-update after all.

giddyhup, 2019-11-22:
@mohyddintash I risked it about a week ago and can now turn my screen on and off again.

mohyddintash, 2019-11-23:
@giddyhup Did the upgrade break anything?

giddyhup, 2019-11-24:
@mohyddintash No, nothing broke but YMMV; I suggest making a backup at least of the boot partition.

mohyddintash, 2019-11-24:
@giddyhup glad to hear that! Sure, I always do make backups before such action. Thanks!

lubagov, 2019-12-09:
We have a lor Rpi, and this bug happens on several. Even on last version kernel. I make some table, on several configurations:

RPI firmware | SPI bootloader | kernel version | display_power working? |  
-- | -- | -- | -- | --
Dec 5 2019 14:42:20 | May 10 2019 19:40:36 | 4.19.86-v7l+ | not working |  
dfab159feffeba8de79ecb6dc973fab33d07004f | d2402c53cdeb0f072ff05d52987b1b6b6d474691 |   |   |  
Nov 19 2019 16:42:26 | Sep 10 2019 10:41:50 | 4.19.84-v7l+ | working |  
aeebba4c03968ede49097db077673eadc2888a22 | f626c772b15ba1b7e0532a8d50a761b3ccbdf3bb |   |   |  
Nov 19 2019 16:42:26 | May 10 2019 19:40:36 | 4.19.84-v7l+ | working | 2016, 2027 device
aeebba4c03968ede49097db077673eadc2888a22 | d2402c53cdeb0f072ff05d52987b1b6b6d474691 |   |   |  
Nov  8 2019 18:12:12 | May 10 2019 19:40:36 | 4.19.81-v7l+ | not working |  
e3ba08775a44334ecfda5e37fafad3fb07a3dc7e | d2402c53cdeb0f072ff05d52987b1b6b6d474691 |   |   |  
Oct 11 2019 18:39:26 | Sep 10 2019 10:41:50 | 4.19.79-v7l+ | working |  
d4a63855399ed5c2e6cd94e6dbcd850d881c856d | f626c772b15ba1b7e0532a8d50a761b3ccbdf3bb |   |   |  
Sep 12 2019 16:31:46 | May 10 2019 19:40:36 | 4.19.71-v7l+ | working |  
1a1172e606ae6e16d13a6da2c7834ee06f69c48f | d2402c53cdeb0f072ff05d52987b1b6b6d474691 |   |   |  

and i can't see logic. in version from 19 November it working, but from 5 December and 8 November not....

JamesH65, 2019-12-10:
@lubagov What specifically isn't working?

frankgould, 2019-12-20:
I never returned to this site to confirm this fix is working fine for both Buster and Arch Linux ARM. I will close this one out and if others need help, they can reference this. That is unless @JamesH65 prefers I reopen this issue.

closed 2019-12-20 by frankgould

lubagov, 2019-12-21:
display_power, it was fixed and working, in 4.19.84 and agein beoken in 4.19.86..

lubagov, 2019-12-21:
i have 120 Raspberry and i need to fix it in somewhere... found where it broken... becouse different devices have different kernels :-( by this reason i tru to update one to last version and check. And check also some other, with different to found minimal and fixed versions. And in version from 5 dec it is not working for me. But working in ksernel from 19 november and again display power not working in 8 november version.

lubagov, 2019-12-22:
I can confirm, it again fixed in 4.19.88 kernel, and it is working. But, when display connected to HDMI1 (not in HDMI0), now working, and can off/on display, but status return -1:
```
pi@lurity-prod-67:~ $ vcgencmd display_power 0
display_power=0
pi@lurity-prod-67:~ $ vcgencmd display_power 1
display_power=1
pi@lurity-prod-67:~ $ vcgencmd display_power
display_power=-1
```

JamesH65, 2019-12-23:
It's very odd you see it working on some kernels and not others - it's all been committed to the 4.19 tree so should just 'work' past a certain point, not sometimes work and sometimes not.

As for the error on HDMI1, will take a quick look.

helgeerbe, 2019-12-23:
I can confirm that vcgencmd display_power is not working on 3B+ to get the display status.
Toggeling the display on and off works!
I just upgraded to buster.

`pi@picframe:~ $ vcgencmd display_power 0
display_power=0
pi@picframe:~ $ vcgencmd display_power 1
display_power=1
pi@picframe:~ $ vcgencmd display_power
display_power=-1
pi@picframe:~ $ vcgencmd version
Dec 11 2019 15:14:22 
Copyright (c) 2012 Broadcom
version 44e6357b14365c3029d973dc32956fc5730dc334 (clean) (release) (start)
pi@picframe:~ $ uname -a
Linux picframe 4.19.88-v7+ #1284 SMP Wed Dec 11 13:46:41 GMT 2019 armv7l GNU/Linux`

JamesH65, 2019-12-23:
Looking at the code, specifying the display number in the vcgencmd might help.

So something like

vcgencmd display_power -1 <display id>

where display id is 2 for HDMI1 and 7 for HDMI2 will give 1 for on, -1 for off. Note, that should really be 0, I've found that bug and will fix it. There is a slight different issue when omitting the display id, where it was defaulting to the HDMI status even on HDMI2, which if nothing is connected to HDMI1 will return -1 (the previous bug!). Will fix that at same time.

helgeerbe, 2019-12-23:
Tested it od my 3B+

`pi@picframe:~ $ tvservice -l \n
1 attached device(s), display ID's are : 
Display Number 2, type HDMI 0
pi@picframe:~ $ vcgencmd display_power -1 2
display_power=-1
pi@picframe:~ $ tvservice -s
state 0xa [HDMI DMT (68) RGB full 16:10], 1920x1200 @ 60.00Hz, progressive
`
So specifying display id doesn't help

JamesH65, 2019-12-23:
Ah, it's possible that some of the change were not made on the pre-4 firmware branch. Will check - may or may not be made there. Nothing will hit rpi-update until the new year anyway.

helgeerbe, 2019-12-23:
@JamesH65 Thanks for the quick feedback. So I will wait. If you need someone for testing or other support let me know. Merry Christmas and a Happy New Year.

JamesH65, 2019-12-23:
It's clear from the code what the problem is, just need to fix, test and get it code reviewed, and then released.

JamesH65, 2020-01-02:
OK, this fix has been merged, just need to wait for the next release. Guys involved on holiday at the moment.

helgeerbe, 2020-01-07:
Fix is not complete for 3B+

Without parametes the correct display status is returned. But with parameters I still get -1 as result.

`pi@picframe:~ $ tvservice -l
1 attached device(s), display ID's are : 
Display Number 2, type HDMI 0
pi@picframe:~ $ vcgencmd display_power -1 2
display_power=-1
pi@picframe:~ $ vcgencmd display_power
display_power=1
pi@picframe:~ $ uname -a
Linux picframe 4.19.93-v7+ #1286 SMP Mon Jan 6 13:18:44 GMT 2020 armv7l GNU/Linux
pi@picframe:~ $ vcgencmd version
Jan  6 2020 13:05:25 
Copyright (c) 2012 Broadcom
version f3e39fed4763ce4efbffabb2d4787aaacddeda0f (clean) (release) (start)
`

tripplehelix, 2022-09-14:
I'm still having the original problem on an upto date Rpi4

`pi@kiosk | /home/pi
0:09:48 ▶ vcgencmd version
Aug  9 2022 13:44:40
Copyright (c) 2012 Broadcom
version 273b410636cf8854ca35af91fd738a3d5f8b39b6 (clean) (release) (start)

pi@kiosk | /home/pi
0:08:24 ▶ /usr/bin/vcgencmd display_power 0

display_power=1

pi@kiosk | /home/pi
0:08:29 ▶ /usr/bin/vcgencmd display_power 1

display_power=1

pi@kiosk | /home/pi
0:08:31 ▶ /usr/bin/vcgencmd display_power 0

display_power=1

pi@kiosk | /home/pi
0:08:34 ▶ /usr/bin/vcgencmd display_power

display_power=1`

popcornmix, 2022-09-15:
vcgencmd display_power is a control for the firmware display driver.
We not use the kernel driver by default (vc4-kms-v3d), so display_power should not be used.
See https://github.com/raspberrypi/firmware/issues/1740

bagong, 2022-10-27:
Is there another, vc4-kms-v3d compliant way to turn hdmi off and on?

[14 more replies]