raspberrypi/firmware #1226: Custom 3440x1440 EDID selecting wrong mode

JamesH65, 2019-08-13:
Customer has reported that their display, 3440x1440 is selecting a different mode. I've tried their EDID, and actually get 2560x1080p 59Hz once the V3D takes over.

JamesH65, 2019-08-13:
[edid_pi4_ultrawide.tar.gz](https://github.com/raspberrypi/firmware/files/3496872/edid_pi4_ultrawide.tar.gz)

JamesH65, 2019-08-13:
@6by9 FYI

JamesH65, 2019-08-13:
Original forum post https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=248037

JamesH65, 2019-08-13:
I think I could get this through the system with the following config entries, assuming you are using a provided edid file. Remove first two lines if not. Note, this was causing scrambling asserts in the firmware and my 4K monitor would not sync to the signal.

hdmi_edid_file:0=1
hdmi_edid_filename:0=edid_pi4_ultrawide.bin
hdmi_timings=3440 0 88 64 8 1440 0 4 5 32 0 0 0 60 0 533120000 0
hdmi_pixel_freq_limit=600000000
hdmi_group:0=2
hdmi_mode:0=87

6by9, 2019-08-13:
James: Add an option to arm_loader.c to stop the firmware adding the "video=HDMI-A-1:XxY@N" line to the command line and see whether KMS then picks the correct mode. IIRC There's already an ```if(kms_mode)``` clause to switch between adding the above vs adding the ```bcm2708_fb.XXX``` parameters, so want to add a further conditional in that first case to add neither.

The firmware never will select 3440x1440 by default as it doesn't fully parse the detailed timings, and instead finds the nearest mode from the standard tables. This can cause grief if KMS then doesn't find the mode in the list it parses from the EDID as it adds that as a mode based on the VESA GTF formula.

6by9, 2019-08-13:
Actually for a test I believe you should be able to add
```
video=HDMI-A-1:3440x1440@59
```
to the end of /boot/cmdline.txt to make KMS choose that mode instead of the firmware specified one (the last entry should take precedence).

JamesH65, 2019-08-14:
Out of office now, so not able to continue with this for a couple of weeks.

JamesH65, 2019-09-04:
I've got some beta firmware that may well fix this. Check the forum post linked above for details.