raspberrypi/firmware #1227: Video files not played/freezed when forcing 1080i50 - Raspberry Pi 4

triangle-stream, 2019-08-14:
Hi there! 

I'm facing an endless problem with my Raspberry Pi 4 (1GB) (I also have a Raspi 3 B+). 
Both of them are "forced" to work in 1080i50 because the right CEA mode 20 was not available under raspi-config (no interlaced mode was available). 
Both of the raspi are connected to a video switcher (working on 1080i50) via HDMI ports; both of them with the same OS version (Buster, last update).

The problem is that, on the Raspi 4, when i try to play a video file (.mp4, .flv, RTMP, 720p, 1080p, 1080i, everything) with omxplayer or ffplay, the video freezes on the first frame until i stop the process (ctrl+c). The only way to achieve a different result was playing a 1080i50 RTMP stream (wich i was able to see clean on other devices): in this case i have a video that plays at something like 1fps.

With the same setup, and same video files and players, the Raspi 3 B+ works really good without any problem. 

On te Raspi 4 i tried to: disable GL, disable 4K60 and any other advice i found on this forum and also on github etc. Nothing worked for me. 

At this point i tried to clone the SD card (same model, same size) from Raspi 3 B+ to Raspi4: everything worked good. But the video player in HDMI forced mode does not work. Same problems. 

Does anyone know this issue? I have no problems on Raspi 4 when i set a resolution from the raspi-config , and every video file is played properly, but i need that the raspi MUST be on 1080i50. Why the same thing, same os, same config, same everything, works differently on Raspi 4?

**To reproduce**
ffplay http://cdn.mainstreaming.tv/vodprogressive/TiRcOn3-h264_main_42_720_1280_2300000_aac_128000.mp4 

 omxplayer http://cdn.mainstreaming.tv/vodprogressive/TiRcOn3-h264_main_42_720_1280_2300000_aac_128000.mp4 

**Additional context**
MAYBE Something interferes with omxplayer output in non-fkms interlaced modes on pi4 (possibly incorrect vsync interrupts).
FKMS doesn't support interlaced modes.

matiaspl, 2019-09-29:
The lack of interlaced modes in the FKMS driver is a pain for all "broadcast" users of the Pi. I'd too be really happy to see them in the new driver.

6by9, 2019-09-30:
Sorry, interlaced support is something that is a fair way down the priority list.
The firmware supports it fine, but X in particular is currently misbehaving badly if you try rendering to it with DRM/KMS. 

On your Pi3, enable the full KMS driver (which does claim to support interlacing), and start X. It looks like it is missing every other line, or some similar error.

You can enable it on the Pi4 with FKMS by altering the line at https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/gpu/drm/vc4/vc4_firmware_kms.c#L1588 to read ```connector->interlace_allowed = 1;```` and rebuilding your kernel, but I was seeing odd effects with windows bouncing up and down as you dragged them around. These sorts of errors take a fair while to track down as it could be in any of a number of areas of the composition or rendering pipelines.
If you're not using X, then you may get reasonable results with that one line change. It's not one that we'll merge without having X running properly as having the desktop running properly is a higher priority than interlace support.

Disabling the vc4-fkms-v3d overlay in config.txt does work on the Pi4 and allow interlaced modes to be requested (eg ```hdmi_mode=20``` ``` hdmi_group=1``` in config.txt for 1080i50), however you will lose the GL drivers.

matiaspl, 2019-10-04:
I get the picture, thanks for the explanation. I'm aware of the workaround and my current workflow is switching between legacy and FKMS and restarting depending on the requested video mode. Not that bad, but not perfect either ;)

6by9, 2019-10-04:
Looking at the original report, I'll agree that something is upset in legacy mode when using omxplayer. Sorry I'd originally read it as a complaint that FKMS didn't support interlaced output, but it's more significant than that.

I've had a very quick look, and as you guessed we're not getting end of frame callbacks from the hardware. There's a 1.2s timeout on updates should we not get the correct callbacks, and that is why you should be getting jerky playback (I get a frame at roughly that rate).

I think it's an obvious reason - a timer is being set on the start of frame signal for near the end of the frame. With interlaced images I suspect we're getting a start at the beginning of each field (not frame), which will reset the timeout value, therefore the timer never expires.
It looks a simple one to solve, but if we're getting field interrupts instead of frame interrupts then it requires a touch more investigation to ensure that we're doing the right thing.

NB ffplay has no hardware acceleration, therefore ends up nearly maxing out the CPUs.
VLC in Raspbian has various optimisations to use the hardware acceleration. In legacy mode that works pretty well as all rendering is done to a single framebuffer that is constantly on the screen - there are no updates for double buffering or similar.

6by9, 2019-10-04:
It was an obvious one to fix too. Patch pushed for review internally, so I'd expect it to make rpi-update in the next week. It should be tagged against this issue when it gets pushed.

NB This does NOT fix supporting interlaced modes with FKMS, only omxplayer in legacy mode.

popcornmix, 2019-10-11:
Latest rpi-update firmware may have a fix for this.