raspberrypi/firmware #1229: After upgrade to the latest firmware, images from Pi Camera now purple in absolute darkness

max17048, 2019-08-18:
Referring to #1167

Used HW:
RpiZero-W /w 5MP no-name
Rpi3B /w 8MP No-IR
No additional IR-light used

Used SW:
Stretch, upgraded 2019-08-07. Before, everything workes fine. Maybe there is a gap of weeks since last upgrade.
PiCamera 1.13, used as realtime stream, capuring a picture every 10 seconds und create a timelapse video.

Both cameras are set to: --mode 0 --exposure night --awb auto --drc off

On both systems (No-IR & /w IR-filter), in absolutely darkness the images switch to purple (No-IR) respectively other colors on the camera /w IR-filter. On the No-IR, there is a continued disorder switching normal for seconds, on the camera /w IR-filter disorders happens for seconds with minutes of normal pictures.

On daylight there are no problems.

Image 8MP No-IR:
[image]

Image 5MP IR-filter:
[image]

6by9, 2019-08-19:
https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=245994
#1167

max17048, 2019-08-19:
> https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=245994
> #1167

Yes, I read this. What happens to cameras /w IR-filter?