raspberrypi/firmware #1230: No audio from HDMI

tiagoboldt, 2019-08-18:
**Describe the bug**
No hdmi audio after a recent upgrade. Setup is an LG tv into a sony dn1080. Rpi 4 with arch linux and kodi-rbp4 into dn1080 as well. Also, if the pi is on, the av receiver cannot use arc from tv, for example, when I'm using netflix from the TV. `sudo /opt/vc/bin/tvservice -o` makes ARC start to work immediately, but I have to reboot the pi to get the screen back `tvservice -p` does nothing.

My config is:
> dtparam=audio=on
hdmi_drive=2
gpu_mem=512
dtoverlay=pi3-disable-bt
dtoverlay=pi3-disable-wifi
dtoverlay=vc4-fkms-v3d
hdmi_ignore_cec_init=1
hdmi_force_edid_audio=-1
initramfs initramfs-linux.img followkernel

I've played with multiple configs, changing resolutions, changing kodi output config from alsa to hdmi and back. nothing provided results. 

**To reproduce**
Just boot with recent firmware. Currently using `linux-firmware-20190815.07b925b-1-any` and arch linux.

**Expected behaviour**
Audio is sent via hdmi to av receiver like it was about a month ago.

**Actual behaviour**
No audio via HDMI. 

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? 
> pi4
* Which OS and version (`cat /etc/rpi-issue`)? 
> arch linux
* Which firmware version (`vcgencmd version`)?
> Aug 15 2019 12:03:51
Copyright (c) 2012 Broadcom
version 9f8431fb7839c7f00f52b81f5822ddab2b31d0db (clean) (release) (start)

* Which kernel version (`uname -a`)?
already running the linux-raspberrypi4 kernel released these past days, but the issue started with the generic linux-raspberrypi kernel. 
> Linux rpi 4.19.66-1-ARCH #1 SMP PREEMPT Sat Aug 17 02:33:31 UTC 2019 armv7l GNU/Linux

**Logs**
The only thing weird in dmesg is this very threatning line:
> snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.

tiagoboldt, 2019-09-02:
I've been playing with way to much stuff and eventually *something* made it work again. Unfortunately, I cannot pinpoint what it was. Closing the issue though.

closed 2019-09-02 by tiagoboldt

Bufhimat, 2019-10-14:
Can you recall any of the steps you took, or if there is any documentation you followed?

giddyhup, 2019-10-21:
I encountered something related to this and solved it. Perhaps it helps. After the (dist) upgrade, my Pi had switched to (unconfigured) Pulse audio for output. I resolved it by selecteding my USB soundcard as output via `alsamixer`.