raspberrypi/firmware #1231: HDMI->DVI output on PI 3B not working

comdata, 2019-08-19:
**Describe the bug**
I have an EIZO DVI monitor that is

- WORKING with a PI ZERO W
- NOT WORKING with a PI 3 B

Raspbian images are in both cases up to date and unmodified.

**To reproduce**
Connect screen and power:

- PI ZERO W => see rainbow colors => OK
- PI 3B => screen goes dark => NO SIGNAL => NOK

**Expected behaviour**
Picture is being displayed.

**Actual behaviour**
No Signal

**System**
 System Information
------------------

Raspberry Pi 3 Model B Plus Rev 1.3
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage4

Linux display 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
Hardware	: BCM2835
Revision	: a020d3
Serial		: 00000000e48ac098
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Aug 15 2019 12:06:42 
Copyright (c) 2012 Broadcom
version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30361804 2835912  26244220  10% /
devtmpfs          315928       0    315928   0% /dev
tmpfs             447512       0    447512   0% /dev/shm
tmpfs             447512    6008    441504   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             447512       0    447512   0% /sys/fs/cgroup
/dev/mmcblk0p1    258096   41162    216934  16% /boot
tmpfs              89500       0     89500   0% /run/user/1000

Filename				Type		Size	Used	Priority
/var/swap                               file		102396	0	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20190717
[... 464 lines cut ...]
[    7.481647] brcmfmac: power management disabled
[    7.718576] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    7.718588] 8021q: adding VLAN 0 to HW filter on device eth0
[   13.171170] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   13.830238] Bluetooth: Core ver 2.22
[   13.830334] NET: Registered protocol family 31
[   13.830341] Bluetooth: HCI device and connection manager initialized
[   13.830363] Bluetooth: HCI socket layer initialized
[   13.830375] Bluetooth: L2CAP socket layer initialized
[   13.830415] Bluetooth: SCO socket layer initialized
[   13.851068] Bluetooth: HCI UART driver ver 2.3
[   13.851083] Bluetooth: HCI UART protocol H4 registered
[   13.851169] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   13.851408] Bluetooth: HCI UART protocol Broadcom registered
[   14.064609] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.064616] Bluetooth: BNEP filters: protocol multicast
[   14.064630] Bluetooth: BNEP socket layer initialized
[   14.185585] Bluetooth: RFCOMM TTY layer initialized
[   14.185611] Bluetooth: RFCOMM socket layer initialized
[   14.185633] Bluetooth: RFCOMM ver 1.11

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**

comdata, 2019-08-19:
EDID Parser output:
HDMI:EDID version 1.2, 0 extensions, screen size 34x27 cm
HDMI:EDID features - videodef 0x80 standby suspend active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor S/N descriptor tag 0xff
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 59-61 Hz, horizontal is 31-64 kHz, max pixel clock is 110 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is L557
HDMI:EDID found preferred DMT detail timing format: 1280x1024p @ 60 Hz (35)
HDMI:EDID established timing I/II bytes are A1 08 00
HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
HDMI:EDID standard timings block x 8: 0x8180 0101 0101 0101 0101 0101 0101 0101 
HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 0
HDMI:EDID filtering formats with pixel clock unlimited MHz or h. blanking unlimited
HDMI:EDID best score mode initialised to DMT (4) 640x480p @ 60 Hz with pixel clock 25 MHz (score 0)
HDMI:EDID best score mode is now DMT (4) 640x480p @ 60 Hz with pixel clock 25 MHz (score 36864)
HDMI:EDID best score mode is now DMT (9) 800x600p @ 60 Hz with pixel clock 40 MHz (score 57600)
HDMI:EDID best score mode is now DMT (16) 1024x768p @ 60 Hz with pixel clock 65 MHz (score 94370)
HDMI:EDID best score mode is now DMT (35) 1280x1024p @ 60 Hz with pixel clock 108 MHz (score 5260929)
HDMI0:EDID preferred mode remained as DMT (35) 1280x1024p @ 60 Hz with pixel clock 108 MHz
HDMI:EDID has only DVI support and no audio support
edidparser exited with code 0

6by9, 2019-08-20:
You have ```dtoverlay=vc4-kms-v3d``` to load the ARM side GL driver and DRM/KMS driver for display. GL performance with that driver on the Pi0 will be very low as the ARM CPUs will struggle on the shader compilation and similar. It also makes the output from tvservice/edidparser redundant or incorrect, as the ARM driver has taken over.
I'd actually expect performance to be better on the 3B, at least for GL.

Can you test without ```dtoverlay=vc4-kms-v3d``` to confirm it's not an interaction there? (unlikely as the 4 raspberries are done by the common firmware)
Does the 3B work on any other monitors?
Can you swap the SD card from the 0 into the 3B and see if that boots?

valtteric, 2019-08-21:
Noticed this yesterday with a CM3+ & ViewSonic VS14833 DVI display, few comments to reproduce for what it's worth:

Way 1:
1. Flash fresh 2019-07-10-raspbian-buster-lite
2. Boot ok, enable full KMS via raspi-config.
3. Reboot, stucks on the start messages. Can be accessed via SSH.

Way 2:
1. 2019-04-08-raspbian-stretch-lite
2. Enable full KMS via raspi-config, reboot OK.
3. apt-get update & upgrade
4. Reboot, stucks on the start messages. Can be accessed via SSH.

Didn't notice any errors in dmesg with a quick look, don't know where else to look for better help. Does not reproduce with other display HDMI -> HDMI.

comdata, 2019-08-21:
Quick reply:

1. removed the line for the dtoverlay=vc4-kms-v3d, it was only active in the pi4 section => not working.
2. tried other monitors, SAMSUNG => not working.
3. swapped SD card => not working.

The PIs are booting but simply show no picture, I can access them via SSH.

Additionally a PI 2B is working with the same image and monitor.

Also tried a PI4B with that monitor and it is also NOT working.

comdata, 2019-08-21:
PI 1 B is also working.

So 

WORKING: 1B, 2B, Zero W 
NOT WORKING: 3B, 4B

polaco1782, 2019-08-22:
> PI 1 B is also working.
> 
> So
> 
> WORKING: 1B, 2B, Zero W
> NOT WORKING: 3B, 4B

I have the same issue as you. Seems to be a bootloader issue, mine works fine with an older version, when I upgraded it from the last release It broke.

polaco1782, 2019-08-22:
I replaced the bootloader with a previous version and It worked.
download this version https://codeload.github.com/RPi-Distro/firmware/zip/1.20180619

Uncompress and replace bootloader files from boot/ directory on your SD card

For RPI3+ only. There is no support for the 4+ version in this bootloader.

tallbl0nde, 2019-08-31:
I'm also experiencing a similar issue with HDMI (no adapter though) on my RPi 3B (using the latest firmware). I've described the problem on the forums: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=250124

If I replace my bootloader with the files linked to by @polaco1782 I get HDMI working as expected without having to mess around with unplugging and replugging cables etc.

suromark, 2019-08-31:
I can corroborate that; it happens on my Raspberry Pi 3b+ also. I do have an older microSD card that works, and an up-to-date clone of that on an USB-SSD. Before the update, video output would default to HDMI with rainbow-on-boot. After the update, HDMI is not enforcable and video output is only at Composite Out with 720x480. If  I use the microSD card to boot then redirect to the SSD, HDMI works.

dimov-cz, 2019-09-10:
> I can corroborate that; it happens on my Raspberry Pi 3b+ also. I do have an older microSD card that works, and an up-to-date clone of that on an USB-SSD. Before the update, video output would default to HDMI with rainbow-on-boot. After the update, HDMI is not enforcable and video output is only at Composite Out with 720x480. If I use the microSD card to boot then redirect to the SSD, HDMI works.
> 
> Edited to add: Copying the files bootcode.bin, start*.elf and fixup*.dat from firmware 1.20180619 restored all HDMI functionality.

Confirmed, mirroring boot dir to SD card completely solves this. Thank you!!

suromark, 2019-09-11:
Another update: After installing the previous firmware/GPU files, HDMI worked again, **but** camera was no longer available. Reverting to the newer firmware restored camera, but broke HDMI.
I ended up doing a clean install of the Buster image onto the SSD, then copied over my user dir and reinstalled my extra packages. Now both HDMI and Camera are operational again.