raspberrypi/firmware #1232: Raspberry Pi 4 no HDMI output after plymouth boot screen

Septatrix, 2019-08-19:
**Describe the bug**
When specifying hdmi_group=2 and hdmi_mode=27 the raspberry pi outputs a signal of the correct resolution during boot but stops to do so after showing the plymouth boot screen. Forcing an output using `tvservice -e "DMT 27 HDMI"` however fixes this.

**To reproduce**
1. Burn a fresh image to an sd card
2. Add the following to config.txt
    ```
    hdmi_force_hotplug=1
    hdmi_group=2
    hdmi_mode=27
    hdmi_drive=2
    ```
3. Boot the raspberry pi
4. (Optionally update and retry)
5. Login via ssh and execute `tvservice -e "DMT 27 HDMI"`

**Expected behaviour**
Raspberry Pi boots and outputs a signal.

**Actual behaviour**
Raspberry Pi boots, outputs a signal (of the correct resolution and refresh rate) until it shows the plymouth boot screen. After that no signal is output.

**System**

* Raspberry Pi Model: RPi4B
* Display: Beetronics 7TS4
* Which OS and version (`cat /etc/rpi-issue`)?
    ```
    Raspberry Pi reference 2019-07-10
    Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage4
    ```
* Which firmware version (`vcgencmd version`)?
    ```
    Aug 15 2019 12:03:51 
    Copyright (c) 2012 Broadcom
    version 9f8431fb7839c7f00f52b81f5822ddab2b31d0db (clean) (release) (start)
    ```
* Which kernel version (`uname -a`)?
    ```
    Linux raspberrypi 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux
    ```

**Additional context**
EDID data from the display: 
[edid.zip](https://github.com/raspberrypi/firmware/files/3517331/edid.zip)

Septatrix, 2019-08-20:
After commenting out `dtoverlay=vc4-fkms-v3d` in `config.txt` I could comment out `hdmi_{group,mode}` and the correct resolution was chosen automatically.

This resolves my problem but there still may be a bug in the graphics driver - therefore I will leave this open until an official response thinks this requires no further investigation.

devInterests, 2019-10-06:
Any news on that issue?
I have the issue after I have updated my pi.
After plymouth boot screen, no screen detected.

4k 60HZ is on, cable is on HDMI0.

`Linux raspberrypi 4.19.75-v8+ #1271 SMP PREEMPT Mon Sep 30 13:53:45 BST 2019 aarch64 GNU/Linux`

```
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage4
```

`state 0xa [HDMI CEA (97) RGB lim 16:9], 3840x2160 @ 60.00Hz, progressive`

HDMI1 is working, but i would like to have 4k 60HZ.

Tried different config options in config.txt, no luck.

velkovb, 2019-10-07:
I did a forum thread as well but not luck :( [RPI 4 Getting No signal after boot screen at 4kp60](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=251411&p=1544782#p1544782)
I have tested on another TV with HDR port and it work there, on non HDR ports the same behaviour.

ephestione, 2019-10-25:
#MeToo
Screen was being turned off right after the lines "found device /dev/serial1" and "found device /dev/disk/by-partuuid/<bootpuuid>" were shown.
Commented out the same `dtoverlay=vc4-fkms-v3d` line in config.txt and the screen, which is a Samsung 1320x760 or something TV, outputs this status:
`state 0xa [HDMI DMT (4) RGB full 4:3], 640x480 @ 60.00Hz, progressive`
by commenting out also the hdmi mode lines.