raspberrypi/firmware #1235: RPi4B: eMMC emulated SD card doesn't work / different CMD sequences found

uugear, 2019-08-23:
**Describe the bug**
We have a eMMC->micro SD adapter, which is a PCB with eMMC (16GB) chip soldered and emulate the micro SD interface, so you can insert it into Raspberry Pi and use it as a 16GB SD card, with better read/write speed. It works perfectly in Raspberry Pi Zero, Zero W, 3B and 3B+, but it doesn't work in Raspberry Pi 4B.

**To reproduce**
The eMMC->micro SD adapter does these wirings to emulate an SD card:
```
eMMC DAT2 (A5) -> microSD DAT2
eMMC DAT3 (B2) -> microSD DAT3
eMMC CMD  (M5) -> microSD CMD
eMMC VDD       -> microSD VDD
eMMC CLK  (M6) -> microSD CLK
eMMC VSS       -> microSD VSS
eMMC DAT0 (A3) -> microSD DAT0
eMMC DAT1 (A4) -> microSD DAT1
```
- Insert this emulated SD card into a card reader and flash the latest Raspbian Buster (2019-07-10) into it.
- Insert this emulated SD card into Raspberry Pi 3B and then have power connected, it boots the system without any problem.
- Insert this emulated SD card into Raspberry Pi 4B and then have power connected, it doesn't boot and the green LED just blinks slowly like it doesn't recognise the SD card at all.

**Expected behaviour**
This eMMC emulated SD card work as well in RPi4.

**Actual behaviour**
This eMMC emulated SD card is not recognised by RPi4.

**System**
* Which model of Raspberry Pi?
Raspberry Pi 4B (1GB)
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage4
* Which firmware version (`vcgencmd version`)?
Jun 20 2019 16:04:31 
Copyright (c) 2012 Broadcom
version 407b1da8fa3d1a7108cb1d250f5064a3420d2b7d (clean) (release) (start)
* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.50-v7l+ #895 SMP Thu Jun 20 16:03:42 BST 2019 armv7l GNU/Linux

**Additional context**
I used a SparkFun microSD Sniffer and a logic analyzer to observe the communication between Raspberry Pi and the emulated SD card. In order to make comparison, I tested these combinations:
- Raspberry Pi 3B with Kingston 16GB micro SD card (boot OK)
- Raspberry Pi 3B with eMMC emulated SD card (boot OK)
- Raspberry Pi 4B with Kingston 16GB micro SD card (boot OK)
- Raspberry Pi 4B with eMMC emulated SD card (not recognised)

I monitored the CMD and CLK lines to figure out the actual command sequence in each case. Below I list the command sequences I got:

**Raspberry Pi 3B with Kingston 16GB micro SD card (boot OK)**
```
CMD0
CMD8
CMD8-RESP
CMD55
CMD55-RESP
ACMD41
R3-RESP
(repeat the 4 steps above 3 times)
CMD2
CMD2-RESP
CMD3
CMD3-RESP
...
```
**Raspberry Pi 3B with eMMC emulated SD card (boot OK)**
```
CMD0
CMD8
CMD0
CMD55
CMD1
R3-RESP
(repeat the 2 steps above 2 times)
CMD2
CMD2-RESP
CMD3
CMD3-RESP
...
```
**Raspberry Pi 4B with Kingston 16GB micro SD card (boot OK)**
```
CMD0
CMD8
CMD8-RESP
CMD55
CMD55-RESP
ACMD41
R3-RESP
(repeat the 4 steps above 29 times)
(the clock signal lasts ~50ms without any command)
CMD55
CMD55-RESP
ACMD41
R3-RESP
CMD2
CMD2-RESP
CMD3
CMD3-RESP
...
```
**Raspberry Pi 4B with eMMC emulated SD card (not recognised)**
```
CMD0
CMD8
CMD0
CMD55
CMD0
CMD1
R3-RESP
(repeat the 2 steps above 29 times)
CMD2
CMD0 (drop the process and start over again)
CMD8
…
```
By comparing the command sequences, I can see some differences between implementations in RPi3 and RPi4.

The first thing is that RPi4 repeats the ACMD41 or CMD1 command so many times, which doesn't seem normal. Although I can't tell if it causes the emulated SD card not recognised, but I guess something could be optimised here?

I also noticed that, when using the eMMC emulated SD card, RPi4 inserts a CMD0 before the CMD1, and RPi3 doesn't do that. I don't know if this behavior is relevant to this issue.

Obviously the command sequence for real micro SD card and that for eMMC emulated SD card are different (in both RPi3 and RPi4). For example the ACMD41 command is used for real SD card, while the CMD1 command is used for eMMC emulated SD card. Maybe those older models of Raspberry Pi are supporting this emulated SD card with some quirks, but it would be nice to let RPi4 has the same compatibility.

timg236, 2019-08-23:
Sorry EMMC hasn’t been tested at all on Pi4 yet. It’s fairly low priority but it’s probably quite simple and could be a “Friday afternoon task”. If you can post details of common hardware configurations then we can order some and give it a go.

uugear, 2019-08-23:
> Sorry EMMC hasn’t been tested at all on Pi4 yet. It’s fairly low priority but it’s probably quite simple and could be a “Friday afternoon task”. If you can post details of common hardware configurations then we can order some and give it a go.

Thank you for the quick reply, and I am glad to know it is not difficult to solve. I can send you two pieces of this eMMC->microSD adapter for testing. Please kindly sent me an email (to shawn [at] uugear.com), and let me know your shipping address (with telephone number please). I will sent them to you via UPS. Thank you.

ODwyerPW, 2019-11-01:
Did you all pursue this? It would be neat to find something faster than a microSD Card..

uugear, 2019-11-02:
> Did you all pursue this? It would be neat to find something faster than a microSD Card.

The latest Raspbian Buster (2019-09-26) includes new firmware and can support eMMC emulated SD card on Pi4. However you will need to upgrade the bootloader beforehand. You can find beta version of bootloader [here](https://github.com/raspberrypi/rpi-eeprom/tree/master/firmware/beta). The version I tested is "pieeprom-2019-10-16.bin".

If we can wait long enough, eventually the newly manufactured Pi4 might come with the latest bootloader. However the lead time could be very, very long.