raspberrypi/firmware #1236: Kernel load issue with start.elf start with 1.20190620+1

pvizeli, 2019-08-27:
**Describe the bug**

I'm the main developer of HassOS / A distribution with run a system on docker, called Hass.io

We use GPT in Hybrid Mode and load u-boot they load the Kernel. That work without any issue until the Firmware of 1.20190620+1 / exactly the `start.elf` or `start4.elf`. The Kernel will not load correctly and different things will not work. On a RPi3+, the HDMI have yellow lines and on a RPi4 it sees only 1GB memory. I think it messes up the FDT. I can also load the kernel directly with config.txt without u-boot and it ends up in the same issue.

Not sure if that the same issue as #1097. For the RPi4 1GB issues: https://github.com/home-assistant/hassos/issues/439 / https://github.com/home-assistant/hassos/issues/437

Without correct loaded Kernel/FDT, I can't release a new version. With Firmware before 1.20190620+1 everything works like a charm but there is no RPi4 support.

**To reproduce**
Download our Image (hassos-rpi3_3.3.img.gz) from https://github.com/home-assistant/hassos/releases/tag/3.3 and install it on a RPI3+, after boot you can see the issue or the Image for RPi4 on a 2GB model and you see only 1GB memory. Other unstable issues follow after a while.

**Expected behaviour**
The `start.elf` should work like before 1.20190620+1

**System**
HassOS dev-tree with all latest (github/raspberrypi/linux) / (github/raspberrypi/firmware)

6by9, 2019-08-27:
Yellow lines on HDMI - https://github.com/raspberrypi/linux/issues/3139?
U-boot is modifying the device tree that is passed to the kernel - sorry, not our problem. Easiest to disable bcm2708_fb.

pvizeli, 2019-08-27:
@6by9 thanks for your response. I have the same issue also if I load the kernel directly with `config.txt` from your first stage loader.

pvizeli, 2019-08-27:
I need future tests, but it looks like also the Memory issue with RPi4 going to address by FB/GPU. I did so many tests and different variation last three days that I lost the focus. I come back after next test series 👍 thanks @6by9 for the progress.

pvizeli, 2019-08-28:
I can confirm, all issues are fixed. Thanks for the fast progress.

closed 2019-08-28 by pvizeli