raspberrypi/firmware #1237: ffmpeg h264_omx: bitrate limits not respected on 60fps input

tmm1, 2019-08-29:
**Describe the bug**
Using ffmpeg's h264_omx encoder with 30fps input produces expected output bitrate within limits specified. But when feeding in 60fps input, the output bitrate climbs continuously up to 28mbps

**To reproduce**

>ffmpeg -i 30fps.mpg -c:v h264_omx -b:v 3000k -maxrate:v 3200k -bufsize 6000k -f mpegts -y /dev/null
>fps= 39 time=00:00:20.70 bitrate=27296.6kbits/s

>ffmpeg -i 60fps.mpg -c:v h264_omx -b:v 3000k -maxrate:v 3200k -bufsize 6000k -f mpegts -y /dev/null
>fps= 36 time=00:00:39.70 bitrate=3294.3kbits/s

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi4B

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2

* Which firmware version (`vcgencmd version`)?
Aug 22 2019 13:55:50
Copyright (c) 2012 Broadcom
version 5bcbe61d88681477f5415cf60ae211bcdcd416d3 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux tmm1-pi4 4.19.67-v7l+ #1254 SMP Fri Aug 23 15:17:21 BST 2019 armv7l GNU/Linux

**Additional context**
The h264_v4l2m2m encoder in ffmpeg doesn't have this issue.

I'm trying to figure out if the bug is in ffmpeg's setup of the omx ports, or in the omx implementation itself.

Maybe some sort of overflow issue in the Q16 xFramerate?

tmm1, 2019-08-29:
Interesting: if I omit the `xFramerate` altogether (leaving it as zero which the spec says means "unknown, variable, or is not needed"), then the 60fps encoded bitrate maxes out at 2x of what I requested (6000kbps output on -b:v 3000k) instead of climbing all the way to 28mbps;

peel70, 2019-08-29:
@tmm1 sorry to bother you,
but have you checked if ffmpeg is using GPU for conversion?
I've performance problems with Yocto/GStreamer/Qt Mediaplayer
thank you for your answer

tmm1, 2019-08-29:
> Maybe some sort of overflow issue in the Q16 xFramerate?

Yep that was it. This fixes the issue:

```diff
diff --git a/libavcodec/omx.c b/libavcodec/omx.c
index 837f5df666..c745860242 100644
--- a/libavcodec/omx.c
+++ b/libavcodec/omx.c
@@ -473,9 +473,9 @@ static av_cold int omx_component_init(AVCodecContext *avctx, const char *role)
     in_port_params.format.video.nFrameWidth  = avctx->width;
     in_port_params.format.video.nFrameHeight = avctx->height;
     if (avctx->framerate.den > 0 && avctx->framerate.num > 0)
-        in_port_params.format.video.xFramerate = (1 << 16) * avctx->framerate.num / avctx->framerate.den;
+        in_port_params.format.video.xFramerate = (int64_t)(1 << 16) * avctx->framerate.num / avctx->framerate.den;
     else
-        in_port_params.format.video.xFramerate = (1 << 16) * avctx->time_base.den / avctx->time_base.num;
+        in_port_params.format.video.xFramerate = (int64_t)(1 << 16) * avctx->time_base.den / avctx->time_base.num;

     err = OMX_SetParameter(s->handle, OMX_IndexParamPortDefinition, &in_port_params);
     CHECK(err);
```

tmm1, 2019-08-30:
Interesting: if I pass in a xFramerate of 119/120, the encoder crashes and I have to reboot the device to be able to use it again. Again, h264_v4l2m2m has no problem with the same values. (Some deinterlacing filters in ffmpeg double the framerate).

> Interesting: if I omit the `xFramerate` altogether (leaving it as zero which the spec says means "unknown, variable, or is not needed"), then the 60fps encoded bitrate maxes out at 2x of what I requested (6000kbps output on -b:v 3000k) instead of climbing all the way to 28mbps.

This seems to vary by content, but in general higher bitrates are produced when xFramerate is missing. I'm not sure why this is the case..

I also don't understand why bitrates are more consistent with h264_v4l2m2m when both encoders re-use the same vchiq backend.

peel70, 2019-08-30:
@tmm1 I've rebuilt yocto image with your patch for gstreamer-1.0-omx but the problem seem not fixed I'm able to play in GPU this pipeline

`gst-launch-1.0 -e -vvv filesrc location=big_buck_bunny_720p_h264.mov ! qtdemux ! queue ! h264parse ! v4l2h264dec capture-io-mode=4 ! kmssink`

but I'm not able to play mp4 
`GST_DEBUG="omx:2" gst-launch-1.0 -e -vvv filesrc location=bbb_sunflower_1080p_30fps_normal.mp4 ! qtdemux ! h264parse ! omxh264dec ! videoconvert ! kmssink`

give this error:
`0:00:00.332141903   634  0x1430d20 ERROR                    omx gstomx.c:1892:gst_omx_component_set_parameter:<omxh264dec-omxh264dec0> Set video_decode parameter at index 0x02000001: Bad parameter (0x80001005)
0:00:00.332392237   634  0x1430d20 ERROR                    omx gstomx.c:2072:gst_omx_port_update_port_definition:<omxh264dec-omxh264dec0> Updated video_decode port 130 definition: Bad parameter (0x80001005)`

I know that this is not the correct issue to write to but maybe you can help me?

tmm1, 2019-08-30:
@peel70 My patch is for ffmpeg and you are using gstreamer. My patch is for video ENCODE, but you are using video DECODE.

I cannot help you here. I suggest you start a new issue and fill out all the details requested in the issue form describing what problem you are facing.

tmm1, 2019-08-30:
This was a bug in ffmpeg. I have submitted a patch for review which I plan to merge in the next few days. I will also backport it to the 4.1 tree for the next point release.

closed 2019-08-30 by tmm1