raspberrypi/firmware #1238: ffmpeg h264_v4l2m2m decode performance on RPi4

tmm1, 2019-08-30:
I'm curious what kind of decoding performance can be expected from the v4l2 DECODE component. In my tests with ffmpeg on the RPi4, it seems doing software decoding on the CPU is much more performant than trying to use the v4l2 device.

However, this may be an issue with the ffmpeg decoder implementation. It appears to throw a lot of errors about timestamps, and there is atleast one known bug where packets from the h264 bitstream are randomly dropped when output buffers fill up (https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/multimedia/ffmpeg/patches/v4l2/ffmpeg-999-z-fixes.patch).

On my device, decoding a 1920x1080 progressive h264 video stream with h264_v4l2m2m maxes out at 57fps, whereas if I use ffmpeg's software h264 decoder I get 108fps. With 1280x720 progressive h264, the hardware decoder produces 130fps whereas the software decoder goes up to 243fps.

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi4B

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2

* Which firmware version (`vcgencmd version`)?
Aug 22 2019 13:55:50
Copyright (c) 2012 Broadcom
version 5bcbe61d88681477f5415cf60ae211bcdcd416d3 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux tmm1-pi4 4.19.67-v7l+ #1254 SMP Fri Aug 23 15:17:21 BST 2019 armv7l GNU/Linux

6by9, 2019-08-30:
This is not an issue or bug to be fixed.
Please take discussion threads onto the forums - https://www.raspberrypi.org/forums/index.php ("Graphics, sound and multimedia" would be the most appropriate place).

6by9, 2019-08-30:
Rereading your numbers it sounds like you are getting pretty much the performance expected from the hardware. It was originally designed for a maximum of 1080p30, but will now cope with most 1080p60 streams. There is currently a format conversion on the output, so that will reduce performance slightly, mainly due to the memory bandwidth hit.

The A72 is a pretty powerful CPU (or 4), so it may be able to exceed the performance of the hardware. It'll burn a lot more power in doing so.

tmm1, 2019-08-30:
Thanks! I will move to the forum for future topics.

closed 2019-08-30 by tmm1