raspberrypi/firmware #1240: RPI4B: Random bootup fail.

zuoyeben, 2019-08-31:
My RPI4 board random startup failure probably due to the instability of DRAM. I want to test the stability of DRAM with u-boot's memtest. It's not a good idea to run U-boot in DRAM and test DRAM, so I wonder if U-boot can be run in SRAM to test DRAM, but I don't know the location and size of SRAM. 

Can anyone know this information?

Thanks,
H.K

ghollingworth, 2019-09-02:
There is no SRAM to use...   Unless you confine you application to the cache and make sure it doesn't write back.

closed 2019-09-02 by ghollingworth

zuoyeben, 2019-09-02:
> There is no SRAM to use... Unless you confine you application to the cache and make sure it doesn't write back.
Did you mean L2 can't be used as SRAM?
There is a config "disable_l2cache" to disable CPU access GPU's L2, I  thought maybe i can use this area.

timg236, 2019-09-02:
The firmware is already running at that stage and will be using the VPU L2 cache.