raspberrypi/firmware #1241: PWR-LED switched off at system halt

Melanie-BB, 2019-08-31:
After shutdown of the Raspberry Pi, the status of the red PWR-LED actually is as follows:

- When the system is powered off (`systemctl poweroff`), the PWR-LED remains on.
- When the system is halted (`systemctl halt`), the PWR-LED is switched off.

In my opinion the red PWR-LED should remain on in both cases (poweroff and halt) to indicate that the Pi is still connected to a power source.

My system configuration:

- Raspberry Pi 1 B+
- Firmware 2019-06-26
- Kernel 5.2.5

burtyb, 2019-08-31:
If you want that behaviour you can set [POWER_OFF_ON_HALT=0](https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md).

Melanie-BB, 2019-09-02:
> If you want that behaviour you can set POWER_OFF_ON_HALT=0.

As described in [1], this setting is only for the Raspberry Pi 4 EEPROM, so not applicable for the previous models (Pi 0/1/2/3).

[1] [https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md](https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md)

timg236, 2019-09-02:
IIRC the Linux kernel is turning off the LED as part of the reboot/halt sequence

Melanie-BB, 2019-09-08:
> IIRC the Linux kernel is turning off the LED as part of the reboot/halt sequence

Are the 10 flashes of the ACT LED after poweroff also driven by the kernel?
Is the LED behaviour at shutdown (halt/poweroff) configurable?

pelwell, 2019-09-08:
The 10 flashes are driven by the firmware after a reset. It is not configurable.