raspberrypi/firmware #1242: Pi 2 B not filtered correctly in config.txt

juergh, 2019-09-03:
I have a Pi 2 Model B and try to load a custom kernel using the following config.txt but the kernel is not picked up:

    pi@raspberrypi:~$ cat /boot/config.txt
    dtparam=audio=on
    [pi2]
    kernel=foo.img
    [all]

If I change the filter to [pi3] it works. Seems like the Pi 2 shows up in the wrong section.

    pi@raspberrypi:~$ cat /proc/device-tree/model 
    Raspberry Pi 2 Model B Rev 1.2

    pi@raspberrypi:~$ vcgencmd version
    Aug 30 2019 17:57:30 
    Copyright (c) 2012 Broadcom
    version 91518454866504fb0ff9164391cb66b83f05ba07 (clean) (release) (start)

pelwell, 2019-09-03:
[ Resisting the urge to insert the obvious Shakespearian quote... ]

My theory is that you have one of the newer 2Bs with a BCM2837 processor (the same as the Pi 3). It used to be that 2Bs (and only 2Bs) all used BCM2836s, so although the documentation says:
```
[pi2] | Model 2B (BCM2836- or BCM2837-based)
```
the code assumes that it's only a Pi 2 if the processor is a BCM2836.

To me this looks like a straightforward, easy-to-fix bug, but @popcornmix may have a different opinion.

popcornmix, 2019-09-03:
Moving from check against processor type to board type should be fine.

pelwell, 2019-09-03:
Expect an internal Merge Request.

pelwell, 2019-09-04:
The latest firmware, available via rpi-update, includes a likely fix for this issue.

juergh, 2019-09-04:
Works. Thanks for the quick turn-around!

closed 2019-09-04 by pelwell