raspberrypi/firmware #1243: HDMI 1 on Pi 4 doesn't have sound

himuura, 2019-09-04:
**Describe the bug**
HDMI 1 doesn't provide any sound whatsoever on Pi 4

**To reproduce**
Just plug a micro HDMI cable to HDMI 1 (and not connecting anything to HMDI 0)

**Expected behaviour**
When using a standalone HDMI 1 connection there should be sound coming to the TV via HDMI.

**Actual behaviour**
Video is OK but i have no sound coming from HDMI 1. HDMI 0 works fine.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi 4 4Gb
* Which OS and version (`cat /etc/rpi-issue`)?
Linux r4spberry 4.19.69-v7l+ #1256 SMP Fri Aug 30 16:37:01 BST 2019 armv7l GNU/Linux (Raspbian Buster latest)
* Which firmware version (`vcgencmd version`)?
Latest to date.
Static hostname: r4spberry
Icon name: computer
Machine ID: 8bbbe84f0f9144239cec2269adaafcad
Boot ID: 180ebb732a7048edade61ecbe8be43ec
Operating System: Raspbian GNU/Linux 10 (buster)
Kernel: Linux 4.19.69-v7l+
Architecture: arm

* Which kernel version (`uname -a`)?
Linux 4.19.69-v7l+
**Logs**
Dont have any.

**Additional context**
Already tried hdmi_preferred=1 and hdmi_force_hotplug=1 without success. Also tried switching from HDMI to analog and back to HDMI again witouth any results.

6by9, 2019-09-04:
Not a firmware bug.

The lxplug-volume plugin calls ```amixer -q cset numid=3 2``` to route the one audio device to one of a number of outputs. 2 is used for HDMI, but that is only ever HDMI0. It needs to call ```amixer -q cset numid=3 3``` for HDMI1. (1 is used for analog).

The tricky bit is that it doesn't know whether HDMI0 or HDMI1 is in use when using a single display.
@spl237 This is your domain really. I don't know if there is a sensible solution here for detecting the connections. One option would be to create the additional displays but advertise HDMI0 as disconnected. (Hotplug is something that is on my list, but a moderate way down it). There aren't any valid combinations of displays currently where you can use HDMI1 as well as DPI or DSI, it has to be HDMI0.

pelwell, 2019-09-04:
Could we not add a "first active HDMI" option?

himuura, 2019-09-04:
> Not a firmware bug.
>
> The lxplug-volume plugin calls `amixer -q cset numid=3 2` to route the one audio device to one of a number of outputs. 2 is used for HDMI, but that is only ever HDMI0. It needs [...]

For a few minutes i thought my Pi 4 had a defect :P thank you for your reply mate, it's all i wanted to "hear" (read). Perhaps a fix will be available in around 6 months or so?

6by9, 2019-09-04:
@himuura Run ```amixer -q cset numid=3 3``` and you'll get audio through HDMI1.

@pelwell It's a disjoin between the audio driver and video. There's also two "versions" of the audio driver - one that has one device and switches between the outputs, and one that presents all endpoints as different ALSA devices. Add ```snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_compat_alsa=0``` to get the latter behaviour.
Discussions at https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=249204#p1526690 whilst you were on holiday.
Even with the independent ALSA devices, they are always all presented to the user even if they are disconnected.
I found a discussion the other day over the KMS driver and how it should be presenting HDMI audio. The input from the Intel and AMD folk was that there was a back door from KMS into the ALSA side to add and remove the devices dynamically as things are hotplugged. Doing that with FKMS is going to be a right pain.

pelwell, 2019-09-04:
> Doing that with FKMS is going to be a right pain.

[ Backs away, hastily... ]

himuura, 2019-09-04:
> @himuura Run `amixer -q cset numid=3 3` and you'll get audio through HDMI1.
>
> @pelwell It's a disjoin between the audio driver and video. There's also two "versions" of the audio driver - one that [...]

i have a dumb question, does this alter some sort of config file or is it just a session value? I mean, after a reboot/shutdown, will it persist or does it need to be set again? i used to work with alsa mixer a couple of years back, don't quite recall how this goes (STT-TTS speech recog work). Thank you both for helping me out, mates :)

spl237, 2019-09-04:
> @spl237 This is your domain really. I don't know if there is a sensible solution here for detecting the connections. One option would be to create the additional displays but advertise HDMI0 as [...]

If I call xrandr --listmonitors, I can see the names of the currently connected HDMI devices. I could parse that in the case where there is a single monitor to determine whether it is HDMI-1 or HDMI-2, which are the display names. If I can assume that HDMI-1 is always HDMI0 (amixer cset numid=3 2) and HDMI-2 is always HDMI1 (amixer cset numid=3 3), then it's a trivial fix. But if that assignment isn't consistent, then I'm not sure what I can do....

spl237, 2019-09-04:
Just checked - no, with a single monitor on port HDMI1, xrandr still calls that HDMI-1, which is wrong (ought to be HDMI-2).

So unless someone can suggest some way I can tell which monitor I have if there is only one, this doesn't seem to be something I can fix.

JamesH65, 2019-09-04:
You could call 'tvservice -l' to determine what is attached where.

6by9, 2019-09-04:
Let's avoid tvservice if we can as then things have a chance of working with full KMS too when it happens.

I think the more sensible thing in the mid-term is to get hotplug working (unless there are going to be shouts of "it can't because of <reason>").
- If DPI or DSI displays are found /configured then that replaces HDMI1. You'll get HDMI-1 and DSI-1.
- Otherwise HDMI0 & HDMI1 are both initialised by the firmware and created by FKMS (HDMI-1 and HDMI-2 to xrandr). They then need to abide by hotplug and report back their connected status correctly.
- (Enabling SDTV is the third mode where we only get the VEC due to clocking requirements. I can't recall what the display will be called, but it doesn't have an associated audio path).

I'll create a new issue under Linux to propose and track that.

Separate issue/question, but it'd be nice if we could switch to using the alternate mode for snd_bcm2835 which exposes the outputs as individual devices. The drop down then wants to redirect ```default``` rather than posting a magic cset command to the one audio device.

6by9, 2019-09-04:
> i have a dumb question, does this alter some sort of config file or is it just a session value? I mean, after a reboot/shutdown, will it persist or does it need to be set again? i used to work with alsa mixer a couple of years back, don't quite recall how this goes (STT-TTS speech recog work). 

I don't recall either, and the volume plugin may store it independently of the system default.
@spl237 may be able to answer the latter, but it's probably simplest to just test it for yourself.

spl237, 2019-09-04:
> I don't recall either, and the volume plugin may store it independently of the system default.
> @spl237 may be able to answer the latter, but it's probably simplest to just test it for yourself.

ALSA device number is stored in ~/.asoundrc. The specific output on the internal Pi audio device is persistent over a reboot - it's written to some NV storage within the device. So if you have an external device, it is set as the default in .asoundrc; if you are using an output on the internal device, the internal device itself is set as the default in .asoundrc, and the Pi itself remembers which of the outputs on that device is in use.

JamesH65, 2019-09-17:
@6by9 Any progress on this? Is there a stopgap we should put in to get people around the problem until a final solution is available?

6by9, 2019-09-17:
I have a couple of hacks that mean that FKMS always gets told about both HDMI ports, and hot plugs between them. I believe that means that xrandr will report predictable output names.
It's not hooked up to know about DPI or DSI displays though, so not in a mergable state.

I think we want to be driving people towards the independent ALSA devices option rather than the magic cset thing. I'd need to check again what the lxplugin does at the moment with that.

JamesH65, 2019-09-17:
So, multiple ALSA devices vs one device with sub devices? A bit more work, but I think the firmware would be able to deal with that fairly easily, IIRC.

6by9, 2019-09-17:
It's done, but the kernel module defaults don't enable it.
Add ```snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_compat_alsa=0 ``` to cmdline.txt and you'll get multiple devices.
I think it is that the lxplugin then gets confused by the names and assumes it is still the old device with the cset switch. Change the device names slightly, or remove the check special handling from the plugin, and I think you're good to go.
I don't know if we lose support for the IEC958 passthrough though - it's not one that I have checked.

rebroad, 2020-01-25:
I put:-
`if [ `tvservice -l|tail -1|cut -d' ' -f6` -eq 1 ]; then
    echo Sound to HDMI1
    amixer -q cset numid=3 3
else
    echo Sound to HDMI0
fi`
in my .profile. This seems to work for speaker-test but not for omxplayer.