raspberrypi/firmware #1244: Pi 4B (4GB) cannot set gpu_mem above 512.

Kraust, 2019-09-06:
**Describe the bug**
Values of `gpu_mem` above `512` are not valid on the Pi 4B

**To reproduce**
I've tried setting the following values of `gpu_mem`:
- 512 (works)
- 768 (hangs on boot)
- 944 (hangs on boot)
- 1024 (hangs on boot) - this is expected as the maximum in the docs is 944.

**Expected behaviour**
Values up to 944 should successfully boot. Preferably, the user should be allowed to allocate more than 1GB of VRAM on a Pi 4 with more than 1GB of RAM.

**Actual behaviour**
Setting `gpu_mem` hangs on the values I've described above (768, 944, 1024).

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
- Pi 4B (4GB)

* Which OS and version (`cat /etc/rpi-issue`)?
```Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2
```

* Which firmware version (`vcgencmd version`)?
```Aug 15 2019 12:03:51 
Copyright (c) 2012 Broadcom
version 9f8431fb7839c7f00f52b81f5822ddab2b31d0db (clean) (release) (start)
```

* Which kernel version (`uname -a`)?
```
Linux wired 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux
```

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
I am trying to get video output stable on dual 4k30 monitors. If there is a newer firmware version (that one seems old), or any other remedies, I am open to suggestions. What is the suggested VRAM for a dual 4k30 setup?

timg236, 2019-09-06:
@JamesH65 I think you updated the documentation to say that allocating more than 512MB VideoCore memory is pointless on Pi4 because graphics texture memory is allocated from Linux system memory not VideoCore memory
Suggest close - NGTF

JamesH65, 2019-09-06:
Indeed. Anything over 512 on Pi4 (and actually, all Pi models I suspect) is a waste of time, untested, and will give no performance benefits. Not sure where this idea that these large values for GPU memory give better performance comes from.

closed 2019-09-06 by JamesH65

6by9, 2019-09-06:
gpu_mem and the CMA heap must both fit in the bottom 1GB of RAM. (Also VPU firmware).

With vc4-fkms-v3d loaded the CMA heap defaults to 256MB.
All resources for vc4-fkms-v3d come either from the CMA heap, or from general system RAM (mainly 3D textures).