raspberrypi/firmware #1246: Pi4 cannot discover and connect to 5GHz WiFi

thomas201008, 2019-09-07:
We need a minimal Linux with **5GHz WiFi connection** ( 2.4GHz WiFi or wired is not suitable ), as a thin client, and anything else will read from or store to an NFS server. So we made a small initramfs as rootfs.

We have deployed more than 50 pieces of **RPi3B+** with this minimal linux, and they works well.

The whole system architecture is easy to deploy, upgrade or maintain, effeciently.

As RPi4B released with better performance, we decided to use RPi4B in the coming deployment.

So we upgraded the minimal linux with **firmware-stable-4.19.66**, and RBi3B+ works well as before.

**But with the mininal Linux, RBi4B cannot discover any 5g wifi signal, why? Is there anything  special, essential for RPi4B we omited?**

Of course, with the official OS Raspbian Buster, RPi4B can discover and connect to 5ghz wifi.

**initramfs file list: ( wpa_supplicant and busybox are static built )**
./bin/wpa_supplicant
./bin/sh
./bin/busybox
./etc/wpa_supplicant.conf
./etc/udhcpc/default.script
./etc/init.d/rcS
./etc/profile
./etc/mdev.conf
./etc/services
./etc/ld.so.nohwcap
./etc/protocols
./init
./lib/firmware/brcm/brcmfmac43430-sdio.txt
./lib/firmware/brcm/brcmfmac43430-sdio.bin
./lib/firmware/brcm/brcmfmac43455-sdio.bin
./lib/firmware/brcm/brcmfmac43455-sdio.clm_blob
./lib/firmware/brcm/brcmfmac43455-sdio.txt
./lib/modules/4.19.66-v7+/kernel/net/wireless/cfg80211.ko
./lib/modules/4.19.66-v7+/kernel/net/rfkill/rfkill.ko
./lib/modules/4.19.66-v7+/kernel/drivers/net/wireless/broadcom/brcm80211/brcm80211/brcmutil/brcmutil.ko
./lib/modules/4.19.66-v7+/kernel/drivers/net/wireless/broadcom/brcm80211/brcm80211/brcmfmac/brcmfmac.ko
./lib/modules/4.19.66-v7l+/kernel/net/wireless/cfg80211.ko
./lib/modules/4.19.66-v7l+/kernel/net/rfkill/rfkill.ko
./lib/modules/4.19.66-v7l+/kernel/drivers/net/wireless/broadcom/brcm80211/brcm80211/brcmutil/brcmutil.ko
./lib/modules/4.19.66-v7l+/kernel/drivers/net/wireless/broadcom/brcm80211/brcm80211/brcmfmac/brcmfmac.ko

**/etc/init.d/rcS script:**
#!/bin/sh
/bin/busybox mkdir -p /boot /dev /mnt /proc /sbin /sys /tmp /var/log /var/run
/bin/busybox --install -s /sbin

mount -t proc proc /proc
mount -t sysfs sysfs /sys
mount -t devtmpfs devtmpfs /dev
mkdir -p /dev/pts
mount -t devpts devpts /dev/pts
echo "/sbin/mdev" > /proc/sys/kernel/hotplug
depmod -a
modprobe brcmfmac
read -rt3 && return

wpa_supplicant -B -Dnl80211 -i wlan0 -c /etc/wpa_supplicant.conf -d -f /var/log/wpa_supplicant.log
udhcpc -i wlan0 -s /etc/udhcpc/default.script
... ...

**/etc/wpa_supplicant.conf:**
ctrl_interface=/var/run
country=CN
network={
    scan_ssid=1
    ssid="mySSID"
    psk="myPWD"
}

**log on RPi3B+, scanned 5g wifi and discovered 5g wifi**
......
nl80211: Scan included frequencies: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825
wlan0: Event SCAN_RESULTS (3) received
wlan0: Scan completed in 2.805478 seconds
nl80211: Received scan results (6 BSSes)
wlan0: BSS: Start scan result update 1
wlan0: BSS: Add new id 0 BSSID e8:fc:af:98:3e:6f SSID 'DEBIAN8' freq 5745
wlan0: BSS: Add new id 1 BSSID c8:3a:35:03:44:58 SSID 'Tend_808' freq 2422
wlan0: BSS: Add new id 2 BSSID e8:fc:af:98:3e:6b SSID 'Debian8' freq 2417
wlan0: BSS: Add new id 3 BSSID 8c:a6:df:6d:26:87 SSID 'TP-LINK_2687' freq 2412
wlan0: BSS: Add new id 4 BSSID 68:89:75:08:29:78 SSID '' freq 2412
wlan0: BSS: Add new id 5 BSSID 40:16:7e:2c:c9:c0 SSID 'DebianEX' freq 2447
BSS: last_scan_res_used=6/32
wlan0: New scan results available (own=1 ext=0)
......

**log on RPi4B, scanned 5g wifi but found 2.4g only**
......
nl80211: Scan included frequencies: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825
wlan0: Event SCAN_RESULTS (3) received
wlan0: Scan completed in 2.559010 seconds
nl80211: Received scan results (6 BSSes)
wlan0: BSS: Start scan result update 2
BSS: last_scan_res_used=6/32
wlan0: New scan results available (own=1 ext=0)
wlan0: Radio work 'scan'@0x1d377e8 done in 2.582936 seconds
wlan0: radio_work_free('scan'@0x1d377e8): num_active_works --> 0
wlan0: Selecting BSS from priority group 0
wlan0: 0: c8:3a:35:03:44:58 ssid='Tend_808' wpa_ie_len=24 rsn_ie_len=20 caps=0x411 level=-60 freq=2422
wlan0: skip - SSID mismatch
wlan0: 1: e8:fc:af:98:3e:6b ssid='Debian8' wpa_ie_len=22 rsn_ie_len=20 caps=0x1411 level=-39 freq=2422
wlan0: skip - SSID mismatch
wlan0: 2: 8c:a6:df:6d:26:87 ssid='TP-LINK_2687' wpa_ie_len=22 rsn_ie_len=20 caps=0x431 level=-71 freq=2412
wlan0: skip - SSID mismatch
wlan0: 3: 24:05:0f:79:29:c8 ssid='360\xe5\x85\x8d\xe8\xb4\xb9WiFi-C8' wpa_ie_len=24 rsn_ie_len=20 caps=0x431 level=-72 freq=2437
wlan0: skip - SSID mismatch
wlan0: 4: 24:05:0f:a6:f8:2e ssid='160WiFi_F82F' wpa_ie_len=0 rsn_ie_len=20 caps=0x431 level=-76 freq=2412
wlan0: skip - SSID mismatch
wlan0: 5: 40:16:7e:2c:c9:c0 ssid='DebianEX' wpa_ie_len=22 rsn_ie_len=20 caps=0x411 level=-79 freq=2427
wlan0: skip - SSID mismatch
wlan0: No suitable network found
......

[RPi3Bplus.wpa.log.txt](https://github.com/raspberrypi/firmware/files/3586510/RPi3Bplus.wpa.log.txt)
[RPi4B.wpa.log.txt](https://github.com/raspberrypi/firmware/files/3586509/RPi4B.wpa.log.txt)

pelwell, 2019-09-09:
1. Make sure your "minimal OS" is using the same firmware files as Raspbian (/lib/firmware/brcm/).
2. Use `iwlist channel` to look for differences between the 3B+ and 4B.

I'm closing this because it's a question rather than a bug in our kernel, but you will still be able to post updates.

closed 2019-09-09 by pelwell

thomas201008, 2021-01-28:
1. The allowed 5GHz WiFi channels in China are 149, 153, 157, 161 and 165, according to the following link.
https://www.arubanetworks.com/vrd/OutdoorMIMOVRD/wwhelp/wwhimpl/common/html/wwhelp.htm#context=OutdoorMIMOVRD&file=AppA.html

2. To solve this issue in RPi4B, /lib/firmware/regulatory.db and regulatory.db.p7s are required.

Undefined-User, 2021-02-16:
> 1. The allowed 5GHz WiFi channels in China are 149, 153, 157, 161 and 165, according to the following link. [...]

Yes, but China government allow to use 36, 40, 44, 48, 52, 56, 60, 64 channels for many years https://en.wikipedia.org/wiki/List_of_WLAN_channels
I have same issue in RPi4B now, can you describe solution more clearly?

Undefined-User, 2021-02-16:
> 1. Make sure your "minimal OS" is using the same firmware files as Raspbian (/lib/firmware/brcm/).
> 2. Use `iwlist channel` to look for differences between the 3B+ and 4B.
> 
> I'm closing this because it's a question rather than a bug in our kernel, but you will still be able to post updates.

I think the Wi-Fi channels data is too old in RPi4B firmware, when I set Wi-Fi region to `CN`, I only can use channel `149, 153, 157, 161 and 165`, but channel `36, 40, 44, 48, 52, 56, 60, 64` is also official support in region `CN`. [List_of_WLAN_channels](https://en.wikipedia.org/wiki/List_of_WLAN_channels)

I trying change RPi4B Wi-Fi region to `US`, I can see Wi-Fi name but hardware will keep disconnecting and connecting because region problem.

thomas201008, 2021-02-16:
Fetch the following two files from Raspberry Pi OS or other distributions,
/lib/firmware/regulatory.db
/lib/firmware/regulatory.db.p7s
and place them in the same directory structure in your OS image.
Then, wpa_supplicant parameter country=CN/US will take effect.
I think you are making a very lightweight image.

thomas201008, 2021-02-16:
To use channels 36, 40, 44, 48, 52, 56, 60, 64, you have to set country=US, instead of CN.

thomas201008, 2021-02-16:
According to my debug procedure,  without the following two files, wpa_supplicant will not understand  the parameter country=**, instead, it will use default value 00. And RPi3BPlus will say, oh yeah, my country code is 00, I can use any channels.
But RPi4B will say, oh no, my country code is 00, I can only use limited channels.

/lib/firmware/regulatory.db
/lib/firmware/regulatory.db.p7s

The firmware from Broadcom is not open source, then we cannot find the reason why RPi3BPlus and RPi4B behave different with the same firmware, but you can trace dmesg log and wpa_supplicant log, to find some hints.

/lib/firmware/brcm/brcmfmac43430-sdio.bin
/lib/firmware/brcm/brcmfmac43455-sdio.bin
/lib/firmware/brcm/brcmfmac43455-sdio.clm_blob

Undefined-User, 2021-02-17:
> To use channels 36, 40, 44, 48, 52, 56, 60, 64, you have to set country=US, instead of CN.

Yes, I know set RPi4B country=US can discover and connect channels `36, 40, 44, 48, 52, 56, 60, 64`, but I have same problem with [this link(in Chinese)](https://yangwang.hk/?p=923), Raspberry Pi OS keep disconnecting and reconnecting Wi-Fi if I not also set router country to US, maybe RPi OS or RPi4B firmware use some way to temporary set Wi-Fi country code same with router, I think the only way to solve this problem is modify `regulatory.db` and `regulatory.db.p7s` file.

Undefined-User, 2021-02-17:
> According to my debug procedure, without the following two files, wpa_supplicant will not understand the parameter country=**, instead, it will use default value 00. And RPi3BPlus will say, oh yeah, [...]

I read #1359, download and replace newest `regulatory.db` and `regulatory.db.p7s` from [here](https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/) (it have correct channels value in readable file `db.txt`, I think binary file `regulatory.db` and `regulatory.db.p7s` is correct too), but the result of command `iwlist channel` not changed when set region=CN, I think there is a problem in the wireless driver.

Undefined-User, 2021-02-18:
Temporarily resolved, see https://github.com/raspberrypi/firmware/issues/1359#issuecomment-781081935

All3xJ, 2023-10-31:
lol changing localization options -> wlan country inside raspi-config, fixed it! I changed to China which is the country that seems to use the 149 channel (that my Xiaomi/Redmi phone hotspot was using) and I FIXED!