raspberrypi/firmware #1248: tvservice fails to switch HDMI mode with NOOBS/PINN.

procount, 2019-09-09:
(I believe this is just a framebuffer issue and unrelated to kms/fkms changes).

NOOBS & PINN have a feature to switch display modes between:
1) HDMI preferred (tvservice -p)
2) HDMI safe (tvservice -e 'DMT 4 DVI')
3) Composite PAL (tvservice -c 'PAL 4:3')
4) Composite NTSC (tvservice -c 'NTSC 4:3')
Up until the release of the Pi4, these worked ok.

At Pi4 launch, NOOBS 3.2 and PINN 3.3 still worked fine on older models, but on Pi4, they could no longer switch to the composite modes. I understand this is due to having to change the clock setup, which is not possible without rebooting, so nothing to be done about that. For both of these, the rpi-firmware commit is e5b11a9d32cf4388b09097c33ab7bffbe1ca721c from the private repo. (PINN 3.3, just copied the bootcode.bin and recover*.elf files from NOOBS 3.2).

Subsequent releases of PINN have tried to use a comparable commit in the public firmware repo starting with 64b5649a41b69d09bfe0ed05448d28a66be3edfd. However, when trying to use any pi4 compatible firmware from the public repo, the tvserice commands to switch the HDMI between 'preferred' and 'safe' modes have also stopped working on all models. 

When switching to 'safe' mode (by pressing key '2'), the monitor changes to 640x480 mode, but it looks like the framebuffer does not, and a weird display is shown. Then when pressing key '1' to switch back to 'preferred' mode, the monitor will switch to preferred, but the screen mostly goes black. Occasionally there will even be a seg fault.

I have tried various combinations of
kernel commits:
```
2ff97445cd97d872b08ad01f084bdfcb3c892b64 (vc5) 4.19.37
be0a940de66666c10c9071cddafce6284c400734 (vc5) 4.19.46
db83d4b7c10d0a4251780d2c7a033c2a1216d47c (pub) 4.19.46
883a1a5bf14abc9b3a014a3907772cd0b3129c8a (pub) 4.19.50
71d47f4c4bd7fd395b87c474498187b2f9be8751 (pub) 4.19.50
```
rpi-userland commits:
```
de4a7f2e3c391e2d3bc76af31864270e7802d9ac (pub) 4.14?
95b29b556a4068725ea9948f6e78790ab3aa153e (pub) 4.19.46?
17d2fdc1abd370e09ba7074753294c7976dd6b0d (pub) 4.19.?
```
and firmware commits:
```
e5b11a9d32cf4388b09097c33ab7bffbe1ca721c (pri) 4.19.46?
64b5649a41b69d09bfe0ed05448d28a66be3edfd (pub) 4.19.50
e10b2880e26708afc0b32a485c1231207037fc45 (pub) 4.19.?
```
Which lead me to think the problem is restricted to the firmware changes starting soon after the e5b11a9d32cf4388b09097c33ab7bffbe1ca721c private commit, or maybe a mismatch between the firmware and the kernel or userland.

I'm at a bit of a loss of how to advance the firmware to anything pi4 compatible on the public repo that avoids this issue. Whilst the composite mode switching may have to be deprecated, at least on the Pi4, it would be nice to keep the HDMI resolution switching for problematic displays.

JamesH65, 2019-09-17:
I was under the impression that changing mode using tvservice always destroyed the framebuffer, and you need to do an fbset to get it back again (in non-fkms mode anyway). So not sure why this might ever have worked. Have I misunderstood the problem?

procount, 2019-09-23:
Yes, I understood that too, and I think that must happen during the call to setMode() here -> https://github.com/raspberrypi/noobs/blob/master/recovery/mainwindow.cpp#L777 .
So maybe the problem is not tvservice, but the later versions of firmware are not setting the framebuffer correctly in  QScreen::instance()->setMode(...) ...? 
But IIRC, this problem occurs even when I run `tvservice...; fbset -depth...` at the command shell prompt.

JamesH65, 2020-01-02:
@procount Hi, after all the changes to various bits of code in this area over the last couple of months, does this issue still exist?

procount, 2020-01-02:
I'll have to rebuild and test.
I currently just copy the firmware used in NOOBS as it's still in the private repo.

Could you suggest some consistent commit-IDs I should use for the kernel, userland and firmware repos?

procount, 2020-01-02:
I rebuilt PINN with rpi-firmware-0c01dbefba45a08c47f8538d5a071a0fba6b7e83.tar.gz.
On 3B+ it's still as bad as before, so no, it has not been fixed.

JamesH65, 2020-01-03:
OK, thanks for checking, will leave issue open.

procount, 2020-01-13:
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=261818 sounds like the same issue.

JamesH65, 2020-01-27:
Been taking a look. With the latest kernel and firmware, things appear to work as expected. What I did need to do is specify the x and y resolution in the fbset command in order for the correct display to come up. I wonder if there has been a slight change underneath such that it no longer reads the current HDMI resolution prior to creating the framebuffer, and just uses what the last value was. Sounds possible - this area has changed a lot with the multi fFB support being added, along with changes to tvservice to support that.

procount, 2020-01-27:
> Been taking a look. 

Thanks James,
> With the latest kernel and firmware, things appear to work as expected.

Is that with Raspbian (cos I think that's ok) or have you got a new NOOBS build with the latest kernel and firmware?
If in NOOBS, did you try pressing keys '2' and '1' to change the mode?
Or had you dropped to the cmdline (ctrl-alt-f2) and used the tvservice and fbset commands in NOOBS?

JamesH65, 2020-01-27:
This is with Raspbian - I have no idea how NOOBS or PINN work. I think @XECDesign may be about to do a NOOBS build with latest kernel for further checking.

procount, 2020-01-27:
Yes, so I believe, with all latest kernel and firmware etc. 
For a quick check, I took an existing PINN version and just replaced bootcode.bin and recovery.elf from rpi-firmware 0c01dbefba45a08c47f8538d5a071a0fba6b7e83 (attached as bad.zip)
[bad.zip](https://github.com/raspberrypi/firmware/files/4117595/bad.zip). I suspect NOOBS will exhibit the same problem.
When booting PINN and pressing key '2' to get HDMI safe mode, the attached image shows what it looks like with this firmware, or any other since pi4 launch:
[image]

procount, 2020-01-29:
Somewhere between the commit IDs I tested above, and the latest packages that XECDesign used, this issue seems to have been fixed, judging by the basic test I've done. So I'll close this for now. Thanks.

closed 2020-01-29 by procount