raspberrypi/firmware #1249: Raspberry Pi v4 - hdmi_mode=31 (1080p50) not working

LOMF, 2019-09-09:
Hi,

So far I haven't been able to find any similar bug reports that describes this issue.
My knowledge for Linux systems/raspberry pi is quite limited so I apologize in advance if this issue is due to my own ignorance. I'm running the lastest build of Dietpi but I believe this issue is reproducible on a clean installation of Raspbian as well.

**Describe the bug**
I'm not able to set the raspberry pi to output CEA 1080p50 resolution ( hdmi_mode=31 ) on my Raspberry Pi v4.
It boots up for a second and then looses signal (if a monitor is in power saving it's usually to quick for the monitor to display anything before it looses the signal again).

Setting it to 1080p60 ( hdmi_mode=16) works without any issues. 

I know 1080p50 might not be the most common format but it's the PAL standard used in European broadcasts and my project needs to output 1080p50 to a video switcher. 

The "monitor" I use is a Blackmagic Design Micro HDMI to SDI converter - and I use those with other sources with 1080p50 all the time so I know that the format is supported.

**To reproduce**
Change the following settings in /boot/config.txt
hdmi_group=1
hdmi_mode=31

**Expected behaviour**
Video signal from the device in 1080p50 format.

**Actual behaviour**
Lost video signal after boot.

**System**
https://pastebin.com/ZjrgYRfS

6by9, 2019-09-09:
Pi4 is expected to be used with the vc4-fkms-v3d driver as there is no firmware GL/GLES driver. You appear to have disabled that. Is that for any particular reason?
FKMS does rely on the display device providing a sensible EDID, but I would have expected a Blackmagic device to do that.

1080p50 is working fine with all the monitors I have tried it with. It may be that your device requires a slightly altered version of the timings rather those dictated by CEA mode 31.

LOMF, 2019-09-09:
Hi and thank you for your reply.

The only reason why FKMS was disabled was because I was testing around with different settings to see if I could get anything to work.

I've also tested plugging in different monitors directly with the same outcome so the blackmagic adapter does not seem to be the issue in itself.

Question - did you test with a Pi v4 or a earlier version? because I have gotten the exact same setup to work with Pi v3 before.

JamesH65, 2019-09-17:
I've just tried this with a Dell monitors (1920x1200) and pi4 running fkms.  Set the required mode in config.txt and it actually started in a 60Hz mode, probably because the screen config utility last set it up in that mode. However, changing the frequency using the configuration util did result in a 50Hz 1080p mode being selected, as shown by `tvservice -s`. So not sure what is happening in @LOMF case. Are you still encountering the problem?

LOMF, 2019-09-19:
Hi
It think I finally found the issue and it was not related to the raspberry pi after all. 

For any googles that find this in the future, in my case it looks like 1080p50 has issues when using a Blackmagic Design Micro converter for HDMI to SDI and then a Blackmagic BiDirectional adapter for converting back from SDI to HDMI. When I swapped the BiDirectional adapter for another one it worked. What's wierd is that this issue does not appear using 1080p59.94 or 1080p60.

Thanks for the help testing it out and sorry for wasting time.

closed 2019-09-19 by LOMF