raspberrypi/firmware #1252: rpi 2 - ethernet led issue

Schlunze, 2019-09-22:
**Hello,**

after set the following options in config.txt (rpi 2)
> dtparam=eth_led0=14
dtparam=eth_led1=14

ethernet leds still on after reboot. 
I expected them to be off on pi2 according to your [readme](https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README)?
OS is libreelec (9.1.501) if this matters.

tx in advance for your reply

trejan, 2019-09-23:
The documentation needs to be updated. eth_led0 and eth_led1 are only for the LAN7515 USB Ethernet chip used in the 3B+. The 3B and earlier boards have a completely different USB Ethernet chip which doesn't use those two options. You need to download and run the lan951x-led-ctl utility at each boot to turn off the LEDs.

Schlunze, 2019-09-24:
Thank you for the useful information trejan.

Greetings

closed 2019-09-24 by Schlunze