raspberrypi/firmware #1253: Stereoscopic support is broken on firmware newer than 4.19.57

realizator, 2019-09-23:
I have 2 cameras system based on RPi Compute Module 3+ Lite.
The latest firmware with working stereoscopic support is 4.19.57 (Buster). Under "stereoscopic support" I mean appropriate work of _raspistill_ and _raspivid_ with "-3d" option.

Commands used for tests:
```
raspistill -3d sbs -o 1.jpg
raspivid -3d sbs -w 1280 -h 720 -o 1.h264
```

With all newer versions (4.19.66, 4.19.71 and 4.19.73) we have for both raspivid and raspistill:
- preview is started
- preview image freezes
- adding '-n' option (no preview) has no any effect
- no image or video saved
- system can not be shut down correctly (sudo poweroff), and frozen preview image is always on the screen

6by9, 2019-10-08:
Sorry, not had time to investigate this one as yet. 
I have just had a daft thought. The regression is in the right time frame to be due to the AWB algorithm change. Could you try running ```sudo vcdbg set awb_mode 0``` BEFORE running raspivid and see if it is happy?

realizator, 2019-10-08:
> Sorry, not had time to investigate this one as yet.
> I have just had a daft thought. The regression is in the right time frame to be due to the AWB algorithm change. Could you try running `sudo vcdbg set awb_mode 0` BEFORE running raspivid and see if it is happy?

@6by9 I confirm that `sudo vcdbg set awb_mode 0` do the trick!
Both raspistill and raspivid works. Tested on 4.19.76

6by9, 2019-10-08:
Thanks. I'll get someone to look at the AWB algorithm again.

realizator, 2020-04-15:
Current situation update on 15 of April: stereoscopic support works again out of the box after rpi-update.
It has been tested on both current and "common" firmware, for 4.19.114-v7+ and 5.4.29-v7+ kernels.
Related discussions:
[stereoscopic mode](https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=85012&e=1&view=unread#p1643535) tests (RPi forum)
[common firmware](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=267576&start=25#p1643032) tests (RPi forum)

pelwell, 2020-04-15:
Is there any reason not to close this issue?

realizator, 2020-04-15:
> Is there any reason not to close this issue?

Ok, I'm closing it. Can not wait for this firmware update will promoted to apt. :-)

closed 2020-04-15 by realizator