raspberrypi/firmware #1255: 32-bit SMP boot is broken

evangreen, 2019-09-28:
**Describe the bug**
My 32-bit bare-metal kernel.img cannot start the other cores using the mailbox.

**To reproduce**
My config.txt is nearly empty: kernel_address is 0x8000. When I load my bare metal OS and write to mailbox 3 set, I find that the other cores never boot, and mailbox 3 clear is never cleared.

I did a little dumping of memory around 0, notice the last line:
```
00000000 ea000007 b	0x00000024 [0x00000024]
00000004 0124f800 msreq	cpsr_s, r0
00000008 e3001131 movw	r1, #305  ; 0x131
0000000c ee011f11 mcr	p15, 0, r1, c1, c1, 0
00000010 e30001da movw	r0, #474  ; 0x1da
00000014 e16ff000 msr	spsr_cxsf, r0
00000018 e1b0f00e movs	pc, lr
0000001c 00063fff strdeq	r3, [r6], #-255
00000020 400000cc <<<<< notice this
```
The address the APs are spinning on is the old BCM2836 mailbox address at 0x400000cc, not the BCM2711 address at 0xff8000cc.

If I patch that address in start4.elf, then the other cores boot fine:
```
$ md5sum start4.elf.orig
15fd86732474dcd39c902c7a804bd9dd *start4.elf.orig

$ cmp -l start4.elf.orig start4.elf
2330263   0 200
2330264 100 377

```

Please fix the 32-bit spin address in start4.elf to point to 0xff8000cc rather than 0x400000cc, where there's just garbage memory and the CPUs jump off into space.

pelwell, 2019-09-30:
I suspect your problem is caused by the firmware's conflation of BCM2711 and GIC support (which I think you don't want), but it can be solved using a custom armstub.

The [armstubs](https://github.com/raspberrypi/tools/tree/master/armstubs) are small executables that contain the very first instructions run by the ARM processors. They perform some very low-level processor initialisation before branching to the "kernel". BCM2711 required a new stub file because the A72 needs some additional configuration _and_ to enable the GIC (ARM's Generic Interrupt Controller) while in secure mode. Raspberry Pi kernels and .dtbs for the 2711 all use the GIC, and the GIC is currently only found on 2711, so the stub build that supports 2711 enables the GIC, hence the name `armstub8-32-gic.bin`; there is no `armstub8-32-nogic.bin`.

The firmware executables (start*.elf) include embedded default stub files, and they choose the one to use based on knowledge of the processor type and whether or not the GIC is wanted. However, I think the rules are wrong: it will currently fall back to using a BCM2710 stub if the BCM2711 version isn't found (or if the GIC is disabled), but the different I/O base means this is bogus. I also believe there should be nothing wrong with using the -gic stub even when using the old interrupt controller.

Fortunately the firmware respects the `armstub=` config.txt setting that forces the use of any stub file you like. You can download the stub sources from the tools repo and build them yourself, or download a prebuilt `armstub8-32-gic.bin` [here](https://drive.google.com/file/d/19Psdx0o0h2wCKNIWM-kIXzw8os4TlVgv/view?usp=sharing). Copy the result to the root of the FAT partition and add `armstub=armstub8-32-gic. bin` to config.txt. (If you want to support multiple Pi models, put the setting in a `[pi4]` section. You should then find that the spin address is correct.

Let us know how you get on. If this workaround is successful in your non-GIC OS we'll make this the default firmware behaviour.

evangreen, 2019-10-01:
Hi Phil,
Thanks for all those details, and for filling in the pieces I failed to describe. That totally makes sense. When I set `enable_gic=1` in config.txt, then all cores come up fine.

In addition, to my absolute delight, I also find that the horrible cache coherency issue I was debugging seems to go away when I set `enable_gic=1`. I was finding that when I tried to enable the MMU & caches, I would end up with random cache-line-size runs of garbage sprinkled throughout the VAs, despite the fact that 1) the PAs were fine, and 2) I invalidated all levels of the cache just before enabling the MMU & caches. 

I wonder if not properly enabling SMP coherency is another symptom of the BCM2710/2711 mixup you pointed out. I think your proposed fix would cover that case as well.

Does this pretty much tell you everything you need to know? Or do you still want me to try out the armstub you linked to?

The funny thing is we do have support for the GIC. I just thought I was saving myself some time in bringup by only changing one variable at a time. The universe has a sense of humor.

popcornmix, 2019-10-11:
Latest rpi-update firmware may have a fix for this.

pelwell, 2020-01-10:
Closing for lack of activity,

closed 2020-01-10 by pelwell