raspberrypi/firmware #1258: No 2160p on Pi4 through 4K Receiver

Off-Load, 2019-10-02:
I'm trying to get 2160p to work trough my Sony STR-DN 1040 receiver, it is connected as following PI4 -- STR-DN1040 -- Samsung UE58 MU6125. This setup gives me only 1080p, changing the resolution to 3840x2160p and 4096x2160p  gives me an black screen, with an No Signal message. This behavior is on both HDMI 0 and HDMI 1.
Connecting the Pi directly to the TV gives me 2160p with no problem.

This issue has been discussed on the Libreelec forum in this thread, [https://forum.libreelec.tv/thread/20449-no-2160-on-pi4-through-sony-4k-receiver/?postID=128271#post128312](url) but also running Rasbian Buster desktop with all updates and upgrades gives the same result, so I was advised by Popcornmix to open an issue here, to see if the problem could be resolved. (black screen No Signal when choosing an 2160p 24/25/30hz resolution)

* Clean install of Rasbian Buster ( no changes in config.txt )
* All cables is 4K certified (tried two MicroHDMI cables, and one MicroHdmi adapter)
* The pi4 is connected to an 4K input on my receiver (did try BD, Video1, and Game)
* Did try with CONFIG_HDMI_BOOST=7 to 9 but no difference ( does it work on pi4? )
* 2160p works trough AVR with an Android S905w box running Coreelec
* UHD is set to "ON" on the TV
* No problem with CEC 
* Tested with Amlogic X96mini and the following modes works through the AVR
__2160p YcbCr 444 8bit 24/25/30hz works
2160p YcbCr 422 10bit 24/25/30hz works
2160p YcbCr 422 8bit 24/25/30hz works
2160p RGB 8bit 24/25/30hz works_
* 2.5A Psu (only wireless keyboard adapter is connected) 

Modes supported by the AVR:
[image]

Modes supported by the TV
[image]

**This is the modes that currently is reported back by tvservice.
tvservice -m CEA**
Group CEA has 18 modes:
mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive
mode 2: 720x480 @ 60Hz 4:3, clock:27MHz progressive
mode 3: 720x480 @ 60Hz 16:9, clock:27MHz progressive
mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive
mode 5: 1920x1080 @ 60Hz 16:9, clock:74MHz interlaced
(prefer) mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive
mode 17: 720x576 @ 50Hz 4:3, clock:27MHz progressive
mode 18: 720x576 @ 50Hz 16:9, clock:27MHz progressive
mode 19: 1280x720 @ 50Hz 16:9, clock:74MHz progressive
mode 20: 1920x1080 @ 50Hz 16:9, clock:74MHz interlaced
mode 31: 1920x1080 @ 50Hz 16:9, clock:148MHz progressive
mode 32: 1920x1080 @ 24Hz 16:9, clock:74MHz progressive
mode 33: 1920x1080 @ 25Hz 16:9, clock:74MHz progressive
mode 34: 1920x1080 @ 30Hz 16:9, clock:74MHz progressive
mode 93: 3840x2160 @ 24Hz 16:9, clock:297MHz progressive
mode 94: 3840x2160 @ 25Hz 16:9, clock:297MHz progressive
mode 95: 3840x2160 @ 30Hz 16:9, clock:297MHz progressive
mode 98: 4096x2160 @ 24Hz unknown AR, clock:297MHz progressive

**Running "tvservice -M" gives indications that HDMI connections are lost when switching to any 2160p resolution 24/25/30/hz.**

_[I] HDMI in HDMI mode
[I] HDMI cable is unplugged. Display 2
[I] HDMI is attached
[I] HDMI cable is unplugged. Display 2
[I] HDMI is attached_

So can anyone have an idea, why I can't get 2160p to work trough the AVR?

pelwell, 2019-10-02:
I'm not near a Pi at the moment, but I think there is an option in the settings GUI to enable 4K support.

6by9, 2019-10-02:
Apologies if you've done this on the libreelec thread, but could you post the raw EDID please for analysis?
Use ```tvservice -d edid.bin``` to save it, and then either attach it here, or use base64 or similar to convert it to text and copy/paste on here.

timg236, 2019-10-02:
4Kp24..p30 should work without config.txt changes, hdmi_enable_4kp60=1 is required for p50 & p50 because the core clock frequency needs to be boosted.

It probably won't help but it would interesting if setting hdmi_deep_color to 5 or 6 makes any difference, it potentially picks different PHY settings but if it's failing on both HDMI ports I'm expecting it to make no difference

6by9, 2019-10-02:
I have a vague recollection that @JamesH65 may have already looked at some of this already, but I don't recall whether it was this device or a different one.

I have seen invalid EDIDs which DRM is interpreting them more strictly than some other parsers, and that gives some strange results. DRM does rely so heavily on the EDID that there is little we can do if it is iffy.

Off-Load, 2019-10-02:
> 
> 
> Apologies if you've done this on the libreelec thread, but could you post the raw EDID please for analysis?
> Use `tvservice -d edid.bin` to save it, and then either attach it here, or use base64 or similar to convert it to text and copy/paste on here.

No problem, I'm happy for all the help I can get 👍 

[Samsung MU6125_edid.zip](https://github.com/raspberrypi/firmware/files/3683580/Samsung.MU6125_edid.zip)

Off-Load, 2019-10-02:
> 4Kp24..p30 should work without config.txt changes, hdmi_enable_4kp60=1 is required for p50 & p50 because the core clock frequency needs to be boosted.
>
> It probably won't help but it would [...]

I tried to add hdmi_enable_4kp60=1 in config.txt, but that didn't help, still no 2160p output. I will not get a 60hz signal anyway, because the AVR doesn't support RGB 4.4.4, but I had to try to se if i could at least help to get 25/30hz 2160p.The AVR supports RGB 4.4.4 up to 30hz on 2160p

How do I change the hdmi_deep_color setting? 

hdmi_deep_color? Where can I change this setting?

Off-Load, 2019-10-02:
> I have a vague recollection that @JamesH65 may have already looked at some of this already, but I don't recall whether it was this device or a different one.
>
> I have seen invalid EDIDs which DRM is [...]

I was thinking this could be an DRM problem to, but it makes no sense, because the X96mini android box has no problem getting an 2160p30hz signal through the AVR and output it on the TV.

But I'm pretty noob when it comes to videosignals modes, DRM, EDID, and such, so I'm barely hanging on 🤒 
But every comments here is appreciated, hopefully there is an solution to this problem 👍

timg236, 2019-10-03:
Add hdmi_deep_color=6 to config.txt in the boot partition.
NB deep colour doesn’t work properly yet, it’s just an experiment to force a different pixel clock

Off-Load, 2019-10-03:
> 
> 
> Add hdmi_deep_color=6 to config.txt in the boot partition.
> NB deep colour doesn’t work properly yet, it’s just an experiment to force a different pixel clock

That makes sense, I never had heard of hdmi_deep_color, therefor I had to ask.

It made some difference, tried both 5 and 6, and when running tvservice -M I didn't get the "[I] HDMI cable is unplugged. Display 2" message like I do when I run it without hdmi_deep_color, but still an black screen, but without the "No Signal" message on the TV. Buster bumps back to 1080p after a few seconds
Tried both 4096 and 3840. on 24/25/30Hz

Thanks 👍

popcornmix, 2019-10-03:
Can you also capture (and post) the EDID when directly connected to the TV.
What you can try is adding ``hdmi_edid_file=1``/``hdmi_force_hotplug=1`` to config.txt and place the ``edid.dat`` file (from just the TV attached) into the boot partition.

If you now reconnect the AVR does it behave any differently?
That should tell us whether the timings from the AVR's EDID are causing the problem or if the AVR just doesn't like the 4kp30 signal that the TV is happy with.

6by9, 2019-10-03:
My Dell 4k monitor is totally happy with that EDID.

There's nothing dubious looking in it. The detailed modes exactly match VICs 16 (1080p60) and 20 (1080i50).
The EDID does advertise YUV420 VICs 96 (3840x2160@50), 97 (3840x2160@60), 101 (4096x2160@50), and 102 (4096x2160@60), but DRM is correctly filtering them out as the clock frequency is too high.

Getting hot plug events does seem a little strange. Hotplug should normally only change state if something significant has changed with the sink (TV or AVR) which isn't the case in this situation. That said, the Pi is generally ignoring hotplug at the moment other than reporting it through tvservice, so it shouldn't make a big difference.

Off-Load, 2019-10-03:
> Can you also capture (and post) the EDID when directly connected to the TV.
> What you can try is adding `hdmi_edid_file=1`/`hdmi_force_hotplug=1` to config.txt and place the `edid.dat` file (from [...]

adding hdmi_edid_file=1/hdmi_force_hotplug=1 and putting EDID.bin in /boot gives me only 1024X768 at max. and only 60Hz  This happens also when connected through the AVR

I will try again without rebooting the Pi, and switch directly.

[direct_to_MU6125_edid.zip](https://github.com/raspberrypi/firmware/files/3685674/direct_to_MU6125_edid.zip)

popcornmix, 2019-10-03:
File needs to be called ``edid.dat``

6by9, 2019-10-03:
> File needs to be called edid.dat

Or you can add ```hdmi_edid_filename=Samsung_MU6125_edid.bin``` as well.

Off-Load, 2019-10-03:
> 
> 
> File needs to be called `edid.dat`

Changed it to edid.dat, but no difference, still an black screen when choosing 2160p.

Off-Load, 2019-10-03:
> My Dell 4k monitor is totally happy with that EDID.
>
> There's nothing dubious looking in it. The detailed modes exactly match VICs 16 (1080p60) and 20 (1080i50).
> The EDID does advertise YUV420 VICs [...]

Thanks for looking into it.

I have no idea whats going on here it seems that the Pi sends an signal that the AVR doesn't like obviously, or its to weak.
Wondering whats makes the cheap X96mini's HDMI work, but not the Pi's HDMI ?

popcornmix, 2019-10-03:
A final sanity check. With ``hdmi_force_hotplug=1`` boot connected to display and 4kp30.
Check you can unplug and replug hdmi to display and image remains.

Now connect through AVR. I'm guessing it won't work.

With ``hdmi_force_hotplug=1`` the Pi won't know when it's unplugged and will continue outputting exactly the same signal over hdmi (presumably a valid one based on your TV being happy and the many other users who can do this).

This suggests the AVR doesn't like the same signal that the TV does like.

Possibilities are the AVR is a little "deafer" than the TV and the signal is too weak. This might be made worse by a less than great hdmi cable.
The AVS is not supporting something we are driving (and the X96 just happens to be defaulting to something else). No idea what this is, but things like whether we output RGB or YUV, full range or limited range, DVI or HDMI drive, CEA or DMT mode, reduced/non-reduced blanking etc.

When directly connected to TV does the TV provide any info about the signal? Are there any differences between X96 and Pi4? (unfortunately displays often don't show much more than width*height@framerate, but some are more useful.

Off-Load, 2019-10-03:
**_With hdmi_force_hotplug=1 boot connected to display and 4kp30.
Check you can unplug and replug hdmi to display and image remains._**

The image remains (3840 × 2160p30Hz)

**> Now connect through AVR. I'm guessing it won't work.**
No, you are correct, No picture, and No Signal message (display on AVR switching between "BD input and "No Stream" 

**> When directly connected to TV does the TV provide any info about the signal? Are there any differences between X96 and Pi4?**

No , If I press the Info button on the remote, its just the usual stuff, like width*height@framerate, 
But I'm going to try to see if I can see anything in the root menu of the TV, if there is any settings there, or something that I can turn on to monitor whats happening.
I will report back tomorrow on that one, since messing with root on an Tizen TV, and beer don't mix so well 👍 

Thanks

Squall1Titus, 2020-01-23:
I have the same problem and it is already mentioned in several forums.
Apparently the 4b and some TVs have problems with 4k.
I have a 4k Samsung and after booting up Noobs --> LibreElec (KODI) the screen switches to black and in the info only 3840x2160 30Hz is shown. I hear sound but see nothing. 
And yes various cables were tested.
On my Pi 3 everything is displayed without problems, so it can't be a HW problem on the non-Pi side.
If I connect the 4b to my 4k monitor, I have picture and everything is as expected. So it must be the combination TV and Pi 4b.

Please have a look at this. Because I am far from being the only one with this problem. And without picture, KODI doesn't make sense.