raspberrypi/firmware #1259: kernel8.img loading broken with device_tree_address specified in config.txt using Sep 3, 2019 or later firmware

satmandu, 2019-10-02:
Simplest test case: copy https://github.com/Hexxeh/rpi-firmware/blob/master/kernel8.img to a sd card, add the current firmware, and have this in the config.txt file:
```
[all]
device_tree_address=0x03000000
enable_uart=1
```
The system halts at the rainbow screen.
This appears to happen for any kernel8.img file.

The Sep 3 commit "firmware: Avoid defaulting to 64-bit kernel on older Pi" appears to be the problem.

I've tried this with other kernels and u-boot. As far as I can tell u-boot doesn't load when I have ```device_tree_address=0x03000000``` in my config.txt. Without that line u-boot has FDT errors and doesn't load, so I need to be able to pass device tree address from config.txt, or have u-boot pick it up automatically. 

As an example I am using this u-boot binary named to be kernel8.img with this config.txt:
```
[all]
device_tree_address=0x03000000
enable_uart=1
```
This does not load.

This does load:
```
[all]
#device_tree_address=0x03000000
enable_uart=1
```
But u-boot needs the FDT address at this point for my usage, and it complains about it here.

(U-Boot is useful on arm64 since one can use a boot script to decompress compressed vmlinuz files, which don't auto-decompress like on 32 bit.)

System info:
```
cat /sys/firmware/devicetree/base/model
Raspberry Pi 3 Model B Plus Rev 1.3

vcgencmd version
Sep  3 2019 23:52:29 
Copyright (c) 2012 Broadcom
version 6dc34ddf57a9cb5e234d00a4490d0b6a31688526 (clean) (release) (start)
```
That's the last working version, and reverting to be9ca72bb474863afa17cf67c5b1dd7630c11365 allows boot to succeed.

(Current bootcode.bin isn't causing a problem. Reverting the start*.elf and fixup*.dat files allows boot to succeed.)

pelwell, 2019-10-02:
This ought to work, so there is a bug/limitation of some kind. However, I'm fairly sure U-boot has a mechanism to use the DT at the location chosen by the firmware - you shouldn't need to use a fixed address. Have you looked at CONFIG_OF_BOARD and what it does?

pelwell, 2019-10-02:
@agherzan and @mbgg should be able to give more definitive answers to the U-Boot side of things.

satmandu, 2019-10-02:
>  However, I'm fairly sure U-boot has a mechanism to use the DT at the location chosen by the firmware - you shouldn't need to use a fixed address. Have you looked at CONFIG_OF_BOARD and what it does?

I'm trying not to muck around with recompiling u-boot with other options until the patches which @mbgg is working on for the single rpi2/3/4 u-boot binary get accepted upstream so I can avoid having multiple u-boot binaries sitting around for various pi generations, but I'll try that as a workaround to see if it helps.

satmandu, 2019-10-02:
For what it is worth, this in config.txt works around it, with no kernel8.bin file in the boot partition.

```
kernel=kernel8.bin
device_tree_address=0x03000000
```

mbgg, 2019-10-03:
@satmandu I don't really understand fully what you want to do. Can you please explain what you want to achieve. 
What options do you use in U-Boot?
How you want to use one binary without my patches?

satmandu, 2019-10-03:
@mbgg Sorry for being obtuse. I'm just booting the arm64 ubuntu 19.10 distribution, and noticed that recent firmware updates kept it from booting on a 3b+ due to the pre-installed u-boot binary not working with the recent firmware due to the device_tree_address config.txt option halting boot.

I'm also booting ubuntu arm64  on a 4b, which of course does require a more recently compiled u-boot binary. But Ubuntu uses a bunch of custom patches for u-boot as per http://archive.ubuntu.com/ubuntu/pool/main/u/u-boot/u-boot_2019.04+dfsg-2ubuntu3.debian.tar.xz and I didn't want to go down the rabbit hole of applying those patches to current and also fiddling with u-boot options like CONFIG_OF_BOARD and trying to figure out what was going wrong, as it seems there are just too many variables. 

So I was hoping to wait until you had your universal rpi binary work with the CONFIG_OF_BOARD stuff sorted out and upstreamed before trying to swap in a personally compiled u-boot binary at this point and opening a ubuntu ticket suggesting those changes.

(The goal for me is get booting working with current firmware using a near stock option u-boot binary so I can suggest to a distribution like debian/ubuntu to just use that.)

juergh, 2019-10-07:
@satmandu There is an unofficial Ubuntu u-boot that supposedly supports the PI 4: https://launchpad.net/~waveform/+archive/ubuntu/ubpi4/+packages. Not sure about the device_tree_address issue though.

mbgg, 2019-10-07:
I suppose what they do is build two binaries and tweak the config.txt to use one or the other depending on the RPi version they are booting from.

satmandu, 2019-10-07:
@mbgg  That seems to be the direction they are headed. But that's all a workaround for not being able to use a single u-boot binary to support all rpi variants. 

It seems... unnecessarily complicated? But I suppose that's the only approach which works with currently available u-boot builds.

pelwell, 2019-10-07:
FYI The original reported bug is high on my list of issues to look at.

pelwell, 2019-10-08:
Doh - this was a trivially simple bug. Providing a sensible `device_tree_end` (the exclusive end point for the merged DT) is a safe workaround until the fix is merged.

satmandu, 2019-10-08:
> Doh - this was a trivially simple bug. Providing a sensible `device_tree_end` (the exclusive end point for the merged DT) is a safe workaround until the fix is merged.

Any suggestions for what a conservative `device_tree_end` should be?

Also, https://www.raspberrypi.org/documentation/configuration/device-tree.md states:
>`device_tree_end` This sets an (exclusive) limit to the loaded device tree. By default the device tree can grow to the end of usable memory, which is almost certainly what is required.

Which is almost certainly _more_ than what is required?

pelwell, 2019-10-09:
> Any suggestions for what a conservative `device_tree_end` should be?

device_tree_start + 1MB is much more than you would need in practise.

> Which is almost certainly _more_ than what is required?

No (and yes), I just meant that the way memory is likely to be used it is unlikely to have any negative consequences.

pelwell, 2019-10-09:
Thanks for the confirmation. The next firmware build will contain the fix.

popcornmix, 2019-10-11:
Latest rpi-update firmware may have a fix for this.

satmandu, 2019-10-12:
I am seeing a resolution of this issue on a 3b+ with current firmware.

closed 2019-10-16 by pelwell