raspberrypi/firmware #1260: Turning LCD Backlight Off Using RPi4/Buster with dpms

frankgould, 2019-10-03:
**What I'm trying to do**
I would like to configure my RPi4B:4GB desktop to shut off the monitor after a certain amount of time using `xset dpms 300 300 300`. Currently, the timeout blanks the screen but does not turn off power to the monitor, as it should. This was fixed for `vcgencmd` to turn off the monitor (link below). The same fix needs to be applied to `dpms` commands. 

https://github.com/raspberrypi/firmware/issues/1224

**Description**
When the `dpms` time expires, a blank screen is displayed but should be powered off.

**To reproduce**
Open a terminal window, type in `xset dpms force off`, notice the screen is still illuminated and not turned off.

**Expected behaviour**
The monitor shuts down using Energy Saver.

**Actual behaviour**
Screen blanks, blue monitor light remains illuminated, screen never turns off nor red light illuminated indicted monitor is powered off.

**System**
https://pastebin.com/PNnyw303

* Which model of Raspberry Pi: RPi4B
* Which OS and version:
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage5
* Which firmware version:
Sep 30 2019 14:37:51 
Copyright (c) 2012 Broadcom
version b1477501b8675de389acd184c2b7e7c4b2e39e4a (clean) (release) (start)
* Which kernel version:
Linux FrankRaspi 4.19.75-v7l+ #1271 SMP Mon Sep 30 13:51:15 BST 2019 armv7l GNU/Linux
Graphics Driver: dtoverlay=vc4-fkms-v3d

JamesH65, 2019-11-05:
I think this should also be fixed by https://github.com/raspberrypi/linux/pull/3289

Please test and confirm (change currently in rpi-update)

frankgould, 2019-11-05:
This appears to be working great on Buster. Thank you @JamesH65 !

closed 2019-11-05 by frankgould