raspberrypi/firmware #1261: Regression: NetBSD no longer boots since "Port file loading revamp to master"

skrll, 2019-10-04:
Since e2e17cd035fb8cd1a0b74d35d6615d7e267e408b NetBSD no longer boots.

nick@zoom:/media/nick/NETBSD$ ls
 bcm2837-rpi-3-b.dtb   bcm2837-rpi-3-b-plus.dtb   bootcode.bin   cmdline.txt   config.txt   dtb   EFI   fixup_cd.dat   fixup.dat   ld.dbg_so   libc.so   LICENCE.broadcom   netbsd.bak.img   netbsd.img   old   start_cd.elf   start.elf  'System Volume Information'
nick@zoom:/media/nick/NETBSD$ cat cmdline.txt 
root=ld0a c0nsole=fb -x
nick@zoom:/media/nick/NETBSD$ cat config.txt
arm_64bit=1
kernel=netbsd.img
kernel_address=0x200000
enable_uart=1
force_turbo=0
nick@zoom:/media/nick/NETBSD$

skrll, 2019-10-04:
The netbsd.img file is available from http://www.netbsd.org/~skrll/rpi-aarch64-netbsd.img

skrll, 2019-10-04:
we track mainline dts - is there a file naming issue?

pelwell, 2019-10-04:
Try with `upstream_kernel=1` - that commit dropped support for the kernel trailer, so the firmware has to be told to use the upstream DTB naming.
 
If that doesn't work, try naming the DTB explicitly with `device_tree=...`.  You can also add `uart_2ndstage=1` to enable logging over the UART from the firmware.

skrll, 2019-10-04:
Thanks for the hint.
upstream_kernel=1 makes it boot.
I guess I can drop the kernel trailer and use upstream_kerneel going forward - is this the expectation from the firmware now?

pelwell, 2019-10-04:
Yes it is - all support for trailers has been removed from the firmware and the rpi-5.3.y tree. The logic behind the change is that trailers were a downstream invention that were never going to be adopted upstream, so expecting builds of the upstream kernel to use them was just wrong.

skrll, 2019-10-04:
ok, thanks

skrll, 2019-10-06:
I tried `upstream_kernel=1` with a kernel without a trailer and it doesn't boot. The `uart_2ndstage=1` output is the same with trailer and no trailer (except file sizes), e.g.

```
Raspberry Pi Bootcode
Read File: config.txt, 120
Read File: start.elf, 2877988 (bytes)
Read File: fixup.dat, 6736 (bytes)
MESS:00:00:01.190691:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.194915:0: brfs: File read: 120 bytes
MESS:00:00:01.275924:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.280172:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:01.484844:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:01.492276:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:01.498299:0: *** Restart logging
MESS:00:00:01.502187:0: brfs: File read: 120 bytes
MESS:00:00:01.537148:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:01.546999:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:01.550961:0: Read command line from file 'cmdline.txt':
MESS:00:00:01.556851:0: 'root=ld0a console=fb'
MESS:00:00:01.565591:0: brfs: File read: 21 bytes
MESS:00:00:01.574942:0: brfs: File read: /mfs/sd/bcm2837-rpi-3-b.dtb
MESS:00:00:01.579603:0: Loading 'bcm2837-rpi-3-b.dtb' to 0x100 size 0x56b2
MESS:00:00:01.682542:0: brfs: File read: 22194 bytes
MESS:00:00:01.686595:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:03.460465:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:03.490586:0: brfs: File read: 120 bytes
MESS:00:00:04.252466:0: brfs: File read: /mfs/sd/netbsd.img
MESS:00:00:04.256345:0: Loading 'netbsd.img' to 0x200000 size 0xbdbab4
MESS:00:00:04.262603:0: Device tree loaded to 0x2effaa00 (size 0x554c)
MESS:00:00:04.270110:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:04.276557:0: uart: Baud rate change done...
MESS:00:00:04.279987:0: uart: Baud rate

```

pelwell, 2019-10-06:
Is your config.txt the same as above, just with upstream_kernel=1?

skrll, 2019-10-06:
yeah

skrll, 2019-12-17:
Recent firmware updates work

closed 2019-12-17 by skrll