raspberrypi/firmware #1263: Camera exposure_speed decreasing each frame

twVolc, 2019-10-04:
This is the same problem highlighted by [https://github.com/waveform80/picamera/issues/475](url), but I thought I should post here as recommended by 6by9.

I have a very simple script to record images

```
import picamera
import time

with picamera.PiCamera() as cam:
    cam.framerate = 1
    cam.shutter_speed = 1000000
    cam.exposure_mode = 'off'
    cam.analog_gain = 1  # I'm using a forked version of picamera (https://github.com/rwb27/picamera) which can set analog gain, but I doubt this is important. I experience the same issue if I remove this line
    cam.shutter_speed = 1000000
    time.sleep(3)
    
    for i in range(100):
        cam.capture('test_{}ag_{}ss.jpg'.format(cam.analog_gain, cam.exposure_speed), bayer=True, format='jpeg')
```
Consecutive images will have reduced exposure speeds. 
My current fix is to just set shutter_speed each time before acquiring, but it would be great if this wasn't necessary

I'm on a Pi 3 B+, running Buster.
Linux pi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
Sep 24 2019 17:39:25 
Copyright (c) 2012 Broadcom
version 6820edeee4ef3891b95fc01cf02a7abd7ca52f17 (clean) (release) (start_x)

Apologies if this hasn't been raised correctly, I'm pretty inexperienced with github.

6by9, 2019-10-04:
Which camera module? OV5647 V1.3 (or clone thereof), or Sony IMX219 V2.1?

https://github.com/waveform80/picamera/issues/475#issuecomment-391756255 says the issue on the IMX219 was fixed May 2018.

twVolc, 2019-10-05:
Ah yes sorry, I'm working with the v1.3 Omnivision chip. I know that's now discontinued but for my application I can't move to the Sony sensor. Is there likely to be a quick fix for the v1.3? If not, it's not the end of the world re-assigning camera.shutter_speed before each capture.