raspberrypi/firmware #1264: The size of the .git directory in this repository has ballooned from 127 MB when I cloned it a month ago to about 12GB when I cloned it just now. I'm not sure which commit was the culprit.

lancesimms, 2019-10-04:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

lancesimms, 2019-10-04:
The size of the .git directory in this repository has ballooned from 127 MB when I cloned it a month ago to about 12GB when I cloned it just now. I'm not sure which commit was the culprit, but this cannot be right given the small footprint of the files being tracked.

bit-bash, 2020-01-02:
Not sure if this helps, but running this huge command (discovered on the interwebs):

`git rev-list --objects --all | git cat-file --batch-check='%(objecttype) %(objectname) %(objectsize) %(rest)' | sed -n 's/^blob //p' | sort --numeric-sort --key=2 | cut -c 1-12,41- | $(command -v gnumfmt || echo numfmt) --field=2 --to=iec-i --suffix=B --padding=7 --round=nearest`

Show a list of all the git blobs, sorted by size.  I've attached a summary of all files > 4MiB.  There are many .img files, and each time a new one is committed, the repo will grow by that size (compression isn't going to help much).

There are 1088713 blobs in the repo, and 1518 of them are > 4MiB.

Also, is it possible the firmware repo has been big for a while and could your recent clone have been a shallow clone ( git clone ... --depth 1)?

[bigstuff4.zip](https://github.com/raspberrypi/firmware/files/4017567/bigstuff4.zip)

popcornmix, 2020-01-07:
There have been 1169 commits to this repo (typically new firmware/kernel).
I'm seeing 13G which equates to about 11M per commit.
That seems perfectly reasonably when there are multiple firmwares and kernels (of a few MB each).

The size of this repo didn't balloon recently. It pretty much grows linearly with time.
If you just want a snapshot then download the zip file from gitub or use ``git clone --depth 1`` to just get the latest. If you want the whole history of over 1000 firmware releases then expect it to be big.

closed 2020-01-07 by popcornmix

lancesimms, 2020-01-08:
Ok, I do not know why the initial clone was so small. I did not use any special directives or anything like that. If I figure out why there was the abrupt change, I'll let you know.