raspberrypi/firmware #1266: [gpio4=0] filter on PI 3

juergh, 2019-10-07:
Hi,

I'm trying to boot the 64-bit kernel on a PI 3 using gpio4, i.e.:
```
[gpio4=0]
arm_64bit=1
```
I've shorted pin 7 to ground but the firmware still loads the 32-bit kernel. This works just fine on a PI 4. What am I missing?

```
pi@raspi-3b-plus:~ $ vcgencmd version
Sep 30 2019 14:40:39 
Copyright (c) 2012 Broadcom
version 6820edeee4ef3891b95fc01cf02a7abd7ca52f17 (clean) (release) (start)

pi@raspi-3b-plus:~ $ sudo vcdbg log msg
001398.797: brfs: File read: /mfs/sd/config.txt
001399.861: brfs: File read: 1854 bytes
001410.539: HDMI:EDID error reading EDID block 0 attempt 0
001411.842: HDMI:EDID error reading EDID block 0 attempt 1
001413.135: HDMI:EDID error reading EDID block 0 attempt 2
001414.430: HDMI:EDID error reading EDID block 0 attempt 3
001415.723: HDMI:EDID error reading EDID block 0 attempt 4
001417.018: HDMI:EDID error reading EDID block 0 attempt 5
001418.311: HDMI:EDID error reading EDID block 0 attempt 6
001419.606: HDMI:EDID error reading EDID block 0 attempt 7
001420.899: HDMI:EDID error reading EDID block 0 attempt 8
001422.194: HDMI:EDID error reading EDID block 0 attempt 9
001423.251: HDMI:EDID giving up on reading EDID block 0
001423.609: HDMI:EDID error reading EDID block 0 attempt 0
001424.908: HDMI:EDID error reading EDID block 0 attempt 1
001426.206: HDMI:EDID error reading EDID block 0 attempt 2
001427.505: HDMI:EDID error reading EDID block 0 attempt 3
001428.803: HDMI:EDID error reading EDID block 0 attempt 4
001430.101: HDMI:EDID error reading EDID block 0 attempt 5
001431.399: HDMI:EDID error reading EDID block 0 attempt 6
001432.698: HDMI:EDID error reading EDID block 0 attempt 7
001433.997: HDMI:EDID error reading EDID block 0 attempt 8
001435.296: HDMI:EDID error reading EDID block 0 attempt 9
001436.359: HDMI:EDID giving up on reading EDID block 0
001453.927: brfs: File read: /mfs/sd/config.txt
001674.933: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
001677.030: *** Restart logging
001677.097: brfs: File read: 1854 bytes
001677.930: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
001679.239: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
001680.542: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
001681.845: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
001683.148: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
001684.451: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
001685.754: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
001687.057: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
001688.360: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
001689.663: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
001690.730: hdmi: HDMI:EDID giving up on reading EDID block 0
001691.097: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
001692.396: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
001693.693: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
001694.989: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
001696.285: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
001697.581: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
001698.877: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
001700.173: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
001701.469: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
001702.765: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
001703.824: hdmi: HDMI:EDID giving up on reading EDID block 0
001703.904: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001703.976: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001707.651: brfs: File read: /mfs/sd/cmdline.txt
001707.744: Read command line from file 'cmdline.txt':
001707.805: 'console=serial0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait'
001712.456: brfs: File read: 121 bytes
001716.454: brfs: File read: /mfs/sd/bcm2710-rpi-3-b-plus.dtb
001716.522: Loading 'bcm2710-rpi-3-b-plus.dtb' to 0x100 size 0x69ae
001870.052: brfs: File read: 27054 bytes
001874.243: brfs: File read: /mfs/sd/config.txt
001874.879: dtparam: audio=on
003609.950: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
003711.058: brfs: File read: 1854 bytes
004078.993: brfs: File read: /mfs/sd/kernel7.img
004079.059: Loading 'kernel7.img' to 0x8000 size 0x510740
004079.133: Device tree loaded to 0x2eff9200 (size 0x6dc1)
004081.385: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
005402.059: vchiq_core: vchiq_init_state: slot_zero = 0xfad80000, is_master = 1
010672.067: brfs: File read: 5310272 bytes

pi@raspi-3b-plus:~ $ raspi-gpio get 4
GPIO 4: level=0 fsel=0 func=INPUT

pi@raspi-3b-plus:~ $ cat /proc/device-tree/model 
Raspberry Pi 3 Model B Plus Rev 1.3
```

pelwell, 2019-10-07:
I also have a _switch-of-64bitness_, so it should work. I suspect you have some other filter active, but post your config.txt (or just the bits you've added).

juergh, 2019-10-07:
```
dtparam=audio=on

[pi4]
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[gpio4=0]
arm_64bit=1

[all]
enable_uart=1
```

pelwell, 2019-10-07:
Yep - thought so. The conditional filters are quite powerful, but with that power comes a few gotchas.

There are a number of orthogonal/independent sets of filters - model, edid and gpio - all of which have to be true for the settings to be used. By default, all filters are set to true, but when the parser finds a filter setting it updates the relevant filter based on the system state. On a Pi 4 all is well, but on Pi 3 by the time it reaches `[gpio4=0]` the model filter is still false because it isn't a Pi 4. There isn't a way to always set just the model filter to true, so the solution is to use `[all]` liberally and re-apply any filters you wish to remain active.

juergh, 2019-10-07:
Hmm... I've added a [pi3] section before the [pi4] section so the model flag should be true when the parser reaches the gpio4 filter, no? But that didn't make a difference so I must not understand your explanation correctly. Can you provide a config.txt that sets arm_64bit=1 when gpio4=0 for all PIs that support the 64-bit kernel?

pelwell, 2019-10-07:
```
dtparam=audio=on

[pi4]
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
[gpio4=0]
arm_64bit=1

[all]
enable_uart=1
``` ?

juergh, 2019-10-07:
Works. Thanks!

pelwell, 2019-10-07:
When it comes to organising larger config files it probably makes more sense to put the blank line (if there is one) after the first `[all]` to mark the end of the `pi4` section rather than the start of the `gpio4=0` section, but that's a cosmetic issue.

closed 2019-10-07 by pelwell