raspberrypi/firmware #1267: Raspberry Pi 3B+ Not boot with Kernel 4.19.76

pheiduck, 2019-10-08:
Can‘t generate logs for this.
After I update to 4.19.76 with rpi-update & reboot the RPi 3B+ doesn’t boot 
Directory is has size 256M /boot

jgarvey955, 2019-10-08:
I am getting...

mmc1: Reset 0x1 never completed

on both my 3B and 3B+ but my Pi4's boot fine with the same SDCard swapped through out.

pelwell, 2019-10-08:
The same rpi-update release boots for me without any problems. What's in your config.txt?

pheiduck, 2019-10-08:
`# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
hdmi_force_hotplug=1
dtparam=i2c_arm=on
dtparam=spi=on
enable_uart=1
dtoverlay=tft35a:rotate=90
program_usb_boot_mode=1`

pheiduck, 2019-10-08:
No I don’t have a Serial Port Cable

pelwell, 2019-10-08:
And my first question?

pelwell, 2019-10-08:
It not clear what could have gone wrong, but we might be able to use the 64-bit kernel as a get-out-of-jail card. Try adding `arm_64bit=1` to config.txt - the worst it can do is not help.

jgarvey955, 2019-10-08:
> It not clear what could have gone wrong, but we might be able to use the 64-bit kernel as a get-out-of-jail card. Try adding `arm_64bit=1` to config.txt - the worst it can do is not help.

this did not help my issue:

mmc1: Reset 0x1 never completed

pheiduck, 2019-10-08:
I don't want to do this right now, because I have troubleshoot this problem here last night and have set up my Pi again. Have there a lot things installed pihole and something like that

pheiduck, 2019-10-08:
only test device that I have is the Raspbberry Pi2 Modell B

pelwell, 2019-10-08:
@jgarvey955 That's a different problem. You'll need to post your config.txt as well.

mmc1 should only affect WiFi. If it fails to boot:
* Can you upload a photo of the boot screen at the point of failure?
* Try adding `dtoverlay=disable-wifi` to config.txt.

If it boots but has no WiFi, post the output of `dmesg`.

pheiduck, 2019-10-08:
I know that, but since yesterday I haven't any problems with `rip-update`

pelwell, 2019-10-08:
Most people who play with loaded guns walk away unscathed.

jgarvey955, 2019-10-08:
> @jgarvey955 That's a different problem. You'll need to post your config.txt as well.
>
> mmc1 should only affect WiFi. If it fails to boot:
>
> * Can you upload a photo of the boot screen at the point of [...]

adding the overlay did the trick. Thank you,

polaco1782, 2019-10-10:
> @jgarvey955 That's a different problem. You'll need to post your config.txt as well.
>
> mmc1 should only affect WiFi. If it fails to boot:
>
>     * Can you upload a photo of the boot screen at the point [...]

I can confirm it works for me too. Mine hangs soon after mmc1: Reset 0x1 never completed.

ikalkov, 2019-10-10:
Seems to be a firmware issue. My 4.9.80-v7+ kernel was working fine on RPi 3B until I manually updated the firmware binaries (fixup.dat, fixup_x.dat, start.elf, start_x.elf). Now I get the same problem during boot:

```
[    3.442987] sdhost: log_buf @ ded08000 (ded08000)
[    3.492392] mmc0: sdhost-bcm2835 loaded - DMA disabled
[    3.599759] mmc1: Reset 0x1 never completed.
[    3.604098] mmc-bcm2835 3f300000.mmc: mmc_debug:0 mmc_debug2:0
[    3.610021] mmc-bcm2835 3f300000.mmc: DMA channel allocated
```

pelwell, 2019-10-10:
`4.9.80-v7+` - seriously?

Do you have any idea how old the previous working firmware was? Have you tried rolling back to find when it starts to work?

pheiduck, 2019-10-10:
Stable Update Works for me:)

ikalkov, 2019-10-10:
The last commit (f8a89246a793911701b9f255bb6ba34a0b93479e) breaks it, Previous one (a16470ad47c0ad66d5c98d98e08e49cd148c8fc0) is working just fine.

pheiduck, 2019-10-10:
> The last commit ([f8a8924](https://github.com/raspberrypi/firmware/commit/f8a89246a793911701b9f255bb6ba34a0b93479e)) breaks it, Previous one ([a16470a](https://github.com/raspberrypi/firmware/commit/a16470ad47c0ad66d5c98d98e08e49cd148c8fc0)) is working just fine.
> 
> EDIT: I've tested a bunch of commits between 1st of September and today - all previous commits are OK,

Can confirm this

pelwell, 2019-10-10:
OK- I've got a theory as to what the problem might be, and reverting the bit of code in question should be harmless, so provided it passes the code review it could appear in an update in the next few days.

Thank you for your co-operation and patience.

popcornmix, 2019-10-11:
Latest rpi-update firmware may have a fix for this.

pelwell, 2020-01-10:
Closing for lack of activity.

closed 2020-01-10 by pelwell