raspberrypi/firmware #1268: Upgrade to latest stable version

akicker, 2019-10-08:
I want to upgrade to the latest stable version !
https://github.com/raspberrypi/firmware/blob/stable/extra/uname_string7l
shows: Linux version **4.19.75**-v7l+ 
but apt full-upgrade remains at Linux version **4.19.66** ?

XECDesign, 2019-10-08:
See https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=253259&p=1548623#p1548559

akicker, 2019-10-08:
Hi I dont think boot size is the reason - df shows:
/dev/mmcblk0p1  253M     40M  213M   16% /boot
and apt upgrade does not show any errors!

XECDesign, 2019-10-08:
To clarify, the package is not in the default repo until a few more people have tested it.

akicker, 2019-10-08:
OK got it, 
I am monitoring dayly the latest stable from https://github.com/raspberrypi/firmware/blob/stable/extra/uname_string7l, 
to find out if a new stable update is available,
so I should monitor the default repo, but where to find it?

closed 2020-02-17 by akicker