raspberrypi/firmware #1269: MMAL_PARAMETER_SUPPORTED_ENCODINGS is returning an empty list on rpi4

djthorpe, 2019-10-08:
I have both a Raspberry Pi 3 and Raspberry Pi 4, and I am not getting the list of supported encodings back on all MMAL components on the Raspberry Pi 4 - only an empty array.

**To reproduce**
Download or clone the repo:
https://github.com/Terminus-IMRC/mmal_list_supported_encodings

```bash
bash%  git clone git@github.com:Terminus-IMRC/mmal_list_supported_encodings.git
bash% cd mmal_list_supported_encodings
bash% make
bash% ./mmal_list_supported_encodings vc.ril.video_render
```

The output on a rpi3 is:
```
* Input port 0
	* Minimum number of buffers: 2
	* Minimum size of buffers: 15360
	* Minimum alignment of buffers: 0
	* Number of buffers recommended: 3
	* Size of buffers recommended: 15360
	* Capability: allocation event_format_change
	* Default encoding: I420
	* Supported encodings:
		 1. I420
		 2. I42S
		 3. YV12
		 4. NV12
		 5. NV21
		 6. I422
		 7. I422
		 8. SAND
		 9. RGB2
		10. RGB3
		11. BGR3
		12. BGRA
		13. RGBA
```

The output on an rpi4 is:
```
* Input port 0
	* Minimum number of buffers: 2
	* Minimum size of buffers: 15360
	* Minimum alignment of buffers: 0
	* Number of buffers recommended: 3
	* Size of buffers recommended: 15360
	* Capability: allocation event_format_change
	* Default encoding: I420
```

The function being called in both cases is

```c
hdrp = mmal_port_parameter_alloc_get(port,MMAL_PARAMETER_SUPPORTED_ENCODINGS, 0, &status);
```

I would expect to see "Supported encodings" on the rpi4 as well. This problem affects all components.

**System**

* Which model of Raspberry Pi? rpi4
* Raspberry Pi reference 2019-06-20
* Firmware: Jun 20 2019 16:06:32 version 407b1da8fa3d1a7108cb1d250f5064a3420d2b7d (clean) (release) (start_x)
* Kernel Linux rpi4 4.19.50-v7l+ #895 SMP Thu Jun 20 16:03:42 BST 2019 armv7l GNU/Linux

Terminus-IMRC, 2019-10-08:
Your kernel may be too old. Try updating it with `apt update && apt upgrade` or so. Kernels until 2019 September 25 seem to have a bug, which prevents users from querying supported encodings https://github.com/raspberrypi/linux/commit/a00177eaa6d5c274664b6c8c86318734657d657e , I've not yet tested it though.

djthorpe, 2019-10-08:
Thanks for this information, Yukimasa and for your diagnosis tool! Let me try and upgrade by kernel and I will post here if that was successful.

6by9, 2019-10-08:
@Terminus-IMRC That commit only applies to the V4L2 drivers, not to userland using MMAL.

However the V4L2 drivers are using exactly that mechanism to get the list of supported formats for each of the codecs, and that is working fine.
I'm suspecting an issue in ```mmal_port_parameter_alloc_get```, but that's only a guess.

6by9, 2019-10-08:
@djthorpe What firmware version are you running on the Pi3?
```mmal_port_parameter_alloc_get``` is being told a smaller value than expected for the array of encodings, so when it allocs the array and rerequests it still gets told ENOSPC (ie array too small).
There are no differences at all in the code implementing that between current Pi3 and Pi4 firmwares, therefore I suspect it's actually a common change to both that has caused the regression.
I've just tried it on a Pi3 with latest firmware and see the same issue.

If you define the structure to be big enough in the first place (which is what the V4L2 drivers do), then the call works fine.

6by9, 2019-10-08:
OK, it is common. The "firmware: mmal_ril: List all 4 Bayer orders if the IL component says it supports Bayer" part of https://github.com/Hexxeh/rpi-firmware/commit/a79abb5d0b3bf9f20e39c12c92f5890ac3eaf2dc
Incrementing the count to be reported was being done in the wrong place.

djthorpe, 2019-10-08:
Yes, you're right, if I change the line to the following it works fine:

```c
#define MAX_ENCODINGS 20
hdrp = mmal_port_parameter_alloc_get(port,
            MMAL_PARAMETER_SUPPORTED_ENCODINGS, MAX_ENCODINGS * sizeof(uint32_t), &status);
```
I will close this issue report.

closed 2019-10-08 by djthorpe

6by9, 2019-10-08:
No, you're totally correct with the report, and I have a firmware fix.
```mmal_port_parameter_alloc_get``` is supposed to allocate a correctly sized buffer for the parameter, and it doesn't. That's due to a firmware bug which wasn't incrementing the counter once it hit the buffer size.

The change linked had to add a one to many mapping between the IL Bayer formats which don't specify Bayer order, to the MMAL ones which do. In doing so I moved the increment of the count that reports back on ENOSPC to the wrong place, it didn't get incremented once it hit the size of the buffer, and so the value reported back was too small.
Silly one, but they happen.

popcornmix, 2019-10-11:
Latest rpi-update firmware may have a fix for this.