raspberrypi/firmware #1270: vl805fw question

acostach, 2019-10-09:
Hi, sorry if this is not the right place to ask about the vl805 firmware upgrade.

There is this forum topic https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=250990 where a new firmware binary was provided, but for testing only, via google drive.

My question is if that firmware will be publicly released, and if so, in what repository will it be available? Will it be available as a debian package only, vl805fw, like it currently is?
Also, wondering if you have any estimated planned release date?

Thank you

acostach, 2019-10-11:
Hi, any info on this? Was also wondering if vl805 tool sources would be published anytime soon so it can be built with yocto. The debian repos seem to have a dinamically linked binary which can't be run outside of raspbian.. Any answer is highly appreciated. Thanks

timg236, 2019-10-11:
Any supported updates to the VLI will be via the rpi-eeprom package we haven't decided on the update mechanism yet which won't necessarily be the vl805 tool.
Sorry no timescales yet, maybe a beta in the next few weeks

acostach, 2019-10-14:
Thank you very much for your answer!

closed 2019-10-16 by pelwell