raspberrypi/firmware #1272: 64 bit size-cells breaks RPi 4 boot

lategoodbye, 2019-10-11:
**Describe the bug**
During upstreaming process of Raspberry Pi 4 support, Florian Fanelli complained about the too same size-cells definition in the root node. AFAIK the BCM2711 is able to address 16 GB RAM so we should extend the size-cell from 1 to 2 (64 bit). Unfortunately this breaks boot of the Raspberry Pi 4.

https://marc.info/?l=linux-arm-kernel&m=157036853723973&w=2

I decided to keep the working entry for initial Raspberry Pi 4 support in Linux 5.5. But it would be nice to get this fixed, so we can also fix the upstream device-tree.

**To reproduce**
Adjust size-cell to 2 and add another 0 to the reg property of the memory node in the DTS.

**Expected behaviour**
Raspberry Pi 4 boot

**Actual behaviour**
Bootloader or firmware stuck during boot

**System**

* Which model of Raspberry Pi? Raspberry Pi 4

pelwell, 2019-10-11:
The firmware currently has a number of templates for the memory node, and it selects between them based on the architecture. It's a fairly simple task to generate them based on the value of `#size-cells`.

pelwell, 2019-10-25:
The latest firmware release should generate memory nodes correctly for the DTB (provided the cells values are 1 or 2).

lategoodbye, 2019-10-26:
I'll try to test this soon and close this

closed 2019-12-05 by lategoodbye