raspberrypi/firmware #1273: vcgencmd not able to query the initramfs config option

Melanie-BB, 2019-10-11:
The command `vcgencmd get_config initramfs` returns `initramfs is unknown`.

The commands `vcgencmd get_config ramfsfile` and `vcgencmd get_config ramfsaddr` are not affected by this issue and return the expected values.

As described in [1] the `initramfs` option specifies both the `ramfsfile` and `ramfsaddr` in one parameter.

As it seems that most users prefer the `initramfs` option in their `config.txt` I think it would make sense to add the ability to query this option directly.

Not sure whether this change would affect the firmware or only the userland-tool, so please advise if this is not the correct location for this issue.

[1] [https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md](https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md)

pelwell, 2019-10-11:
`vcgencmd get_config` runs a common lookup for all string settings. `initramfs` is a command/directive rather than a setting, hence the failure to read back the string value. In my opinion it ain't broke, so I doubt we'll be fixing it.

Melanie-BB, 2019-10-16:
Agreed, closing issue as this is considered normal behaviour.

closed 2019-10-16 by Melanie-BB