raspberrypi/firmware #1274: gamma and brightness control not implemented?

myier, 2019-10-11:
I'd like to use xrandr to change the colour of the display and the brightness below what the screen offers, for example:
    $ xrandr --output HDMI-1 --brightness 0.7
    $ xrandr --output HDMI-1 --gamma 2:2:2
but these commands do nothing and give no error. The redshift [tool](http://jonls.dk/redshift/) (package redshift on debian) also uses methods to get a more red and less bright display, same result.

**System**
Raspberry Pi 4 with raspbian, experimental driver enabled (dtoverlay=vc4-fkms-v3d in /boot/config.txt).

# cat /etc/rpi-issue
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

# vcgencmd version
Sep 24 2019 17:34:30 
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start)

# uname -a
Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

6by9, 2019-10-14:
Having had a quick look at the source for xrandr, brightness also alters the gamma curve, so you've only actually requested one feature.

Implementing gamma control requires suitable hardware in the composition pipe in order to operate. There are some parts of the display pipeline that can be switched between the various output channels - I'd need to check whether gamma falls into that category or not.

Tweaking gamma is not a critical feature of the display pipeline, therefore it won't be a high priority to add the plumbing.

myier, 2019-10-14:
Thank you for taking the time to look at it.

For people using the computer at night it's an important feature, but I understand that there are much less people caring about their eyes compared to those who want to have opengl or other features :)

Askmewho, 2019-10-29:
Hope someone get it implemented, because it not only affect desktop experience... while gaming, some game are unplayable because of the gamma... for example, open jedy academy... ive recently compile it and the brightness its unresponsive, while on rpi3 under mesa that didnt happened

Thovthe, 2019-12-13:
+1 I would use this feature

john2exonets, 2019-12-27:
+1 ...new RP4's are tough on the eyes at night!!  f.lux, redshift, xrandr....none of them are working :(

pjh64, 2019-12-29:
I'm bumping this since the situation of not being able to tweak gamma still hasn't changed.
Like john2exonets stated it is really tough on the eyes.

6by9, 2019-12-29:
So we now have a grand total of 5 people asking for this feature. Compared to the number of devices sold and priorities, the situation still hasn't really changed since my comment on the 14th October

> Tweaking gamma is not a critical feature of the display pipeline, therefore it won't be a high priority to add the plumbing.

Bumps (particularly only two days after the last bump post) add nothing to the discussion, but do spam those involved. If you're really that bothered, then adding emojis shows your interest without spamming everyone.

pjh64, 2019-12-29:
I wouldn't (have a reason to) be that bothered if that part wasn't proprietary. Still I don't think this is the place to discuss that OSS vs. proprietary matter on a fundamental scale since this wouldn't do anything about it.

Yet I still felt the urge to state/underline that it is indeed tough on the eyes. So this is really not ment to just spam on this issue but rather the only thing a user could do about the issues ergonomic nature in the context of closed source software.

I'm gonna add this via emoji next time but didn't think of adding emoji being an actual thing rather than a stylistic choice.

JamesH65, 2019-12-29:
Yes, it would be a useful feature to have, BUT, its not the only option. All monitors have a brightness setting, and most some ability to change colour balance. You can always use that. No need for access to closed source firmware (and TBH, if people with 10 years experience working on the firmware cannot quickly add it, then someone out there isn't going to be able to quickly add it either, even with source access)

Askmewho, 2019-12-29:
I don't know why rpi foundation work with broadcom SOCs if they are unable (or just won't) to supply key source code of their SOC firmware.

pjh64, 2019-12-29:
@Askmewho I think the rather strange situation of having a RPI with important features being kept proprietary is the price of keeping the hardware that affordable. Maybe it is even possible to release a fully open board, but certainly not for the rather low cost you get an RPI for.

Take the Pyra/Pandora for example. AFAIK it is mostly open source, but expensive as hell.

Askmewho, 2019-12-29:
You are right @pjh64 , but how open/closed its rk3399 firmware for example?

JamesH65, 2019-12-30:
If you can suggest a SoC with the same feature set, and an entire development team that knows the SoC inside out, and a SoC manufacturer who is willing to talk to us about what features we need and supply them, then that would be possible. But I suspect you won't be able to find one that fits the bill. 

Broadcom fits the bill. And of course, Eben works for Broadcom, so has a huge amount of input on what goes in the chips. 

We are also trying to move away from the GPU firmware and provide all features accessible from ARM space. But it's a long process because of the huge amount of work that the GPU firmware actually does (I think it's about a millions line of code or something bonkers like that) The KMS driver is a huge step towards that, but that still leaves the codecs, the ISP, and probably a load of other stuff I cannot be bothered to think about right now.

JamesH65, 2020-01-02:
Just been reminded - the KMS driver (available now on Pi3, in a few months on Pi4) is fully open source and will allow you to directly access the underlying systems to add your own gamma support, if it's not in there already by default. Note, not talking about the FKMS driver which still uses the GPU firmware.

Askmewho, 2020-01-02:
Ok, good to hear about that!

smarks86, 2020-04-16:
+1 for the issue. I know it isn't top priority and thanks for the hard work.

Askmewho, 2020-05-31:
@JamesH65  I am confused. It's not full kms available on mainline?? I switched to mainline and it seems it's not..

popcornmix, 2020-05-31:
kms support is in upstream kernel for pi0-3. Pi4 support has been submitted but not fully merged yet.
https://lore.kernel.org/dri-devel/cover.6c896ace9a5a7840e9cec008b553cbb004ca1f91.1582533919.git-series.maxime@cerno.tech/T/#m0275b55e1518be1fb2154d3c95e13c1b7de1f347

Askmewho, 2020-05-31:
ok, yeah. I tried the overlay for full KMS on 5.4 kernel and it does brake my sound with pulseaudio, and while extremely unstable, it did not add gamma correction capabilities neither as far as I can tell. Thanks

6by9, 2020-05-31:
Pi4 uses a PWL to configure gamma curves. DRM uses a lookup table. You're therefore looking at fun and games around curve fitting if you want vaguely accurate results, and that becomes a faff. It's currently explicitly disabled on vc4-kms-v3d for Pi4.

Implementing it is still on the list of tasks, but there are higher priorities.

pmarreck, 2020-08-24:
+1 to making standard Linux utilities that change gamma for eye/sleep protection (Ubuntu's "Night Light", the "gammy" utility, and many more) work on an RPi[4]. As a dev with a sleep disorder who finally has access to a desktop-class ARM computer (I'm currently actually running Ubuntu Server with a GUI and GPU acceleration thanks to https://github.com/wimpysworld/desktopify), this would be huge.

NoSuck, 2020-11-21:
It's like staring into a fluorescent light bulb.

Askmewho, 2020-11-28:
> If you can suggest a SoC with the same feature set, and an entire development team that knows the SoC inside out, and a SoC manufacturer who is willing to talk to us about what features we need and [...]

I forget to answer this. Yes: RK3399

Askmewho, 2021-05-18:
> com fits the bill. And of course, Eben works for Broadcom, 

he worked at broadcom until 2019 then

RaingodSpires, 2021-07-01:
> So we now have a grand total of 5 people asking for this feature. Compared to the number of devices sold and priorities, the situation still hasn't really changed since my comment on the 14th October
> 

I registered on purpose to add me to the request. Maybe we only are a few people requesting that because of:
1. a lot of people doesn't know the existence of that programs
2. other people doesn't have or know, account here.

Using the pi in a old television at night even with ambiental light, with the white backgrounds of websites is a torture.

RaingodSpires, 2021-07-01:
For those that would have the redshift or similar, at the moment maybe you will interest an alternative. In the xfce4 settings there are a "Brightness" option. The process is "pibright", it works. Not sets the brightness of the display itself, but it darks the signal.

So I made a launcher in the notification area, If you start typing pibri... the panel itself suggests to insert the brightness icon. So this at the moment is the best solution for me. I hope you too.

6by9, 2021-07-02:
Gamma control is nearly there for vc4-kms-v3d.
https://github.com/mripard/rpi-linux/tree/rpi/vc4-gamma still needs a few clean ups before it'll be merged though.

There is a restriction in the hardware that means that the pipeline has to be shutdown and restarted on a gamma change, so the display will blank. Not much we can do about that one unfortunately.

I wasn't anticipating making changes to the firmware to support gamma, and due to the requirement for a pipeline restart I suspect that it is more effort and risk than it is worth.

slaminger, 2021-10-02:
No brightness control or sleep function on raspberry pi? What's the deal?

pelwell, 2021-10-02:
The deal is that gamma control is high on the list of things to do but not completed yet - I thought @6by9 made that clear.

slaminger, 2021-10-02:
> The deal is that gamma control is high on the list of things to do but not completed yet - I thought @6by9 made that clear.

I was more asking why the pi foundation didn't have this and sleep supported a long time ago.

pmarreck, 2021-10-02:
I guess the danger in making a “desktop class” Pi is that suddenly people expect desktop-class things.

As is the M.O. with open source, I’m sure someone would be grateful if you wanted to take a crack at it with a PR

Frankiesayscode, 2021-11-29:
Fully feeling this at the moment.

Git hub is a beautifully clean site in the day, but redhift/f.lux etc are all essential for my sleep on other devices, and hopping over here has on the pi with monitor done my insomnia no favours.

Absolutely loving the 4. It really does feel like a desktop device, but will have to vnc in from an android/ios device until there is a native resolution for pi-os or variants. Small screens make me squinty but poor sleep is probably worse for my eyes.

Please bump this up as a priority.

If anyone does come up with a package once gamma is wihin contro... I propose camomile as a name :)

6by9, 2021-11-29:
It's implemented for the full KMS driver (not FKMS)
#4435, #4682, #4664.

As the KMS driver has been adopted by default under Bullseye, I don't see any point in looking to implement it under FKMS.

myier, 2021-12-17:
Thank you 6by9, but could you explain how to use it?
I'm running KMS on bullseye, and neither xrandr commands that I gave at the top or redshift have any effect on the colour of the display. They still do nothing with no error.
How did you manage to change it?

6by9, 2021-12-20:
Use ```sudo rpi-update``` to update to the absolute latest kernel (do not do this on a critical Pi - there is a small chance of regressions).
apt will be updated soon once the kernel/firmware has undergone more testing.

mcgrealife, 2022-01-23:
> I'm running KMS on bullseye, and neither xrandr commands that I gave at the top or redshift have any effect on the colour of the display. They still do nothing with no error. 

Same. My Case:
- Fresh bullseye image
- `sudo rpi-update`
- `dtoverlay=vc4-fkms-v3d` was enabled by default in /boot/config.txt 
- `xandr`, `redshift`, and `sct` gamma commands seem to execute without error, but have no effect on my display

Thank you.

JamesH65, 2022-01-23:
I believe the fix is for KMS, not FKMS. On Bullseye, KMS is the default, not FKMS.

juergh, 2022-01-24:
Yes it's KMS only but it has been disabled again recently: 6d5309c

stacksmith, 2022-02-13:
This is insane!  It is really important to be able to adjust your gamma.  I guess I won't be using it as my main machine.

JamesH65, 2022-02-14:
I've NEVER, in 40 years of this sort of thing, bothered with changing gamma. Am I insane?

Anyway, we have some investigation to do with regard to setting up the HW to use gamma correction, there's some weirdness in *when* you can set it. Once we have figured out the problem, it will be reintroduced.

Jaciss, 2022-02-15:
A work-around might be found using DDC if your monitor supports it. A quick `sudo apt install ddcutil` and running `ddcutil detect` will probably let you know if it's worth trying. Brightness seems pretty reliable, but gamma...ymmv. Not a software/dev issue so much as lack of monitor standards.  I stuck up [a quick-and-dirty guide on the rpi stackedit](https://raspberrypi.stackexchange.com/questions/102279/how-to-change-screen-brightness-on-raspberry-pi-4/136125#136125) since this has been bugging me for some time as well.

stacksmith, 2022-02-16:
In case you are wondering, people do use gamma.  Visual perception is highly individual - as are other senses... There are super-tasters, and there are people who toss a pound of salt on everything they eat.  I happen to like it to create a pleasing, linear-looking display, need it to get the subpixel-rendered fonts not make me noxious, and absolutely require it for things like photography...  And if I get up in the middle of the night, I often adjust the gamma to match lower-brightness settings....

Pretty please!

pelwell, 2022-02-16:
It's been tried once, when it proved to be non-trivial. Getting a functional gamma control is on the list of things to do, but supporting the huge variety of displays out there is more urgent.

stacksmith, 2022-02-22:
That makes sense.  Just want to make sure people know it is not a whim - there are many people not using rpis as main machines because of that.

pmarreck, 2022-02-25:
@JamesH65 If you've been doing this for 40 years, then you may actually have "missed the memo about this"

https://www.scientificamerican.com/article/q-a-why-is-blue-light-before-bedtime-bad-for-sleep/

Light is a funny thing, and apparently not only a subjective thing but a mind-affecting thing. There's a whole group of people for example who get seasonal affective disorder, and the cure (or at least the band-aid) is... a broad-spectrum light (interestingly, literally the opposite of what we're discussing here, lol).

As a person who has struggled with various sleep issues his whole life, the introduction of things like Windows' "Night Light" and macOS' "Night Shift" (all likely inspired by https://justgetflux.com which has existed for some time) was a game-changer for me.

Been occasionally using an RPi4 for dev work because I want to be on the ARM bleeding-edge. ;) Having gotten used to the blue-light-removing features across my other devices when evening hits, the lack of it here was likely even more jarring.

I get that there are technical (possibly including hardware) challenges to implementing it, though.

r1de, 2022-03-24:
+1, add another user who wants this. I hope it can get addressed.
Might be surprised at what some RPi's are used for, I have a dashboard mounted Pi with an HDMI touchscreen, it'd be really nice to have an automatic "Night Mode" via redshift or whatever. :smile:

JamesH65, 2022-03-24:
We know this is wanted by some people, but right now don't have the time to investigate the issue. When we do have time, it will be investigated,. but right now we have a tremendous workload of many more important things to sort out.

stacksmith, 2022-04-02:
I guess I will wait until it's fixed before considering using an RPI with a monitor.

nfstoney, 2022-04-13:
+1 for the issue.

stacksmith, 2022-05-06:
OT: no disrespect intended, but I can buy a refurbished intel I7 box with 8GB for the same price as the the current (2022) raspberry-pi, shortages and gamma be damned.

JamesH65, 2022-05-08:
So, buy that Intel box until the worldwide chip supply crisis affecting absolutely everyone, is over? Our resellers sell at the RRP. Neither of which have anything to do with gamma control, so really not sure why you mentioned it here.

pelwell, 2022-05-08:
Locking until there is something to report.