raspberrypi/firmware #1276: h264 decoder and display problems on the pi4

zoff99, 2019-10-23:
we are using h264_mmal (via ffmpeg) to decode an incoming live stream and the displaying it with OMX (VideoCoreIV OpenMAX interface).
this works without any problems in the Pi3B and Pi3B+ with the latest firmware and kernel.
but on the Pi4 we get video curruption periodically.
even when using the exact same binary of our software.

since no errors do show up, we don't even know where to start looking?
can you help us?

zoff99, 2019-10-23:
pi3b+:

Aug 15 2019 12:08:48 
Copyright (c) 2012 Broadcom
version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start_x)

Linux rpi-B52B5 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

zoff99, 2019-10-23:
pi4:

Sep 24 2019 17:36:31 
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start_x)

Linux rpi-8190D 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

popcornmix, 2019-10-23:
Can you swap sdcards and confirm if the problem swaps?
e.g. is the problem in Pi3 vs Pi4 or the problem in the sdcard (e.g. firmware, kernel, or userland code)

zoff99, 2019-10-23:
no i cant, the pi4 needs a different raspian to work

popcornmix, 2019-10-23:
You can run the pi4 sdcard on the Pi3.

The firmware you have on Pi3 should be new enough to boot on a Pi4.
3d won't work on a Pi4 (assuming Pi3 is on stretch and Pi4 is on buster)
but you can boot to console, or boot without fkms (comment out ``dtoverlay=vc4-fkms-v3d``).