raspberrypi/firmware #1277: raspberrypi-bootloader 20191021-1 breaks boot on Raspberry Pi 2

0x9fff00, 2019-10-28:
After updating `raspberrypi-bootloader` in Arch Linux ARM from 20191012-1 (commit 9f4983548584d4f70e6eec5270125de93a081483) to 20191021-1 (commit ed238d3329e39a68bde03bbc2dfc293c0c3dc93a), my Raspberry Pi 2 no longer boots. It shows the colored splash screen, and then goes to black instead of booting. The red LED is still on, however the green LED is off (doesn't blink).

`/boot/cmdline.txt`:
```
root=/dev/mmcblk0p2 rw rootwait console=ttyAMA0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=ttyAMA0,115200 elevator=noop
```

`/boot/config.txt`:
```
# See /boot/overlays/README for all available options

kernel=zImage
device_tree=dtbs/bcm2836-rpi-2-b.dtb
gpu_mem=64
initramfs initramfs-linux.img followkernel
```

`/dev/mmcblk0p2` is a BTRFS partition.

**Model:** Raspberry Pi 2
**Distribution:** Arch Linux ARM
**Kernel version:** 5.3.7-1-ARCH (`linux-armv7` 5.3.7-1)

For reference, the `raspberrypi-bootloader` package consists of the following files:
* `/boot/bootcode.bin`
* `/boot/fixup.dat`
* `/boot/fixup4.dat`
* `/boot/fixup4cd.dat`
* `/boot/fixup4db.dat`
* `/boot/fixup4x.dat`
* `/boot/fixup_cd.dat`
* `/boot/fixup_db.dat`
* `/boot/start.elf`
* `/boot/start4.elf`
* `/boot/start4cd.elf`
* `/boot/start4db.elf`
* `/boot/start4x.elf`
* `/boot/start_cd.elf`
* `/boot/start_db.elf`

popcornmix, 2019-10-28:
Yes - "kernel=<blah>" was broken. Removing that and renaming zImage to kernel7.img will likely work.
We have a fix for this - I'll push it out shortly.

popcornmix, 2019-10-28:
Try latest rpi-update firmware.

0x9fff00, 2019-10-29:
The latest version works, thanks!

closed 2019-10-29 by 0x9fff00