raspberrypi/firmware #1278: Blank display on boot with 64-bit kernel - Raspberry Pi 3b+

k3lt, 2019-10-29:
I did clean installation of full raspbian image from https://www.raspberrypi.org/downloads/raspbian/ , with no changes to /boot/config.txt it's booting correctly to Desktop. (default armv7 kernel)
Then i did _rpi-update_ and added _arm_64bit=1_ to /boot/config.txt which results in blank screen on boot. Kernel information with _arm_64bit=1_ enabled:
```
uname -a
Linux raspberrypi 4.19.80-v8+ #1274 SMP PREEMPT Mon Oct 21 16:45:40 BST 2019 aarch64 GNU/Linux
```
There are some more logs/outputs i posted here: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=250730&p=1558745#p1558487
Screen is blank but rest of the functionality is there, SSH is working and also i can boot into CLI just fine, but the desktop won't show up.
After some googling i came across this: https://github.com/raspberrypi/firmware/issues/1204
I added code below into /boot/config.txt and the display started working.
_[all]
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
framebuffer_priority=2_

Initially i used all all three lines but found out that only _dtoverlay=vc4-fkms-v3d_ is required for display to work correctly. 
TV i'm using is LG 50UK6950, connected with HDMI. I was asked to also include _tvservice -d edid.dat_ so there it is.
[edid.zip](https://github.com/raspberrypi/firmware/files/3783745/edid.zip)

Let me know if you anything else.

pelwell, 2019-10-29:
This is expected behaviour - you need to use one of the KMS drivers (real or "fake") with the 64-bit kernel.

pelwell, 2019-10-29:
Hmm - ignore me. I was thinking of the upstream kernel. I've just seen @popcornmix with a display running without FKMS, so it does sound more like an EDID-handling problem.

popcornmix, 2019-10-29:
With your edid file I get a display without fkms driver:
```
pi@pi3:~ $ tvservice -s
state 0xa [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive
```
and with fkms driver:
```
pi@pi3:~ $ tvservice  -s
state 0xa [HDMI CUSTOM RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive
```
So I'm seeing basically the same hdmi mode in both cases (it's possible the fkms driver is picking subtly different timings from the edid, although that would be a little surprising for 1080p60).
Can you check you see the same both with and without the fkms line?

k3lt, 2019-10-29:
With fkms driver:
`state 0xa [HDMI CUSTOM RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive`
Without fkms driver:
`state 0xa [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive`

popcornmix, 2019-10-29:
(discussion started [here](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=250730&start=200#p1558484))
You reported:
```
Linux raspberrypi 4.19.80-v8+ #1274 SMP PREEMPT Mon Oct 21 16:45:40 BST 2019 aarch64 GNU/Linux
```
can you also report output of ```vcgencmd version```.

k3lt, 2019-10-29:
vcgencmd version:

```
Oct 21 2019 16:50:09
Copyright (c) 2012 Broadcom
version 509fb1bcb9511d963439429724a6f6dfb8251107 (tainted) (release) (start)
```

popcornmix, 2019-10-29:
Just to be sure, removing ``arm_64=1`` makes the display work again? (with fkms workaround lines removed). 

@6by9 any thoughts? If the timing chosen by fkms and firmware the same for this edid?

popcornmix, 2019-10-29:
Also, do you have another display you can test with?
I'm curious if the display is the key factor in the failure, of there is something else in your setup.

k3lt, 2019-10-29:
> Just to be sure, removing `arm_64bit=1` makes the display work again? (with fkms workaround lines removed).

Yes it does work.
Also something i just tried, with fkms and arm_64bit=1 enabled (so blank screen) i started Kodi through SSH and it displayed it just fine.
Cant test with another display currently..

popcornmix, 2019-10-29:
Can you try adding ``hdmi_drive=2`` to config.txt (without fkms driver).
(If not try ``hdmi_drive=2``)
Also try ``hdmi_group=2``
(If not try ``hdmi_group=1``)

popcornmix, 2019-10-29:
Output of ``vcgencmd dispmanx_list`` when not displaying an image might be interesting.

popcornmix, 2019-10-29:
> vcgencmd dispmanx_list - no output with dtoverlay=vc4-fkms-v3d disabled

Interesting. If you disable "boot to desktop" in raspberry pi configuration, is there any difference.

k3lt, 2019-10-29:
Booting to CLI with dtoverlay=vc4-fkms-v3d enabled:
`display:2 format:XRGB8888 transform:0 layer:-127 1920x1080 src:0,0,1920,1080 dst:0,0,1920,1080 cost:1156 lbm:0`

With dtoverlay=vc4-fkms-v3d disabled:
`display:2 format:RGB565 transform:0 layer:-127 1920x1080 src:0,0,1920,1080 dst:0,0,1920,1080 cost:889 lbm:0`

popcornmix, 2019-10-29:
can you add ``start_debug=1`` and (without fkms and desktop enabled) report output of:
```
sudo vcdbg log msg
sudo vcdbg log assert
```

popcornmix, 2019-10-29:
Can you try booting to console, running:
```sudo rm /usr/share/X11/xorg.conf.d/99-fbturbo.conf```
(don't worry it gets recreated every boot)
and then ``startx``

k3lt, 2019-10-29:
Says there is no such file or directory.

popcornmix, 2019-10-29:
You have fkms disabled?

k3lt, 2019-10-29:
Oops forgot to disable it, here is output of `startx`

`X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.15.0-48-generic armv8l Raspbian
Current Operating System: Linux raspberrypi 4.19.80-v8+ #1274 SMP PREEMPT Mon Oct 21 16:45:40 BST 2019 aarch64
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=5e3da3da-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait usbhid.mousepoll=0 quiet splash plymouth.ignore-serial-consoles
Build Date: 05 June 2019  12:49:54PM
xorg-server 2:1.20.4-1+rpt1 (https://www.debian.org/support)
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/home/pi/.local/share/xorg/Xorg.0.log", Time: Tue Oct 29 17:24:25 2019
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE)
Fatal server error:
(EE) parse_vt_settings: Cannot open /dev/tty0 (Permission denied)
(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional information.
(EE)
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error
Nie udało się uzyskać deskryptora pliku wskazującego na konsolę
`

k3lt, 2019-10-29:
Oh wait, running `sudo startx` loaded the desktop.

popcornmix, 2019-10-29:
Are you running startx from ssh? You should be running it from keyboard (without sudo).

That fix will only last one boot. A permanent fix would be to edit:
/usr/lib/systemd/scripts/gldriver_test.sh

to this: http://paste.debian.net/1111722/

(take a backup copy first).

popcornmix, 2019-10-29:
Okay, thanks for report and testing.

@XECDesign could you update /usr/lib/systemd/scripts/gldriver_test.sh to this:
http://paste.debian.net/1111724/

(I've tweaked it slightly from previous version)

It also treats aarch64 (along with fkms/kms) as a reason to disable fbturbo which allows non-fkms usage of 64-bit kernel.

XECDesign, 2019-10-30:
The updated package should be public within an hour.

GrabbenD, 2023-03-03:
Just encountered this issue today when running a custom kernel (6.1 & 6.2). Thanks for sharing the workaround
```
# arm_64bit=1
dtoverlay=vc4-fkms-v3d
```

popcornmix, 2023-03-06:
@GrabbenD it seems unlikely this is your issue.
This issue was caused by using fbturbo on 64-bit.

If you had this issue then this previous comment should (temporarily) resolve it:

> Can you try booting to console, running:
sudo rm /usr/share/X11/xorg.conf.d/99-fbturbo.conf
(don't worry it gets recreated every boot)
and then startx

GrabbenD, 2023-04-14:
@popcornmix It's most likely a different issue but the solution was the same. When I follow the official instructions for compiling a Raspberry Pi 4B+ kernel on Debian 12 I have to use `dtoverlay=vc4-fkms-v3d` to avoid blackscreen when booting. Can't wrap my head around why.
https://www.raspberrypi.com/documentation/computers/linux_kernel.html